fairecasoimeme/Zlinky_TIC #162: Probleme de connexion au mode routeur

landaisbenj, 2023-01-07:
Bonjour.

J'ai un petit capteur de contact dans le jardin. j'espérai le voir se connecté au Zlinky car la clé conbee est un peu loin et derrière le mur, mais je n'y arrive pas. 

Est-ce qu'il y a quelque chose à savoir pour pouvoir le faire ? Ou alors c'est le zlinky qui est moins puissant qu'une clé conbee ? Merci pour vos retours.

ssenart, 2023-01-11:
Bonjour,

Je confirme que depuis le passage à la V12 (j'étais avant en V4), mon bouton sonoff juste à côté du zlinky ne veut plus passer par zlinky (donc utiliser sa fonction de routeur) pour router ses messages et du coup, il reste off-line. En V4, il passait naturellement et systématiquement par le zlinky.

En contournement, j'ai rajouter à proximité une ampoule Hue pour que le sonoff puisse refonctionner. Ses messages peuvent maintenant atteindre le coordinateur via la Hue.

1) Est-ce un problème connu ? Je n'ai rien trouvé de similaire des les Issues.

2) Y-a-t'il moyen de revenir en V4 ? J'ai fait une mise à jour OTA via zigbee2mqtt pour passer de V4 en V12.

Merci d'avance

landaisbenj, 2023-01-11:
Ah je suis heureux de voir que c’était possible avant, et que je ne suis pas seul. J’espère qu’on pourras donc avoir une solution. 

Je crois qu’un downgrade est possible pas pas en dessous de 5. Peut-être que le mode routeur fonctionne en v11. J’ai pas trop envie de tout casser à essayer, mais s’il n’y a pas de réponse je verrais bien si je peux effectuer quelque test.

fairecasoimeme, 2023-01-11:
Oula, attention il y a des erreurs d'interprétation.

Peu importe la version "legacy" du ZLinky, la fonction de routage des enfants ne diffère pas. 
Il faut savoir que la gestion des routes / maillage des réseau zigbee est intégré dans des librairies qui sont censés répondre à la norme Zigbee 3.0. Le firmware du ZLinky n'a pas d'impact sur ce point.

En zigbee, le choix pour un objet de passer par telle ou telle route est défini par ce qui semble être (pour l'objet qui demande) le chemin le plus approprié, soit par le niveau de qualité emission/réception ou chemin le plus court (on ne sait pas trop, pas beaucoup de doc dessus). Le changement de route pour un objet peut prendre énormément de temps (parfois plus de 24h), ce sont les différentes transactions entre objets zigbee qui font qu'à un moment donné, une nouvelle route est définie.

Si un objet à côté d'un routeur ne passe pas par ce dernier, les raisons peuvent être multiples et difficile de savoir pour quelle raison.

Pour forcer un objet à passer par un routeur, en général, on fait sortir l'objet du réseau puis on le ré-appaire en le collant au routeur pour augmenter les chances. (Certains plugins permettent l'appairage à partir de routeur aussi)

Enfin le downgrade est possible en passant par la mise à jour Non OTA, il suffit de flasher avec un binaire plus ancien.

ssenart, 2023-01-11:
Merci [fairecasoimeme](https://github.com/fairecasoimeme) pour cette explication. Quelles librairies Zigbee as-tu utilisé ?

fairecasoimeme, 2023-01-12:
J'utilise la librairie zigbee de NXP

vincele, 2023-05-16:
Bonjour,
et c'est la même version de cette lib NXP qui est utilisée dans toutes les version du firmware zlinky ?
Quelle est sa provenance, c'est FOSS, binaire, ou bien ?