fairecasoimeme/Zlinky_TIC #163: Update firmware non OTA error

jflefebvre06, 2023-01-11:
Bonjour,
La mise à jour ota ne fonctionnant pas j'ai tenté de faire une MAJ non OTA et j'obtiens systematiquement l'erreur "error read"

J'ai tenté
./DK6Programmer -V6 -s COM9 -P 115200 -e FLASH -Y -p C:/NXP/DK6ProductionFlashProgrammer/ZLinky_router_v12.bin
./DK6Programmer -V6 -s COM9 -P 115200 -Y -p C:/NXP/DK6ProductionFlashProgrammer/ZLinky_router_v12.bin
./DK6Programmer -V6 -s COM9 -P 115200 -Y -p ./ZLinky_router_v12.bin

Une idée ?

jflefebvre06, 2023-01-11:
[image]

Oui j  ai bien apppuyé sur la touche de flash lors de la connexion

jflefebvre06, 2023-01-11:
PS: J ai testé depuis 2 postes différents, meme resultat

fairecasoimeme, 2023-01-11:
Pouvez vous faire une capture du gestionnaire de périphériques ? avec les ports COM ?

jflefebvre06, 2023-01-11:
J'ai bien l'impression que ma Zlinky_tic est complètement out, apres l'avoir rebranchée sur le linky la led ne s'éclaire meme plus

bemble, 2023-03-10:
Vous appuyez bien sur bouton flash sur le zlinky en même temps que vous mettez la clé dans l'ordinateur ?

J'ai eu ce soucis et c'était la cause du problème.

fairecasoimeme, 2023-03-13:
Il arrive lors de mauvaises manip ou mise à jour du Linky que l'alimentation de ce dernier se mette en protection. Il faut laisser débrancher pendant 10 min puis rebrancher. Cela débloque souvent