fairecasoimeme/Zlinky_TIC #165: Stabilité Zlinky Lixee - Multiples appairages et pertes de signal.

HpNoTiQ56, 2023-01-18:
Bonjour,
J'utilise le Zlinky chez moi depuis le début de la semaine sur une VM linux avec Zigbee2MQTT. Je récupère ensuite les infos sur HA avec Z2M Proxy.

1er appairage réussi du premier coup. Après la première nuit, plus de connexion et perte de l'historique.
Désappairage et réappairage au bout de plus de 10 essais, cela a marché 36h et reperte des signaux depuis midi 👍 
[image]
Ce soir, nouvel tentatives de désappairage/réappairage plus d'une dizaine de fois et un dernier interview qui fonctionne mais qui n'a duré que quelques secondes.
[image]

Dois-je renvoyer mon module à votre avis?

Ma configuration : 
- Linky dans la rue dans coffret.
- Clé conbee II sur VM Debian + Z2M sur Freebox Delta (mes autres périphériques Zigbee fonctionnent sans coupure). à 7m du compteur.

J'ai pu faire la mise à jour OTA vers la version 12 avant hier.

fairecasoimeme, 2023-01-19:
Dans un premier temps, il faut regarder le LQI de l'appareil. Avez-vous la version avec antenne externe ? Cela ressemble à un problème de portée.
Ensuite sur Z2M, il semble qu'après une mise à jour, il faut redémarrer Z2M et réappairer l'appareil.

HpNoTiQ56, 2023-01-19:
Le LQI était entre 190 et 255 quand cela marchait. En fonction des appairages, j'ai entre 50 et 190 mais je n'arrive pas à stabiliser la connection.
J'ai la version sans antenne. J'ai bien déjà fait toutes les manips sur Z2M.
Est-il possible de faire un hard reset pour réinitialiser les paramètres du module?

fairecasoimeme, 2023-01-19:
Bizarre cette chute de LQI, la mise à jour n'intervient pas du tout sur ce point. un LQI entre190 et 255, c'est quand l'appareil zigbee est collé à la passerelle ou un routeur...N'y a-t-il pas eu des modifs du réseau zigbee ?

le hard reset se fait lorsque l'on appuie 10 sec sur le bouton Link.

tex0l, 2023-01-20:
J'ai un comportement similaire après màj OTA de 4000-0011 vers 4000-0012 avec ZHA (je n'avais jamais eu de souci de ce type avant, et cela fait plusieurs mois que je l'ai):
- toutes les valeurs ont freezé d'un coup quelque temps après la màj OTA;
- la connexion était hachée (càd que les graphs affichent la valeur freezée par intermittence — je n'ai plus les graphiques puisque j'ai supprimé le Zlkiny et que je l'ai ajouté à nouveau);
- j'ai tenté de reconfigurer l'appareil via ZHA, ça n'a pas fonctionné et a présenté des erreurs;
- j'ai physiquement débranché le zlinky, attendu 10s, je l'ai rebranché;
- j'ai retenté de reconfigurer l'appareil via ZHA, ça a fonctionné;
- 4-5h après ça s'est retrouvé dans le même état qu'après la màj OTA à nouveau;
- j'ai débranché et rebranché sans attendre puis tenté de reconfigurer, ça n'a pas fonctionné;
- j'ai supprimé le ZLinky du réseau, je l'ai débranché, je l'ai rebranché sans attendre, j'ai appuyé sur le bouton de reset pendant 10s (ça a bien éteint la lumière bleue puis l'a rallumée), la procédure d'installation a freezé en plein milieu;
- je l'ai débranche, attendu 10s, rebranché, appuyé sur le bouton reset 10s et l'ai ajouté au réseau, cela fonctionne et tient depuis 9h30 ce matin (soit il y a 3h).

J'ai un réseau avec beaucoup d'appareils, donc je vais tenter de passer au firmware limited, mais je n'ai pas module USB-TTL, j'en ai commandé un.

HpNoTiQ56, 2023-01-20:
Pour ma part, j'ai tout essayé : ma passerelle étant une conbee II fraichement acheté, je la renvoie au profit d'une Sonoff avec antenne.
J'essaierais aussi avec une rallonge et la box à quelques cm !

HpNoTiQ56, 2023-01-21:
Nouvelle tentative hier soir après une deconnection du module de plus de 24h!
Appairage instantané et LQI entre 255 & 180. J'espère que ca va perdurer! 
[image]

fredleger, 2023-01-21:
idem impossible de l'appairer après la mise à jour en 4000-0012 sur zigbee2mqtt. Tout marchait bien avant la mise à jour.
Je met une lampe philips hue avec une rallonge a 10 cm du compteur pile en face porte ouverte. La lampe a un très bon LQI et pas moyen de voir la clé lixee.

tex0l, 2023-02-02:
> J'ai un comportement similaire après màj OTA de 4000-0011 vers 4000-0012 avec ZHA (je n'avais jamais eu de souci de ce type avant, et cela fait plusieurs mois que je l'ai):
>
>     * toutes les valeurs [...]

Je n'ai plus eu de souci d'appairage depuis la dernière opération décrite dans ce post, je n'ai pas changé le firmware vers le no-router.

KipK, 2023-03-01:
Module neuf ici, j'en ai pris un deuxieme pour être sûr.
Sous firmware v13 Limited, le module se deconnecte tout seul après plusieurs heures comme avec le modle précédent ( qui fonctionne parfaitement en firmware v9 ) 
La Led bleu reste allumée en permanence quand je retrouve le module deconnecté.
 il ne revient pas sur le réseau une fois dans cet état , sauf après un power cycle.

fairecasoimeme, 2023-03-01:
Si tu as le module USB TTL, peux tu passer en version "legacy" (mise à jour non-OTA) sur le dernier module ? (Normalement la mise à jour est plus simple car il n'y a pas de PAD mais des trous). J'avais repéré chez un autre utilisateur un problème de routage qui bloque les envois / réception

KipK, 2023-03-01:
il me semble avoir eu le meme soucis avant en firmware normal sur la V12 et V13, puis avoir reswitché en V9 no router. 
Je viens de switcher sur la V10 pour voir je vais monter de version par étape voir quand ca cloche.
Quand je regarde sous Deconz le graph, je vois pleins d'appareils qui affichent quand même une route possible par le module, en mode Limited. 
En fait idéalement il faudrait un firmware pour que l'appareil soit reconnu comme vrai end point et non routeur ( même limité ) . Quitte a simuler une batterie en envoyant la tension 3.3v vu que ca à l'air dans les specs.

fairecasoimeme, 2023-03-03:
Étrange que vous ayez le même problème avec la V13 legacy. En mode limited, normalement, il y a 1-2 routes possibles.
J'ai déjà fait énormément de tests en mode ZED et ce n'est pas du tout concluant ou adapté au besoin.
Si vous pouvez mettre un ordinateur à portée du Linky, je peux vous proposer une version du ZLinky de test qui permettra de capturer ce qui se passe et mieux comprendre ce qui bloque chez vous.

KipK, 2023-03-03:
Depuis le passage en v10 plus de deconnection pour l'instant ( avec la 13 je recevait qd meme des messages de jeedom qui m'indiquait la deco / reco du module. En general quand ca commencait comme ça dans les heures qui suivent il était out. 
Pas encore essayé la 11. 
Contrairement au soucis que j'avais eu au tout debut avec le module qui finissait avec la LED etteinte, maintenant quand il est injoignable, la led reste allumée pour info mais constante, sans aucune modulation rapide comme quand ca fonctionne. C'est peut être un nouveau problème.

Je veux bien le firmware mais je peux le flasher OTA ? Je peux editer les n: de versions des fichiers ota facilement depuis le client Deconz, ce qui me permet de downgrader.

fairecasoimeme, 2023-03-03:
Quand je parlais de la V13 c'est la version legacy que vous devez faire non OTA. 
Pour la version que je vous propose c'est une version hardware avec un module USB TTL.

KipK, 2023-03-03:
Oui j'ai bien compris, mais je vais éviter là car je n'ai qu'un module et si il se passe quelque chose au flashage je vais me retrouver sans délestage. Et vu le froid en ce moment. :)
Je vais peut être pouvoir récupérer un deuxième ce qui me permettra de tester ça correctement.
pour le module TTL , ça va être compliqué en l'état de l'installation. Mon Compteur est dans une pièce sans prise electrique, difficile de laisser meme un raspberry pi branché en permanence. 
Mais j'ai une batterie qui me permet de le faire tourner une dizaine d'heures, ca ppourrait être suffisant.

fairecasoimeme, 2023-03-03:
Pourtant, je pense sincèrement que votre problème peut provenir de la version "limited". J'ai eu un cas similaire au votre où le blocage provenait de là. Pour votre confort, je peux vous envoyer un autre ZLinky avec le firmware V13 legacy. Une fois vos tests effectués, vous me renvoyer l'autre.
Pour cela envoyer moi plutôt un mail : contact[at]lixee.fr

KipK, 2023-03-03:
J'ai déjà testé avec le firmware normal sur le module précédent. C'est comme ça d'ailleurs que je l'ai probablement flingué en voulant mettre le limited pour tester.

Mais c'est gentil de proposer je peux retester avec un autre module oui.
En attendant je continue mon test de remontée de version tout les 3j si ca tiens.
Je suis à la l'aise avec les procédures de flash mais vu l'expérience précédente et les conséquences quand j'ai plus la teleinfo ( chauffage electrique et chargeur VE 32A ) , je préfère pas tenter :)

KipK, 2023-03-10:
Alors plutôt bonne nouvelle, j'ai continué ma remontée en version ( toujours en limited ), et finalement sur la 13 sans soucis deplus plus de 48h ce qui n'était pas arrivé jusque là avec cette version.

Depuis j'ai changé quelques trucs sur le ddf de deconz, réappairé complètement ( d'ailleurs le passage de v11 à v12ou13  oblige un rappairage complet sinon beaucoup de soucis j'en ai fais les frais, ce n'était pas le cas sur les autres montées en version. ) , viré des propriétés dont je me servais pas. On va voir comment ça tiens.

 @fairecasoimeme , j'ai une idée pour simplifier le passage de limited vers normal et vis versa. 

Il faudrait 2 images spéciales qui permettrait de faire la transition:
- de Limited vers Normal:  une image en image_type 2 avec une version à 0, packagée ota comme un n: de version qui sera toujours plus elevé que les versions à venir , mais qui une fois installée demandera en ota une image_type 1.
Il ne suffira qu'à refaire une ota avec l'image type 1 souhaitée.

-de normal vers limited: image_type 1, version compilé sur 0, version dans le package ota sur 65534, demande en ota image_type 2. 

Avec ces 2 images ça permettrait de switcher d'un type à l'autre sans avoir à flassher ttl.

Bazarov42, 2023-03-14:
J'ai aussi des pertes et ré-appairage "sauvage", ça doit faire 2-3 semaines que je n'arrivais pas à appairer et la en revenant de la balade du chien, pouf c'est revenu ^^

J'ai ajouté une prise zigbee à côté pour améliorer le lien comme mon répéteur n'a pas l'air de plaire au Zlinky
La prise et le répéteur son à la même distance du Zlinky il doit y avoir 20 cm de différence. 
 
[image]

Bazarov42, 2023-03-20:
Pour ma part, tout refonctionne et ça a l'air de tenir, j'ai eu quelques décrochage et à force de redémarrer le module, redémarrer le système c'est revenu

antoinelibert, 2023-03-20:
De mon côté (et pour la première fois), je suis en V12 depuis sa sortie (nov 22) et je n'ai eu aucune coupure depuis. Maintenant, j'hésite à mettre à jour ... 😅

HpNoTiQ56, 2023-03-22:
Première coupure depuis mon dernier post : freeze jusqu'à débranchement/rebranchement sur le compteur !