fairecasoimeme/Zlinky_TIC #166: ZLinky ne s'allume pas

bosr, 2023-01-19:
Bonjour,

J'ai reçu mon ZLinky la semaine dernière, et je l'ai branché ce matin pour la première fois : la led ne s'allume pas, et il ne veut pas s'appairer. J'ai tenté aussi de le réinitialiser. Que puis-je faire d'autre ?

J'ai également envoyé cette info avec mon num de commande sur contact@....fr

Merci !

bosr, 2023-01-19:
Merci pour les échanges par mail. Pour l'instant, le problème semble dû à l'alimentation de mon Linky (Sagem) : la tension à vide est de 1 Vrms au lieu de 13 Vrms.

Je mettrai à jour quand j'aurai plus d'infos.

fsegouin, 2023-01-24:
@bosr J'ai le meme soucis, as-tu trouve l'origine du probleme ? Merci !

bosr, 2023-01-24:
Salut, le problème n'est pas encore résolu. Voilà où j'en suis : 

- J'ai contacté Enedis, qui m'a demandé de contacter mon fournisseur (EDF) pour faire activer le mode standard : c'est la demande F185 (note ça car EDF n'a pas l'habitude de faire cette demande et j'ai dû faire une petite partie de ping-pong avec eux et Enedis).
- EDF a effectué la demande auprès d'Enedis, et Enedis s'est exécuté dès le lendemain à distance.
- Le tableau de mon compteur linky indique bien désormais "Mode standard TIC", mais comme prédit par @fairecasoimeme (via mail), ça n'a rien changé à la tension d'alim entre I1 et A
- J'ai recontacté Enedis pour indiquer le problème, et rebelote, ils m'ont indiqué devoir passer par EDF pour faire une demande F920 afin de dépêcher un technicien à mon domicile.
- Je me suis exécuté, et je sais qu'Enedis a bien reçu cette demande, et j'attends l'appel de leur tech pour prendre rdv.

Pour l'instant, je dirais que les conseillers Enedis et EDF sont sympas, mais c'est un peu galère car ils n'ont pas l'habitude de la demande de passage en mode standard (rarissime) et le problème d'alim est également très rare.

Voilà pour l'instant !

bosr, 2023-01-24:
Je note tout cela ici afin d'aider ceux qui auraient un pb similaire, et à terme permettre d'améliorer la section dépannage de la doc. Et pour que ce soit bien clair, à ce stade, je n'ai aucune raison de penser que le ZLinky soit en cause. Je vais modifier le titre d'ailleurs.

bosr, 2023-02-17:
Bonjour,

Enedis a changé mon compteur la semaine dernière, mais le ZLinky ne s'allumait toujours pas, donc @fairecasoimeme a procédé à un échange. Le nouveau ZLinky, reçu hier, fonctionne très bien.

Pour info, c'est un peu la honte, mais bon, j'avais mesuré la tension RMS entre I1 et A avec un multimètre, et évidemment ça ne peut pas retourner une mesure correcte étant donné que la fréquence de la porteuse de ce signal (50 kHz) est bien supérieure à la bande passante du multimètre. Donc, évitez de perdre du temps avec cette mesure naïve, et procédez à un échange. Cela évitera aussi de faire perdre du temps à Énedis ;)

Bonne continuation !

closed 2023-02-17 by bosr