fairecasoimeme/Zlinky_TIC #167: Comprendre PJOURF+1 PPOINTE1 NJOURF et NJOURF+1

technarf, 2023-01-20:
Bonjour,
Je viens de basculer en abonnement TEMPO et si j'arrive à connaître le mode actuel du tarif avec la commande LTARF (current_price) qui m'affiche en toutes lettres le tarif actuel (HP BLEU pour l'instant), je n'arrive pas à trouver comment obtenir la couleur du prochain jour.
Je suis en mode standard sur du triphasé, et mon Zlinky est connecté via Zigbee2Mqtt. 
J'ai l'impression que l'information va se trouver dans PJOURF+1 (days_profile_current_calendar) et/ou PPOINTE1 (days_profile_next_calendar), ou encore NJOURF (days_number_current_calendar) et/ou NJOURF+1 (days_number_next_calendar).
Mais je n'arrive pas à "traduire" l'information, et je n'ai trouvé nulle part comment interpréter tout ça.
A l'heure actuelle, PJOURF+1 me retourne :` 00004001 06004002 16004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE` ,
 PPOINTE1 me retourne : `00004003 06004004 16004003 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE`
NJOURF : `0` et NJOURF+1 : `0`
Le jour actuel est un jour Rouge, mais mon linky et le zlinky m'annoncent tous les 2 un jour bleu (le changement de contrat a eu lieu aujourd'hui avec l'intervention d'un technicien ce matin, c'est peut-être pour ça) et demain sera un jour blanc.
Serait-il possible d'ajouter dans le readme.md (ou ailleurs), comment interpréter les données renvoyées par ces 4 commandes ?
Merci d'avance pour votre aide.

fairecasoimeme, 2023-01-22:
Vous pouvez retrouver les informations sur le lien suivant : 
[Enedis-NOI-CPT_54E - Note externe](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwjd_4zO09v8AhW-TKQEHWrDA4YQFnoECAUQAQ&url=https%3A%2F%2Fwww.enedis.fr%2Fmedia%2F2035%2Fdownload&usg=AOvVaw0Bn9Pv64VK2dhbtmrZD8YD)

Pour les avoir l'information des couleurs du lendemain, il faut utiliser l'étiquette  STGE

technarf, 2023-01-23:
Bonjour, 
Merci beaucoup pour l'info, STGE semble bien remonter chez moi, et avec le lien, j'ai vu comment analyser la valeur...
Y'a plus qu'à attendre que ça bouge, et faire des scénarios pour le rendre visible au premier coup d'oeil...

closed 2023-01-23 by technarf