fairecasoimeme/Zlinky_TIC #168: Le calcule de l'energy dans home assistant  ne marche pas !

mcarbonneaux, 2023-01-23:
j'ai bien configurer la source de consommation dans le dashboard energy. j'utilise l'index EAST, j'ai indiquer un prix fixe.

[image]

j'ai bien les valeurs de consommation qui progresse sur le dashboard energy.

[image]

mais le calcule du prix est faux ou zero (çà dépend des fois) ...

une idée d’où ça viens ?

fairecasoimeme, 2023-01-26:
Bonjour,
Je pense que vous n'êtes pas forcément au bon endroit pour avoir une réponse. C'est plus un problème que vous avez avec le paramétrage de home-assistant
Fred

mcarbonneaux, 2023-02-05:
en fait je sait pas si c'est les valeur retourné par zlinky ou si c'est lier a la configuration.
car j'ai juste ajouter l’attribut EAST qui remonte bien les valeurs, mais le calcule du cout ne se fait pas (a prix fix).

fairecasoimeme, 2023-02-06:
Dans tous les cas, ça ne concerne pas le ZLinky.

slamp, 2023-02-18:
Essaye de poser la question sur le forum HACF
https://forum.hacf.fr/