fairecasoimeme/Zlinky_TIC #172: Papp does not update after ha2023.2.1

Geoffrey54, 2023-02-04:
La puissance apparente ne se met plus a jour avec la dernière version de HA2023.02.01 et z2mqtt 1.30.0-1.

après plusieurs redemarrage de ha et z2mqtt pas de mise a jour. Le module est dans la dernière version également (12)

merci

fairecasoimeme, 2023-02-06:
Il faut je pense vérifier les logs car là, impossible de déduire quoi que ce soit

WarC0zes, 2023-02-07:
la puissance apparente fonctionne chez moi en Z2M 1.30.1 et core 2023.2.2
[image]

Geoffrey54, 2023-02-08:
J’ai supprimé le device sans forcer, ensuite reppairer et redémarrer z2m et c’est ok. On peit clore

closed 2023-02-08 by Geoffrey54