fairecasoimeme/Zlinky_TIC #173: Incohérence lixee_zlinky_tic_tier_1_summation_delivered : données qui ont diminué

luke7101, 2023-02-04:
Bonjour, en observant l'entité `sensor.lixee_zlinky_tic_tier_1_summation_delivered` introduite avec la nouvelle version 2023.2 de Home Assistant je me suis rendu compte de l'incohérence suivante dans l'historique.
Etant donné que je copie le contenu dans un sensor dont la classe est "total_increasing" j'ai eu l'erreur suivante dans les logs : 
`WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.lixee_zlinky_tic_metering_hc from integration template has state class total_increasing, but its state is not strictly increasing. Triggered by state 14705117 (14705.127) with last_updated set to 2023-02-03T23:55:08.303052+00:00. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+template%22`

[image]

[image]

fairecasoimeme, 2023-02-06:
Difficile de dire d'où ça peut venir mais côté ZLinky, il n'y a aucun traitement d'effectué sur les données du Linky. Peut-être regarder s'il n'y a pas un arrondi qui est fait mais très probablement côté HA

luke7101, 2024-01-22:
Oui en effet c'était bien un capteur qui effectuait un calcul arrondi, merci.

closed 2024-01-22 by luke7101