fairecasoimeme/Zlinky_TIC #174: HomeAssistant - base de données remplie à cause d'extra_states_attributes

helepeut, 2023-02-05:
Bonjour, 

Je vous écris car j'ai actuellement un soucis en cours de résolution avec bdraco, qui travaille sur la partie MariaDB de HomeAssistant.
Depuis quelques temps, j'ai d'énormes lenteurs dans la purge des données d'historique.

En regardant de plus prêt, il semblerait qu'il y ait un soucis avec la quantité de données remontées par le ZLinky_TIC (possiblement en mode PRODUCTION). 

bdraco me demande de contacter le développeur de l'intégration - je ne sais pas si c'est à vous que je dois m'adresser, de ce fait j'ai aussi laissé un message à l'équipe Zigbee2MQTT que j'utilise pour remonter les données du ZLinky_TIC. En espérant que cela puisse faire avancer la science !

Le message de bdraco  est le suivant : 

_Please send a note to the dev that develops that integration

https://developers.home-assistant.io/docs/core/entity#generic-properties

DANGER
Entities that generate a significant amount of state changes can quickly increase the size of the database when the extra_state_attributes also change frequently. Minimize the number of extra_state_attributes for these entities by removing non-critical attributes or creating additional sensor entities._

Le thread concernant le problème en question : https://github.com/home-assistant/core/issues/87352
La discussion ouverte avec Z2M : https://github.com/Koenkk/zigbee2mqtt/discussions/16549

fairecasoimeme, 2023-02-06:
@vk496 did you see this issue. What do you think ?

vk496, 2023-02-06:
yes, this is a well-known issue. That's why I created a kind of "filters" to not expose attributes that doesn't make sense with your current tarif. My advice for this issue:

- Select the proper linky_mode, phase, production in Z2M
- Select a tarif or which attributes you want with tic_command_whitelist (extra options Z2M)
- Adjust the minimum time for notifications for zlinky in Z2M
- Restart Z2M to take effect

The first two points can be replaced with disabling the entities from Home Assistant (I would personally do both, just in case)

helepeut, 2023-02-06:
Thanks !
This is more or less what I did early January when I saw the quantity of messages and attributes that were surfacing. So I decided to just keep the 2 attributes I wanted for my Production Linky (using the whitelist). But the database was already filled with these data, and I can't purge it in reasonable time. 
Bdraco used this database to update the db optimizer (I guess for a next release), and he just pointed me this warning in order to send it back to you.
It might be worth adding this warning in the documentation ?

ellmout, 2023-07-12:
Hello @BrunoSpy,
In Z2M, in Settings, Home Assistant Integration, disable "Home Assistant legacy entity attributes".
When enable : 
```
  # Optional: Home Assistant legacy entity attributes, (default: shown below), when enabled:
  # Zigbee2MQTT will send additional states as attributes with each entity. For example,
  # A temperature & humidity sensor will have 2 entities for the temperature and
  # humidity, with this setting enabled both entities will also have
  # an temperature and humidity attribute.
```

BrunoSpy, 2023-07-12:
Thank you. That was well hidden :)