fairecasoimeme/Zlinky_TIC #175: Aucune Remontée suite à l'achat d'un Zlinky

mattzeer, 2023-02-07:
Hello Suite à l'achat d'un Zlkiny, je n'ai aucune informations de la consommation qui remontent

J'ai mis à jour le firmware vers la dernière version (4000-0012)
Mon linky est en TIC Standard (ce qui a été changé dans les paramètres du zlinky)

[image]

Y a t-il un réglage qui me manque ou quelque chose que j'ai mal configuré ?

fairecasoimeme, 2023-02-07:
Bonjour,

Il y a des informations qui ont été remontées, du coup, il faut verifier les logs pour voir s'il y a des timeouts ou s'il manque des demande d'informations. Le LQI est assez faible donc à vérifier

mattzeer, 2023-02-10:
Hello,

Voici le genre de log que j'ai lié au zlinky en  DEBUG
[zlinky.log](https://github.com/fairecasoimeme/Zlinky_TIC/files/10706446/zlinky.log)

[image]

nerthazrim, 2023-03-24:
Bonjour @mattzeer, as-tu trouvé la source de ton problème ?

J'ai réussi à appairer mon Zlinky_TIC tout neuf avec mon install Home Assistant + Zigbee2MQTT sur RPI3 mais j'ai un rendu similaire au tiens dans les données remontées par le Zlinky.

Etonnamment j'ai bien EASF01 de populée.
Par contre, impossible d'avoir autre chose que 0 kWh sur EASD01, qui devrait contenir ma conso instantanée, si j'ai bien suivi ?

A noter que je suis en Standard BASE, monophasé sans production

Iwhite67, 2023-06-02:
Hello, j'ai le meme style de probleme sous jeedom

`4959|[2023-06-02 22:41:46][DEBUG] : http://127.0.0.1:8089/device/attributes type : POST
4960|[2023-06-02 22:41:46][DEBUG] : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":1794,"cluster_type":"in","attributes":[0,256,258,260,262,264,266],"allowCache":0}
4961|[2023-06-02 22:41:50][INFO] : [Maison][LiXee.ZLinky_TIC 00:15:8d:00:05:d2:a9:0a] Erreur lors de la requete : http://127.0.0.1:8089/device/attributes(POST), data : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":1794,"cluster_type":"in","attributes":[0,256,258,260,262,264,266],"allowCache":0} erreur : {"state":"error","result":"[0x54d8:1:0x0702]: Message send failure","code":0}
4962|[2023-06-02 22:41:50][DEBUG] : http://127.0.0.1:8089/device/attributes type : POST
4963|[2023-06-02 22:41:50][DEBUG] : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":1794,"cluster_type":"in","attributes":[32],"allowCache":0}
4964|[2023-06-02 22:41:55][INFO] : [Maison][LiXee.ZLinky_TIC 00:15:8d:00:05:d2:a9:0a] Erreur lors de la requete : http://127.0.0.1:8089/device/attributes(POST), data : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":1794,"cluster_type":"in","attributes":[32],"allowCache":0} erreur : {"state":"error","result":"[0x54d8:1:0x0702]: Message send failure","code":0}
4965|[2023-06-02 22:41:55][DEBUG] : http://127.0.0.1:8089/device/attributes type : POST
4966|[2023-06-02 22:41:55][DEBUG] : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":2820,"cluster_type":"in","attributes":[1288,2312,2568,1290,2314,2570,1293],"allowCache":0}
4967|[2023-06-02 22:41:59][INFO] : [Maison][LiXee.ZLinky_TIC 00:15:8d:00:05:d2:a9:0a] Erreur lors de la requete : http://127.0.0.1:8089/device/attributes(POST), data : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":2820,"cluster_type":"in","attributes":[1288,2312,2568,1290,2314,2570,1293],"allowCache":0} erreur : {"state":"error","result":"[0x54d8:1:0x0b04]: Message send failure","code":0}
4968|[2023-06-02 22:41:59][DEBUG] : http://127.0.0.1:8089/device/attributes type : POST
4969|[2023-06-02 22:41:59][DEBUG] : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":2820,"cluster_type":"in","attributes":[1295],"allowCache":0}
4970|[2023-06-02 22:41:59][DEBUG] : http://127.0.0.1:8089/device/attributes type : POST
4971|[2023-06-02 22:41:59][DEBUG] : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":1794,"cluster_type":"in","attributes":[0,256,258,260,262,264,266],"allowCache":0}
4972|[2023-06-02 22:42:04][INFO] : [Maison][LiXee.ZLinky_TIC 00:15:8d:00:05:d2:a9:0a] Erreur lors de la requete : http://127.0.0.1:8089/device/attributes(POST), data : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":2820,"cluster_type":"in","attributes":[1295],"allowCache":0} erreur : {"state":"error","result":"[0x54d8:1:0x0b04]: Message send failure","code":0}
4973|[2023-06-02 22:42:04][DEBUG] : http://127.0.0.1:8089/device/attributes type : POST
4974|[2023-06-02 22:42:04][DEBUG] : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":65382,"cluster_type":"in","attributes":[1,2,4,5,6,7,8],"allowCache":0}
4975|[2023-06-02 22:42:08][INFO] : [Maison][LiXee.ZLinky_TIC 00:15:8d:00:05:d2:a9:0a] Erreur lors de la requete : http://127.0.0.1:8089/device/attributes(POST), data : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":1794,"cluster_type":"in","attributes":[0,256,258,260,262,264,266],"allowCache":0} erreur : {"state":"error","result":"[0x54d8:1:0x0702]: Message send failure","code":0}
4976|[2023-06-02 22:42:08][DEBUG] : http://127.0.0.1:8089/device/attributes type : POST
4977|[2023-06-02 22:42:08][DEBUG] : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":1794,"cluster_type":"in","attributes":[32],"allowCache":0}
4978|[2023-06-02 22:42:13][INFO] : [Maison][LiXee.ZLinky_TIC 00:15:8d:00:05:d2:a9:0a] Erreur lors de la requete : http://127.0.0.1:8089/device/attributes(POST), data : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":65382,"cluster_type":"in","attributes":[1,2,4,5,6,7,8],"allowCache":0} erreur : {"state":"error","result":"[0x54d8:1:0xff66]: Message send failure","code":0}
4979|[2023-06-02 22:42:17][INFO] : [Maison][LiXee.ZLinky_TIC 00:15:8d:00:05:d2:a9:0a] Erreur lors de la requete : http://127.0.0.1:8089/device/attributes(POST), data : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":1794,"cluster_type":"in","attributes":[32],"allowCache":0} erreur : {"state":"error","result":"[0x54d8:1:0x0702]: Message send failure","code":0}
4980|[2023-06-02 22:42:17][DEBUG] : http://127.0.0.1:8089/device/attributes type : POST
4981|[2023-06-02 22:42:17][DEBUG] : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":2820,"cluster_type":"in","attributes":[1288,2312,2568,1290,2314,2570,1293],"allowCache":0}
4982|[2023-06-02 22:42:22][INFO] : [Maison][LiXee.ZLinky_TIC 00:15:8d:00:05:d2:a9:0a] Erreur lors de la requete : http://127.0.0.1:8089/device/attributes(POST), data : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":2820,"cluster_type":"in","attributes":[1288,2312,2568,1290,2314,2570,1293],"allowCache":0} erreur : {"state":"error","result":"[0x54d8:1:0x0b04]: Message send failure","code":0}
4983|[2023-06-02 22:42:22][DEBUG] : http://127.0.0.1:8089/device/attributes type : POST
4984|[2023-06-02 22:42:22][DEBUG] : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":2820,"cluster_type":"in","attributes":[1295],"allowCache":0}
[... cut ...] : {"state":"error","result":"[0x54d8:1:0x0702]: Message send failure","code":0}
4997|[2023-06-02 22:45:17][DEBUG] : http://127.0.0.1:8089/device/attributes type : POST
4998|[2023-06-02 22:45:17][DEBUG] : {"ieee":"00:15:8d:00:05:d2:a9:0a","endpoint":1,"cluster":2820,"cluster_type":"in","attributes":[1288,2312,2568,1290,2314,2570,1293],"allowCache":0}
4999|[2023-06-02 22:45:18][ERROR] : Le module [Maison][LiXee.ZLinky_TIC 00:15:8d:00:05:d2:a9:0a] n\'a pas envoyé de message depuis plus de 1440 min`

fairecasoimeme, 2023-06-05:
Bonjour,

Si le ZLinky n'est pas accessible, c'est soit qu'il n'est pas appairé, soit qu'il y a des problèmes de portée dû à ma distance ou interférences

Iwhite67, 2023-06-05:
Bonjour, justement, il est accessible car il répond, j'ai juste pas les remontées

fairecasoimeme, 2023-06-05:
Sauf erreur, d'après les logs, il n'a pas l'air de répondre

Coldness00, 2024-04-14:
Bonjour,
Je @mattzeer il semblerait que j'ai un soucis similaire.
Suite a un changement de fournissseur j'ai quelques de remontees de valeurs (PMR, heure, Traf, message, etc), sauf ma consommation d'energie.
As-tu trouve une solution?
Merci

MilesTEG1, 2024-04-15:
Bonjour,
Une potentielle piste serait de vérifier les paramètres dans Z2M, qu'ils correspondent bien aux caractéristiques de l'abonnement : standard ou historique , tarif tempo ou autre , etc.
Dès qu'un paramètre est modifié , il faut redémarrer Z2M. Et être patient pour que les données remontent.
À mon passage de historique à standard , j'ai cru que j'avais un module défaillant, mais j'ai enfin eu les remontées quelques heures après le redémarrage de Z2M.

Coldness00, 2024-04-28:
@mattzeer 
Regarde sur ce post. J ai réussi a solutionner mon soucis.
https://github.com/fairecasoimeme/Zlinky_TIC/issues/256