fairecasoimeme/Zlinky_TIC #176: Abonnement HCHP non reconnu

laguyle, 2023-02-08:
Hello,

Je viens de recevoir mon Zlinky il y a quelques jours (livré directement avec le firmware en v12).
Aucun souci pour l'appairage et le setup, les données de consommation remontent à intervalle régulier.

Par contre j'ai un petit souci : je suis sur un abonnement HC/HP mais visiblement ce n'est pas reconnu car le cluster 0xFF66 / attribut 0x0002 me remonte la valeur BASE : 
[image]

La conséquence est donc que je n'ai que le compteur HC qui s'incrémente (même en période HP).
Le cluster 0x0702 / attribute 0x0102 ne donne aucune valeur : 
[image]

Je précise que : 
- j'ai vérifié chez Engie et je suis vraiment bien en abonnement HC/HP ;-)
- j'ai en parallèle l'intégration myEnedis qui me remonte bien les infos HC/HP
- mon compteur est en mode historique
[image]

Une idée de pourquoi mon abonnement n'est pas reconnu ?

Merci.

fairecasoimeme, 2023-02-08:
Bonjour,

Oui vous n'avez surement pas un abonnement historique EDF (HC/HP) avec une téléinformation historique. Chez les autres opérateurs, il y a des abonnements de type super heure creuse ou heure creuse week end etc ...
Pour régler le problème il faut passer en mode standard pour bénéficier de tous les index.
Qu'y a-t-il marqué sur l'écran du Linky quand vous êtes sur le type d'abonnement ?

laguyle, 2023-02-19:
Bonjour,

Bon et bien je n'ai pas un abonnement HC/HP classique car je suis en tarif HC également le week-end.
Après avoir changé le mode de mon compteur en standard, j'ai bien toutes les données qui remontent.
Désolé j'aurai du commencer par là.

Merci :-)