fairecasoimeme/Zlinky_TIC #177: Problème remontée valeur nulle BBRHPJR - Option Tempo

antoineguilbert, 2023-02-08:
Bonjour,

Je suis passé en début de semaine de l'abonnement HP/HC de Total Energies à l'abonnement Tempo chez EDF. 
J'ai bien changé la configuration sur les paramétrage de Zlinky sur Zigbee2MQTT mais je reste toujours avec des "sans valeur" sur mes nouveaux index à savoir par exemple : BBRHPJR

En revanche, cela fonctionne bien pour BBRHCJB ou encore BBRHPJB. J'ai bien également le PTEC qui remonte avec "HPJR". 

[image]

Voici ce que cela donne au niveau de mon relevé Linky. Note que jusqu'à aujourd'hui, ma valeur de BBRHPJR était à 0kWh commençant mon abonnement mais je viens de passer à 1kWh je pense donc que cela aurait du se mettre à jour ? 

{
    "active_register_tier_delivered": "HPJR",
    "apparent_power": 400,
    "available_power": 30,
    "current_summ_delivered": 0,
    "current_tarif": "BBR(",
    "current_tier1_summ_delivered": 12175.8,
    "current_tier2_summ_delivered": 492.1,
    "linkquality": 255,
    "meter_serial_number": "XXXXXXXX",
    "rms_current": 2,
    "rms_current_max": 90,
    "schedule_h_p_h_c": 0,
    "update": {
        "installed_version": 12,
        "latest_version": 12,
        "state": "idle"
    },
    "update_available": false,
    "warn_d_p_s": 0,
    "current_tier3_summ_delivered": null,
    "current_tier4_summ_delivered": null,
    "current_tier5_summ_delivered": null,
    "current_tier6_summ_delivered": null,
    "start_notice_e_j_p": null,
    "tomorrow_color": null
}

Et la version de mon firmware : 4000-0012

Merci par avance 👍

pipiche38, 2023-02-08:
peut-etre faut-il attendre de passer en jour Blanc et Rouge pour que ca s'initialise

antoineguilbert, 2023-02-08:
Sachant que pour récap : 

- Jour bleu : ça a bien fonctionné 
- Jour blanc : pas encore eu
- Jour rouge : KO pour le moment

cmer81, 2023-02-10:
Bonjour @antoineguilbert,

J'ai eu le même problème aujourd'hui (Avec le tarif BBRHCJB et BBRHPJB ) suite à mon changement d'offre de BASE a Tempo effectuer hier. 
Je voulais également partager avec vous que j'ai réussi à fixer le problème en supprimant complètement le ZLinky device de Zigbee2mqtt et également les ressources dans Home Assistant. Je sais que cela peut ne pas être la solution pour tout le monde, mais cela a fonctionné pour moi.

Maintenant, j'attends de voir si le problème sera le même lors de passage en jour blanc ou rouge. Je pense que la cause pourrait peut-être être due à un index de consommation à zéro. 🤔

[image]

J'espère que la solution sera trouvée rapidement.

antoineguilbert, 2023-02-11:
Je confirme la résolution de @cmer81, j'ai pour ma part redémarré plusieurs fois Zigbee2MQTT et il a fini par récupérer les nouvelles valeurs. C'est ok !

Merci

closed 2023-02-11 by antoineguilbert