fairecasoimeme/Zlinky_TIC #179: Remontées figéees depuis 3 jours

Patoche781, 2023-02-08:
Bonjour.
Ma configuration :
HA 2023.2.3 + Sup 2023.01.1 +OS 9.5 + UI 20230202 + Z2M sous Proxmox sur Celeron 8Go de Ram + SSD 500Go
Clé USB Zigbee SonOff
Lixee V4000-06
Autres périphériques Zigbee: 15
Distance HA Z2M -> Lixee : 5m avec Lqi=30
Contrat EDF:
Monophasé avec abonnement depuis 2mois: HP + HC midi + soir + WE (Vert electrique Auto)

Depuis le 06/02/2023 plus de remontées de Lixee (tout fonctionnait bien depuis Aout 2022).
Les symptomes: les valeurs restent figées. Lixee ne veut pas prendre la mise à jour en OTA.
j'ai redémarré Z2M et Mosquito Broker plusieurs fois
Avez-vous le même problème sinon comment le résoudre ?
Voici mes paramètres
[image] 
[image]
[image]

Merci pour votre retour.
Cdt.

fairecasoimeme, 2023-02-09:
Bonjour,

Pour l'erreur "MAC no ack", il y a eu un issue ici : https://github.com/Koenkk/zigbee2mqtt/issues/2789
Est-ce que ça ne correspondrait pas ?
Dans tous les cas, je vous recommande aussi quand ce sera possible de mettre à jour en v12

Patoche781, 2023-02-09:
Bonjour et merci pour le suivi.
Pour que le Lixee puisse être vu du coordinateur, dois-je faire la procédure suivante: supprimer l'objet Lixee de Z2M, le débrancher du Linky, le rebrancher puis tenter le réappairage ? Car n'étant plus vu de Z2M je ne peux pas faire la mise à jour OTA : c'est la solution ?

fairecasoimeme, 2023-02-09:
D'après l'erreur et de ce que je comprend, il semblerait qu'il y ait des problèmes d'interférences. Vous pouvez tenter de refaire un appairage pour voir s'il le coordinateur le re détecte. peut-être changer de canal zigbee aussi.

Patoche781, 2023-02-09:
Merci pour la rapidité du message!
Je vais tenter ce soir d'abord le refaire le réapparaierage ... Sinon: comment changer le canal zigbee ? Et question très importante: si je change le canal du zigbee vais-je devoir refaire la recoordination de tout mes autres périphériques zigbee ou vont-ils basculer d'eux même sur le nouveau canal ?

Patoche781, 2023-02-09:
Je ma réponds à moi-même pour la question du changement de canal du zigbee: la notice officielle stipule " ZHA préfère utiliser le canal Zigbee 15 par défaut. Vous pouvez modifier cela en utilisant la configuration YAML, mais cela ne fonctionne que s'il n'y a pas de réseau existant. Pour changer le canal d'un réseau existant, la radio doit être réinitialisée et un nouveau réseau doit être formé. Cela nécessite un réappariement de tous les appareils."
C'est ce que je craignais ! On va donc essayer le mode réappairage ...

pipiche38, 2023-02-10:
Bonjour, vous avez le meme problème que les utilisateurs d'un coordinateur non-Zigate. Il vous faut faire un reset du ZLinky;

Patoche781, 2023-02-10:
@pipeche38 Merci pour ton retour!
Je ne suis pas resté inactif hiers soir et essayé et réessayé et re-réessayé mais: choux-blanc! J'ai supprimé de Z2M l'objet "Lixee" puis relancé Z2M avec à la suite relance de Z2M + MQTT+ Redémarage de Poxmox+ HA. Retentative de réappairage de Lixee : appui long pour reset: le lixee s'éteint puis passe au bleu fixe (Linky Ok donc) mais invisible dans Z2M lors de la phase de reconnaissance déclenchée (avant ou après).
Le module Lixee a fonctionné de nombreux mois (car de nombreuses galères sur le github sur les versions qui ont succédées) mais la impossible de l'appairer lors de ces derniers avatars ! Je retente le coup ce soir après une nuit de dépose et de repos de Lixee et de moi même! 
Pour info: tous mes autres périphs fonctionnent à merveille à des distances de plus de 15 m pour certains!  Le Lixee est à 3 m du coordinateur et n'est pas fichu de verrouiller la trame Zigbee !
Je sens que je vais renvoyer ce matos au vendeur et me le faire changer ou rembourser !
Ce matériel n'est pas fiable en tant que tel !
Ps: quand tu dis un reset du Zlinky tu veux dire quoi exactement ? @faicasoieme me dit de changer le canal Zigbee: hors de question de tout casser: j'ai pas envie de monter dans mes combles pour réinitialiser mon detecteur incendie Frient ou mes capteurs T° SonOff et inondations Xiaomi !
Dans l'attente de te lire ...

max5962, 2023-02-13:
Depuis hier.
plus aucune nouvelle du Zlinky ( pourtant je suis sous ZHA / home assistant ) et ça marche depuis 1 ans 1/2 sans broncher ...
Question très très très con, l'heure de déconnection, semble correspondre au moment ou j'ai changé la poignée de porte de mon placard qui contient le Linky ... J'espère que c'est pas semble autant à ce genre d'aléa ^^

Patoche781, 2023-02-13:
Bonjour. Aucune modification de prêt ou de loin sur mon installation !
J'essaie de joindre le SAV de Domadoo qui m'a vendu le produit et qui
manifestement prend son temps pour répondre ma demande. LE N° de téléphone
ne répond pas... Ils m'ont préconisés par mail de faire des mesures de
tension sur le Lixee comme tout préambule. Bon je patiente sur leur réponse
suite à mes mesures transmises après je vais escalader plus haut...

fairecasoimeme, 2023-02-13:
Bonjour,

Vous indiquez que le ZLinky fonctionne correctement depuis aout 2022, vous avez un LQI de 30 (faible). Dans les logs vous avez un message d'erreur "MAC no ack" qui signifie que votre coordinateur n'a pas reçu d'acquittement a une demande d'information.
La led du ZLinky est-elle allumée ?
avez-vous débranché puis rebranché le ZLinky ? (j'imagine que oui)
Du coup, pour moi, c'est un problème de réception du ZLinky.

fairecasoimeme, 2023-02-13:
> Depuis hier. plus aucune nouvelle du Zlinky ( pourtant je suis sous ZHA / home assistant ) et ça marche depuis 1 ans 1/2 sans broncher ... Question très très très con, l'heure de déconnection, [...]

Quelle version du firmware ? avez-vous le condensateur ? https://github.com/fairecasoimeme/Zlinky_TIC#errata-hardware
si vous n'avez pas la dernière version du ZLinky, il est possible qu'il soit planté. (essayez de débrancher/rebrancher puis mettre à jour). Sinon autre possibilité, vous êtes en limite de portée, le fait d'être resté devant la porte à fait changer le réseau de route pour le ZLinky et a mis la pagaille dans le réseau qui fait que maintenant le ZLinky n'est plus à portée du coordinateur.

max5962, 2023-02-13:
> Quelle version du firmware ? avez-vous le condensateur ? https://github.com/fairecasoimeme/Zlinky_TIC#errata-hardware si vous n'avez pas la dernière version du ZLinky, il est possible qu'il soit [...]

Version V5, je vais upgrade à la V12 (je pensais que c'était automatique).

J'ai la version sans le condensateur (acheté le jour la sortie 😁) 

Un restart et c'est de nouveau opérationnel !

pipiche38, 2023-02-14:
le problème est equivalent au problème des Aqara qui disparaissent du réseau.
Le ZLinky quand il n'a pas de retour du coordinateur semble s'endormir et ne se reveille plus (jusqu’à un hard reset de celui-ci)

Quand vous utilisez une clef Texas Instrument avec Z2M ou une clef Silabs avec ZHA celle-ci ne remplie pas son role à 100% quand la partie Z2M ou ZHA est down (update, reboot ....) Si alors lors de cette fenêtre le ZLinky envoie un message de type Report ou autres et que le coordinateur de réponde pas (par exemple pas de retour de Default Response), alors le ZLinky semble se mettre en sommeil.

Peut-être ais-je tord, mais il me semble que les Aqara/Xiaomi qui disparaissent du réseau sont aussi sur un chip set NXP et la même stack que celle du ZLinky

J'ai reproduit ce problème à plusieurs reprise avec une clef Silabs. Ce problème n'arrive pas avec une ZIgate tant que celle-ci reste alimentée car elle gère ZCL et fait les Default Response (contrairement aux autres clefs qui travaille en Raw)

fairecasoimeme, 2023-02-14:
Désolé mais ce n'est pas le même problème que les aqara dans ce cas précis. Les aqara sont des ZED qui décident de sortir du réseau lorsqu'il ne communiquent pas pendant X temps. Le ZLinky est en mode routeur et il n'y a pas d'endormissement. les aqara et ZLinky utilisent la même marque NXP mais pas le même micro et pas la même stack. Le problème que rencontre @max5962 est connu et a été testé. Il est lié à un problème hardware et software. L'absence de condensateur implique un hardfault côté firmware. Pour régler le problème, il faut rajouter le condensateur et mettre à jour le firmware.

Ensuite sur les problèmes de mise à jour, il y a plusieurs choses. mais impossible de conclure sur la ZiGate puisque la mise à jour n'est pas possible au travers de Z2M ou ZHA. De mon côté et au travers de mes tests sur Z2M, j'ai remarqué un affolement et un flood de Z2M sur le ZLinky. Quand je fais les tests avec une ZiGate et un outil basique (avec bien entendu les binds et reports configurés), je n'arrive pas à reproduire les problèmes rencontrés.

pipiche38, 2023-02-14:
Comme tu veux, mais moi je confirme ce que j'ai observé, et que le ZLinky s'endort complétement. et que ce phénomène n'existe pas en Zigate mais avec les clefs non-zigate.
L'idée etait d'apportée une piste d'investigation
Oublie !

pipiche38, 2023-02-14:
qu'il plante ou qu'il s'endorme y-a pas beaucoup de différence pour moi. En tout les cas la root cause pour moi est qu'il est en attente de quelque chose du coordinateur (probablement le default response status) .
Dans mon cas le reset 10s est suffisant, par contre certain ont eu besoin du re-flash

fairecasoimeme, 2023-02-14:
s'endormir n'a rien à voir avec planter. Surtout quand le but est de trouver une solution
Dans le cas de @max5962, le problème est connu et a été testé et corrigé.
Dans ton cas, je crois savoir que tu as encore un prototype donc très probablement le même cas que @max5962 
Pour le cas de @Patoche781, pour moi, il y a clairement un problème de portée, de firmware ou plantage. ça a fonctionné 8 mois ...

pipiche38, 2023-02-14:
comme tu veux . bonne continuation ...

fairecasoimeme, 2023-02-14:
de même

Patoche781, 2023-02-14:
Bonjour à tous.
Ma clé Zigbee est la SonOff USB 3.0 qui fonctionne parfaitement en
production (HA aux dernières M.A.J) avec une grande diversité de matériels
de plusieurs marques: SonOff (T°) , Frient (Incendie), Xiomi (humidité),
Lidl (lumière et détection de présence), Noos (prises électriques) entre
autres. Tous ont été facilement intégrés au réseau zigbee en production dès
leurs achat. A côté de cela j'ai le même matériel et sa configuration à
l'identique en plateforme de test qui fonctionne parfaitement bien mais
libre de tout périphérique (juste utilisée pour tester un nouveau matériel,
les M.A.J, les scripts, etc...). Cette plateforme de test est située à 1,5m
du Linky. La production est elle même à 3-5m soit dit en passant.
Le Lixee (avec condensateur) à fonctionné de longs mois avec tout ce petit
monde (primo appairage capricieux mais ok tout de même)) ! Du jour au
lendemain des pertes de données sont apparues puis silence total (aucun log
de tentative de connexion,)... Les resets + débranchements + rebranchements
les mises au repos n'y ont rien fait. La tentative d'appairage sur ma plate
forme de test distante d'un mètre n'a pas mieux fait alors qu'elle ne
possède que le concentrateur zigbee opérationnel (donc pas de perturbation
possible)!
Conclusion ou le matériel est grillé ou particulièrement instable (mais pas
les autres)  !
Je ne vois pas d'autres solutions et vous ?
Cordialement d'un passionné.

fairecasoimeme, 2023-02-14:
il faut faire attention car avec le ZLinky, dans votre version, quand on appuie 10 sec sur le bouton, cela peut déclencher sur les 2 coordinateurs une mise en mode "permit join" et du coup le zlinky peut s'appairer sur le mauvais coordinateur. C'est déjà arrivé chez certains.
Peut-être débrancher le coordinateur que vous ne voulez pas utiliser pour le ZLinky pour être sur et supprimer avec forçage les ZLinky "fantômes"

Patoche781, 2023-02-14:
Merci pour votre suivi et la recherche de solutions.
@fairecasoimeme  Je souligne que ma plate-forme de test ne fonctionne pas en permanence pour éviter tout interférence.
Par contre je vais éteindre mon HA en production, allumer celui de test uniquement et essayer en fin d'A.M de refaire un appairage du Lixee à 1m ! 
Question: Il faut appuyer combien de temps pour faire le reset ? Y'a t-il un ordre à respecter pour lancer un "Permit join" . Le faire avant ou après le reset? Débrancher le coordinateur non utilisé ? C à dire ? Supprimer avec forçage des ghosts: et si dans mon interface web Z2M je n'ai aucun périphérique c'est qu'il n'y en as pas ? 
Merci de vous lire !

fairecasoimeme, 2023-02-14:
il faut rester appuyer 10 sec.
pas forcément d'ordre. vous pouvez lancer le permit join après le reset.
Si vous éteignez votre HA de production pas besoin de débrancher le coordinateur
Si vous n'avez aucun ZLinky dans l'interface web c'est ok

fairecasoimeme, 2023-02-14:
En ce moment, il y a des mises à jours de Linky qui semble mettre en sécurité l'alimentation du TIC. pour résoudre le problème, il faut laisser débrancher le ZLinky pdt 10 min puis rebrancher.

Patoche781, 2023-02-22:
Bonjour
Problème résolu: le TIC douteux à été renvoyé au vendeur (assez réactif sur le coup) qui m'en a envoyé un nouveau. Dès sa mise en place dans le Linky l'appairage s'est fait immédiatement du premier coup et dans la foulée la dernière MAJ en OTA V13 s'est lancée en 20mn sans défaut. J'ai un LQI de 100 environ. Les datas du Linky sont remontées dans le pannel "Energy" au bout d'une heure (en négatif d'abord ?) puis ce matin tout est OK. Petit problème j'ai perdu l'historique du Linky mais pas grave ! L'appli de mon fournisseur est là ...

closed 2023-02-22 by fairecasoimeme

tomav, 2023-07-13:
Voici mon expérience sur cette erreur.
J'ai eu le même souci aujourd'hui avec un Zlinky récent (acheté en 2023, upgradé tout de suite en version 13, et qui a bien fonctionné pendant des mois):
- aucun changement dans mon installation Z2M
- erreurs `MAC no ack' (233)` sur mon Zlinky seulement. Tout le reste fonctionne bien
- unplug/replug du Zlinky et retour à la normale