fairecasoimeme/Zlinky_TIC #181: Suggestion update : RESET quand plus de syncho avec le linky

nicocha75, 2023-02-12:
Bonjour
Mon linky se trouve en extérieur et le zigbee passe mal, mais ça passe. Je remarque que de temps à autre, lors du changement du relais en mode jour/nuit une perte du zlinky. Je pense qu'il se désynchronise des trames et ne se retrouve plus.
Serait il possible d'avoir un reset automatique du zlinky lorsque par exemple 100 trames sont reçu erroné de suite?
Pour ma part, c'est galère d'aller au compteur pour le reseter.

Merci
Nicolas

navycrow, 2023-02-20:
je suis dans la même situation et c'est vrai que ce serai top ! +1

cantece, 2023-09-14:
comment fait-on un reset hardware sur le Zlinky-TIC?

jacme31, 2023-11-15:
Même problème. Compteur extérieur distant et perte définitive du linky toutes les deux semaines environ. Pour moi ce n'est pas lié au changements de relais jour/nuit  (je suis au tarif base sans HC) . Origine donc indéterminé.
Le reset pour moi passe par extraction du lixee puis remise en place  dans le compteur. 
Je suis en firmware V13 (Micrologiciel : 0x0000000d).