fairecasoimeme/Zlinky_TIC #183: MQTT or Zigbee2MQTT multiple (too many?) log entries

jellium, 2023-02-19:
Firmware used: router 13
Messages in MQTT are numerous: a first message is published, followed by `zlinky config:` line, immediately followed by 9 messages (all 10 messages look identical).
```
Feb 19 09:50:31 ZIGBEE2MQTT npm[22904]: Zigbee2MQTT:info  2023-02-19 09:50:31: MQTT publish: topic 'zigbee2mqtt/linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":1640,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":4798.157,"current_tier2_summ_delivered":4879.599,"linkquality":220,"meter_serial_number":"XXX","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null,"warn_d_p_s":0}'
Feb 19 09:50:38 ZIGBEE2MQTT npm[22904]: zlinky config: historique, single_phase, false, HC..
Feb 19 09:50:39 ZIGBEE2MQTT npm[22904]: Zigbee2MQTT:info  2023-02-19 09:50:39: MQTT publish: topic 'zigbee2mqtt/linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":1640,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":4798.157,"current_tier2_summ_delivered":4879.599,"linkquality":220,"meter_serial_number":"XXX","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null,"warn_d_p_s":0}'
Feb 19 09:50:39 ZIGBEE2MQTT npm[22904]: Zigbee2MQTT:info  2023-02-19 09:50:39: MQTT publish: topic 'zigbee2mqtt/linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":1640,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":4798.157,"current_tier2_summ_delivered":4879.599,"linkquality":220,"meter_serial_number":"XXX","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null,"warn_d_p_s":0}'
Feb 19 09:50:39 ZIGBEE2MQTT npm[22904]: Zigbee2MQTT:info  2023-02-19 09:50:39: MQTT publish: topic 'zigbee2mqtt/linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":1640,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":4798.157,"current_tier2_summ_delivered":4879.599,"linkquality":216,"meter_serial_number":"XXX","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null,"warn_d_p_s":0}'
Feb 19 09:50:39 ZIGBEE2MQTT npm[22904]: Zigbee2MQTT:info  2023-02-19 09:50:39: MQTT publish: topic 'zigbee2mqtt/linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":1640,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":4798.157,"current_tier2_summ_delivered":4879.599,"linkquality":216,"meter_serial_number":"XXX","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null,"warn_d_p_s":0}'
Feb 19 09:50:39 ZIGBEE2MQTT npm[22904]: Zigbee2MQTT:info  2023-02-19 09:50:39: MQTT publish: topic 'zigbee2mqtt/linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":1640,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":4798.157,"current_tier2_summ_delivered":4879.599,"linkquality":220,"meter_serial_number":"XXX","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null,"warn_d_p_s":0}'
Feb 19 09:50:39 ZIGBEE2MQTT npm[22904]: Zigbee2MQTT:info  2023-02-19 09:50:39: MQTT publish: topic 'zigbee2mqtt/linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":1640,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":4798.157,"current_tier2_summ_delivered":4879.62,"linkquality":224,"meter_serial_number":"XXX","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null,"warn_d_p_s":0}'
Feb 19 09:50:39 ZIGBEE2MQTT npm[22904]: Zigbee2MQTT:info  2023-02-19 09:50:39: MQTT publish: topic 'zigbee2mqtt/linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":1640,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":4798.157,"current_tier2_summ_delivered":4879.62,"linkquality":212,"meter_serial_number":"XXX","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null,"warn_d_p_s":0}'
Feb 19 09:50:39 ZIGBEE2MQTT npm[22904]: Zigbee2MQTT:info  2023-02-19 09:50:39: MQTT publish: topic 'zigbee2mqtt/linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":1640,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":4798.157,"current_tier2_summ_delivered":4879.62,"linkquality":232,"meter_serial_number":"XXX","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null,"warn_d_p_s":0}'
Feb 19 09:50:39 ZIGBEE2MQTT npm[22904]: Zigbee2MQTT:info  2023-02-19 09:50:39: MQTT publish: topic 'zigbee2mqtt/linky', payload '{"active_register_tier_delivered":"HP..","apparent_power":800,"available_power":45,"current_tarif":"HC..","current_tier1_summ_delivered":4798.157,"current_tier2_summ_delivered":4879.62,"linkquality":232,"meter_serial_number":"XXX","rms_current":3,"rms_current_max":90,"schedule_h_p_h_c":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null,"warn_d_p_s":0}'
```

bosr, 2023-02-19:
Salut, j'ai la même situation en v12.

À court terme, dans la page Settings (générique, pas settings(specific)) de cet appareil, j'ai mis le paramètre `debounce` à `30`. Les messages sont toujours envoyés sur le réseau zigbee, mais au moins mqtt et surtout la base de données HA prennent moins cher.

Dans mon cas, j'ai pas mal de petites variations de `linkquality`.

Je suis débutant dans le domaine, alors je suis preneur d'une solution propre à ce souci.

PS : I can translate that to english if it helps.

jellium, 2023-02-19:
Hi @bosr, thanks for your input.
I have too set debounce to 30 (seconds) with default poll_intervalle set to 60 seconds (even though there seem to be 2 timers ongoing, one with 1 message broadcasted and one with several (8) messages broadcasted on the Zigbee network).
Also, is it normal that `zlinky config: historique, single_phase, false, HC..` occurs continuously every minute or so? And how is that line related to `measurement_poll_interval`? Finally, changing that last parameter `measurement_poll_interval` doesnt change the MQTT messages broadcast frequency (even if Zigbee2MQTT is restarted after changing that parameter).