fairecasoimeme/Zlinky_TIC #184: Z2M mise a jour par OTA firmware Limited.

WarC0zes, 2023-02-20:
Bonjour,
je suis passer sous Z2M et je n'est jamais mis a jour sous Z2M mon Zlinky en firmware Limited.
Je suis en V12 Limited et voudrais mettre a jour en V13 Limited, mais es que Z2M va récupérer le firmware limited si je le met a jour a partir de Z2M ?
Ou va falloir que je le fasse manuellement ?
Si manuellement, quelqu'un peu expliquer la procédure.
Merci d'avance.

molusk, 2023-02-25:
Je suis dans le même cas que toi et je n’ai pas encore réussi a faire passer la maj.
Z2M va chercher la version complète même quand je lui force le json d’OTA en local avec le firmware à côté.
Je dois faire quelque chose mal mais je ne trouve pas quoi...

On en parle déjà sur le forum hacfr, je ne sais pas quel est le meilleur endroit pour poursuivre cette discussion et partager le résultat de nos tests.

fred-gb, 2023-02-26:
Bonjour,
Je viens de voir ce sujet juste quelques minutes après après posté sur le github de Z2M.
https://github.com/Koenkk/zigbee2mqtt/issues/16823

En espérant une réponse.

Bon après-midi

molusk, 2023-02-27:
Merci pour la piste ! Je vais me laisser tenter par un custom converter :+1: 
Si ça se passe bien, je le partagerai ici même.

molusk, 2023-02-27:
Bon, j'ai tenté une solution TRÈS sale, modifier le fichier js qui définit comment faire des OTA pour le Zlinky pour pointer vers le fw limited :
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/lib/ota/lixee.js
changement de la ligne 21 pour trouver le fichier qui finit par limited.ota au lieu de .ota.

Ça m'avait l'air pas mal en théorie mais en pratique, ma modification n'est pas prise en compte alors qu'elle persiste après redémarrage du conteneur... Peut-être que je n'ai pas modifié la bonne copie du fichier... je ferai plus de tests dans les jours à venir.

molusk, 2023-02-28:
Une solution plus propre a été donnée sur le post de @fred-gb https://github.com/Koenkk/zigbee2mqtt/issues/16823
Je testerai dès que possible :)

En fait, le converter lixee utilise la fonction d'ota spécifique lixee (celle que j'ai tenté de modifier sans succès) où tout est en dur quand la fonction standard zigbeeOTA.js interprète correctement l'option pour écraser le chemin de l'index d'ota.

Bref, en forçant le converter y pointer sur zigbeeOTA, le chemin de l'index pourra être modifié pour pointer vers le firmware limited et la mise à jour poussée.
Par contre, c'est du temporaire, très temporaire, aucune autre mise à jour ne pourra passer avec cette configuration et une fois l'index custom retiré, il faudra repasser sur le converter standard de lixee sans quoi tu n'auras ni mise à jour, ni notification de disponibilité de la mise à jour pour le zlinky.

Au final, c'est le fichier ota/lixee.js qu'il faudra modifier pour gérer les version limited et/ou les index custom.
Je connais rien à JS mais je vais peut-être tenter de bidouiller lors de ma prochaine soirée d'ennui :rofl:

molusk, 2023-02-28:
Mise à jour en cours avec la bidouille indiquée sur l’autre post 👍 

Pour ce qui est de modifier le fichier OTA lixee, il va falloir attendre le prochain firmware pour le test/débogage mais je vais me pencher dessus en attendant (à moins qu’on puisse forcer un rollback sans risque ?)

WarC0zes, 2023-03-08:
Bonjour,
personne a de solution ? 
étonnant qu'on soit que quelques utilisateurs a utiliser un firmware limited sous Z2M. 
C'est mon seul regret d'être passer sous Z2M, pour mettre a jour le Zlinky. C'était bien plus simple sous ZHA...

PEGARifrouf, 2023-03-13:
Salut,
J'avais réussis downgrader vers un firmware limited grâce à [ce post en lien ici](https://github.com/fairecasoimeme/Zlinky_TIC/issues/89#issuecomment-1236422701)

J'étais ensuite passé au v12 limited avec la même procédure.

MilesTEG1, 2024-03-18:
Hello,
J'ai le même soucis que vous tous.
J'ai réussi à flasher le firwmare v13 limited ce WE (après pas mal de galères), et Z2M me dit qu'il n'est pas à jour et qu'il y a un firwmare v13 à flasher.
Sauf que ça ne fonctionnera pas, comme l'a souligné @WarC0zes puisque Z2M veut flasher un firmware legacy, et mon ZLinky est Limited.

WarC0zes, 2024-09-02:
Bonjour,
je relance le post, aucune avancé pour mettre à jour un firmware limited sous Z2M ?
En plus un V14 vient de sortir, ça serait cool de pouvoir l'installer.

MilesTEG1, 2024-09-02:
> Bonjour, je relance le post, aucune avancée pour mettre à jour un firmware limited sous Z2M ? En plus une V14 vient de sortir, ça serait cool de pouvoir l'installer.

Je suis moi aussi preneur d'une solution de MAJ OTA de la version limited.
Ça doit bien être possible, vu qu'il y a un .ota... mais je ne connais pas la méthode.

J'envisage presque de reflasher une version router...

dinde, 2024-09-15:
> Je suis moi aussi preneur d'une solution de MAJ OTA de la version limited. Ça doit bien être possible, vu qu'il y a un .ota... mais je ne connais pas la méthode.
>
> J'envisage presque de reflasher une [...]

Tout pareil. J'ai sans arrêt des propositions de maj OTA mais rien à faire pour mon zlinky en firmware limited.
La maj USB n'est clairement pas envisageable en production et cela fait assez longtemps que nous sommes dans l'attente d'une solution. 🙏🏻

molusk, 2024-09-19:
Honnêtement, je pense que je suis trop incompétent pour réussir à m'en sortir sur ce sujet... Si un gars calé en JS passe dans le coin, pleeeeeeeeeeeeeeeaaaaaaaaaaaaaase help !

WarC0zes, 2024-10-15:
@molusk , ça avance :D

https://github.com/Koenkk/zigbee-herdsman-converters/pull/8136

https://github.com/Koenkk/zigbee-OTA/pull/568

WarC0zes, 2024-10-16:
Les PR sont merge, plus cas attendre la prochaine release de Z2M, début novembre et on pourra mettre à jour en OTA le firmware limited .

closed 2024-10-16 by WarC0zes

WarC0zes, 2024-11-01:
Bonjour,
la nouvelle maj de Z2M est sorti 👍 
MAJ par OTA pour le limited, est bon. Plus cas testé.

WarC0zes, 2024-11-03:
Bonjour,
mise a jour v14 limited sous Z2M OK.
[image]

molusk, 2024-12-03:
J'ai l'impression que mon compteur délivre pas assez de puissance au zlinky et qu'il échoue au milieu de l'OTA mais au moins, il n'est pas briqué !

WarC0zes, 2024-12-03:
> J'ai l'impression que mon compteur délivre pas assez de puissance au zlinky et qu'il échoue au milieu de l'OTA mais au moins, il n'est pas briqué !

Salut,
tu peut utiliser une alimentation externe , au lieu du linky.

https://github.com/fairecasoimeme/Zlinky_TIC/issues/290#issuecomment-2415070713

molusk, 2025-01-02:
> 
> Salut, tu peut utiliser une alimentation externe , au lieu du linky.
> 
> [#290 (comment)](https://github.com/fairecasoimeme/Zlinky_TIC/issues/290#issuecomment-2415070713)

Merci pour l'info mais le compteur est au bout du jardin, ça va être compliqué pour tirer une alim... autant tirer des fils pour la téléinfo et mettre le zlinky dans mon garage avec l'alim externe.
Bon, je viens de voir que suite à la dernière tentative d'OTA, il est offline... Il va falloir que j'aille essayer d'ouvrir le coffret à la serrure pétée pour le débrancher et rebrancher, ça m'enchante !
Ça ne serait peut-être pas une si mauvaise idée de tirer des câbles :smiley_cat: