fairecasoimeme/Zlinky_TIC #185: Acquisition Time and Power measurement

Karreg, 2023-02-20:
Hello,

I have set up my ZLinky successfully, but there's still a value I'm having an issue with : Power (in Watt).
The value is updated every 30 minutes, even though Acquisition Time is set to 10 seconds, and I have changed it to 7 seconds.
I don't have this issue with other measures.
Does it mean the value is updated by Linky only every 30 minutes?
If yes, is there a way to change that?
If no, is there any way to accurately measure instance power delivered?
My purpose is to instantely measure the gain, or added consumption of devices.

Thanks!

CladStriffe, 2023-03-02:
Bonjour
Ce que tu évoques ne dépend pas du zlinky, mais bien du compteur Linky directement qui ne donne qu'une valeur moyenne sur une période définie par le fournisseur d'énergie (de 10 à 30 minutes de ce que j'ai pu voir). 
Regarde ici 
#117

Karreg, 2023-03-03:
Merci pour le lien @CladStriffe ! Je n'arrivais pas à trouver l'information.
Les échanges autour des W et VA est intéressantes, et la connaissant j'ai pu la retrouver sur internet.
Et du coup j'ai comparé, et en effet le profil est comparable, mais je trouve quand même une sacré différence : 

[image]

[image]

Alors ça pourrait être lié au fait qu'il y a des pics d'un côté, et une moyenne de l'autre, mais la différence est la même sur les plats, donc pas un effet de bord de la moyenne.

J'avais trouvé qu'il fallait, pour passer de l'un à l'autre, connaître le Power Factor, qui n'est pas remonté par le zlinky : 

[image]

https://www.powerstream.com/VA-Watts.htm

> In other words, volt-amps x power factor = watts. Similarly, KVA*PF = KW

Se pourrait-il que ce power factor change selon le linky, la région, je sais pas quel autre critère ?

CladStriffe, 2023-03-03:
Le facteur de puissance n'est pas constant et dépend des équipements que tu as chez toi. 
Pour les résistances pures (ampoules à incandescence, grille pain, bouilloire, chauffe eau, radiateur électrique, ...) le facteur de puissance vaut 1. 
Ça se complique quand il y a des transformateurs ou moteurs électriques (frigo, pompe à chaleur, plaque à induction, alimentation pour PC, chargeurs de tel etc) et là le facteur de puissance est inférieur à 1. Ça veut dire que toute la puissance délivrée par le réseau (puissance apparente), nécessaire pour que l'équipement fonctionne, n'est en fait pas utilisée et une partie peut repartir sur le réseau (la puissance réactive). 
Pour comprendre d'où ça vient, pense à un condensateur par exemple : en courant alternatif, il va se charger puis se décharger. Il faut lui fournir de la puissance, mais il la restitue après. 

Au final:
- SINSTS est la puissance apparente, c'est bien ce que le Linky voit passer comme puissance sur ton réseau. Et c'est la seule valeur instantanée qu'on peut lire
- L'énergie qu'on paye en tant que particulier n'est que liée à la puissance active (donc on ne paye pas la puissance réactive. C'est par contre facturé dans l'industrie). C'est donc bien ce qui nous intéresse, malheureusement le Linky ne sait pas la donner en temps réel car il doit faire des moyennes pour regarder aussi ce qui "ressort" de ton réseau (je schématise, mais c'est l'idée de cette puissance reactive). 
- pour calculer le facteur de puissance, tu peux en approximation regarder le rapport entre SINSTS et CCASN sur tes valeurs constantes la nuit. Ça donne une idée, mais ce ne sera pas complètement représentatif le reste de la journée. Par exemple, la nuit je suis autour de 0,6 ou 0,7. Mais quand mon chauffe eau se met en route, comme c'est une grosse puissance active et rien en réactif, mon facteur de puissance est proche de 1. 
- c'est pour ça que je préfère quand même utiliser SINSTS pour suivre ma puissance. 

Ça fait un gros pavé, je ne suis absolument pas expert du domaine, c'est juste le résultat de mes lectures après m'être posé les mêmes questions que toi.

Karreg, 2023-03-03:
Merci, c'est très clair.

J'en déduis qu'on peut peut pas vraiment suivre la puissance d'un appareil en suivant la puissance apparente à cause de ce facteur de puissance, et que rien ne peut remplacer un Wattmètre. C'était mon besoin initial, je comptais regarder le delta à l'allumage d'un appareil, dommage...

Par contre pour suivre la consommation globale, la puissante apparente est à la fois la valeur réellement facturée, et une approximation de la puissance instantanée plutôt proche de la réalité.

Merci beaucoup pour ces éclaircissements, du coup je ferme le ticket :)

closed 2023-03-03 by Karreg

T0ytoy, 2023-03-23:
Personnellement j'utilise la valeur en kwh remontée par le ZLinky et j'appliquer une dérivée dessus pour obtenir des watts en "temps réel" (à chaque remontée du ZLinky, donc toutes les quelques secondes. Ça semble fonctionner correctement et me donne l'info que je veux.