fairecasoimeme/Zlinky_TIC #186: Mise a jour usb sur PCB 1.4

gsemet, 2023-02-25:
Bonjour,

J'ai le PCB 1.4 que je souhaite mettre a jour pour voir si la remontée des valeurs d'énergie injectée sur le réseau marche mieux, mais je vois que je n'ai pas le connecteur:
[image]

Est-ce que les connections 5V/GND/TX/RX/DTR,3V sont bien les mêmes que sur vos photo du READE? C'est normal de ne pas avoir le connecteur?

Merci

fairecasoimeme, 2023-02-27:
Bonjour, 
l'injection ne peut être remontée sur la TIC uniquement si vous avez un contrat avec EDF.
Quelle version du firmware avez-vous ?
Oui sur la v1.4, il n'y a pas de connecteur mais on peut mettre à jour en utilisant le module USB-TTL

gsemet, 2023-02-27:
Bonjour. J'ai réussi à faire la mise à jour et apparement j'ai le Linky en mode historique, je comprend donc pourquoi je n'ai pas de remontée de la production. Je demande a total énergie de faire le changement de mode :)

fairecasoimeme, 2023-02-27:
je pense que si vous étiez en mode historique, vous n'avez donc pas de contrat de production avec EDF. Donc ça ne suffira pas pour récupérer avec la TIC les données d'injection

gsemet, 2023-02-27:
Si j'ai bien un contrat de revente avec EDF OA depuis 2018.

fairecasoimeme, 2023-02-27:
Parfait, alors ça devrait être OK. Juste bizarre que lors du contrat, on vous ait pas passé d'office en mode standard.

gsemet, 2023-03-02:
Bonjour. Effectivement tout fonctionne maintenant. Je recommende néanmoins de rajouter quelques remarques dans votre readme car les gens ne connaissent pas ces histoires de mode standard/historique:
- le linky a 2 modes historiques et standard qui peuvent facilement etre lu dans l'un des affichages du linky
- certaines fonctions ne peuvent etre activee qu'en mode standard
- pour changer le mode d'historique a standard, il suffit de faire la demande au service client de votre fournisseur qui transmettrera la demande a enedis.
(j'ai fait la demande a TotalDirect Energie le lundi, le mode a ete change sur mon linky le mercredi)

closed 2023-03-02 by gsemet