fairecasoimeme/Zlinky_TIC #187: Z2M: Failed to configure - Error: ConfigureReporting

dmartinpro, 2023-03-13:
Hi,

I've noticed this error message in Z2M logs:

`Zigbee2MQTT:error 2023-03-13 19:35:17: Failed to configure '0x00158d0005be99f5', attempt 1 (Error: ConfigureReporting 0x00158d0005be99f5/1 haElectricalMeasurement([{"attribute":"totalReactivePower","minimumReportInterval":60,"maximumReportInterval":900,"reportableChange":1}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 49642 - 1 - 11 - 2820 - 7 after 10000ms)
`
I've tried to restart Z2M, same issue.
ZLinky version is is 0013, Z2M is up to date.

What does it mean and how to get rid of it?

raspbeguy, 2023-07-26:
I have the same behaviour. Data are correctly reported but this error shows up every time I restart z2m.

razmoket89, 2023-08-05:
Same issue here.