fairecasoimeme/Zlinky_TIC #188: Changement de trame TIC historique

bax137, 2023-03-14:
Pour info depuis hier sans que je ne demande rien, la trame de mon linky en mode historique est passée à la trame standard. J'ai vérifié sur l'écran du linky, je suis toujours bien en mode historique.

Home assistant ne recevait plus en mqtt les données de la trame historique, pour résoudre le problème j'ai forcé le mode standard dans zigbee2mqtt, je reçois bien la nouvelle trame.

bax137, 2023-03-22:
De plus en plus curieux, depuis 10 jours NGTF me remonte en permanence "HC.." que je sois en heures creuses ou en heures pleines. Les indexes HC et HP sont corrects.

Gonioul, 2023-05-06:
Bonjour, en ce qui me concerne, j'ai enfin réussi à demander le passage TIC en standard dans mon compte EDF, je pense que ça fait suite à ma demande de passage en mode horaire.

Par la suite, un unplug/plug électrique du zlinky, suppression de l'ancien device dans HA (sinon pas de mise à jour), restart de Zigbee2MQTT, et les nouveaux paramètres sont dispos, mais bon avec perte des anciens.