fairecasoimeme/Zlinky_TIC #189: Recup Data depuis MQTT vers InfluxDB puis Grafana

steflp, 2023-03-27:
Hello
 je voudrais recuperer les data prvenant du Zlinky via le broker mqtt ( Z2M installé)  pour pousser le data dans InfluxDB puis Grafana
qqn aurait-il deja fait ca ? 
qqn peut m'aider sur mqtt et la recup de data ?
 J'utilise MQTT Explorer pour lire les "topic" mais je suis un peu perdu
Merci

nicocha75, 2023-03-28:
Bonjour
Je fais exactement ça en utilisation un outils qui s'appelle telegraf
Voici la config pour MQTT, je te laisse cherché pour connecté telegraf à influxdb

```
 [[inputs.mqtt_consumer]]
   servers = ["tcp://192.168.1.8:1883"]
   topics = [
	  "zigbee2mqtt/linky/#",
	  "Energy/#",
   ]
   data_format = "json"
```

ensuite il n'y a pas plus simple dans grafana pour récupéré les données.

Nicolas

steflp, 2023-03-28:
Hello
Merci  pour  le fichier de configuration qui  m'aidera bcp 
J'ai également Telegraf d'installler mais  coté connecteur  pas facile le paramétrage  !
( j'ia deja un mqtt pour une station meteo connectée) 

L'autre souci du  moment ; reussir a RE-associer mon Zlinky avec ma clef USB SonOff  je l'ai supprimé de ma config et impossibe a la remettre  ! 

Des que ca tourne , je vous fait un feedback 

Merci

steflp, 2023-04-20:
Bon et bien , j'ai reçu  le modele avec antenne et c'est toujours pareil  ... aucune connexion si ca continue je vais devoir deplacer mon compteur Linky et le coller a  ma clef SonOff .. je ne m'en sors pas !

fairecasoimeme, 2023-04-21:
Avez-vous un raspberry Pi avec disque SSD ?
https://www.youtube.com/watch?v=tHqZhNcFEvA

steflp, 2023-04-21:
Non PC dell 3020 micro avec un hub zigbee reseau + PA Dlink wifi à proximité

steflp, 2023-04-21:
Re, avec une rallonge USB2.0 "active"  c'est bon mon Zlinky est reconnu mais pas bcp de  puissance  ... 
 je continue  mes tests ce we en deplacant mon "serveur" DELL pres du compteur Linky

steflp, 2023-04-21:
[image]
Bon je crois que l'on peut fermer cette demande  
Merci  encore  :-)

steflp, 2023-04-28:
Bonjour @nicocha75   
 j'ai  un petit souci  pour recupérer certaines data  : 
`# Read metrics from MQTT topic(s)
[[inputs.mqtt_consumer]]
  servers = ["tcp://127.0.0.1:1883"]
  topics = [
  "zigbee2mqtt/Zlinky/#",
  ]
  data_format = "json"`

je n'arrive pas à récupérer cette valeur "active_register_tier_delivered" 

De  plus  est-ce normal que la valeur   "tomorrow_color" reste toujours a cette valeur   "----" ? 

Merci  !
Stef

fairecasoimeme, 2023-04-28:
La couleur du lendemain n'est disponible qu'à certaines heures
https://github.com/fairecasoimeme/Zlinky_TIC/issues/139#issuecomment-1344400845

closed 2023-05-12 by steflp