fairecasoimeme/Zlinky_TIC #191: NJOURF & NJOURF+1 & PJOURF+1 qui ne remontent pas

deexiz, 2023-04-06:
Bonjour,

J'ai mon Linky et Zlinky en mode standard, avec un abonnement Tempo.
La couleur du jour remonte bien (LTARF & NTARF) mais je n'ai pas l'information de la couleur du lendemain.
Ça fonctionnait quand j'étais en mode historique.

Voici les informations que j'ai : 

- NJOURF = 0
- NJOURF+1 = 0
- PJOURF+1 = 00004001 06004002 22004001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE
- PPOINTE1 = 00004003 06004004 22004003 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE

Tout le reste semble fonctionner. 
J'ai déjà supprimé et réappairé le Zlinky mais ça n'a rien changé.
Le Zlinky est en firmware 4000-0013.

Une idée d'où peut se situer le problème ?

fairecasoimeme, 2023-04-06:
Bonjour,

Sur le mode standard la couleur du lendemain se trouve dans l’étiquette STGE : registre de statut pour les abonnements de type historique Tempo.
Vous pouvez retrouver ces informations dans le document : Enedis-NOI-CPT_54E.pdf

Il faut transformer la valeur de l’étiquette STGE (qui est en hexadécimal) en valeur binaire et consulter les bits 24 à 27.

Ci-joint un extrait du tableau du registre :
 
[image]

Je vous concède que c’est pas évident par rapport au mode historique mais malheureusement, c’est comme cela que ça fonctionne en mode standard.

deexiz, 2023-04-06:
Merci beaucoup pour l'information ! 
C'est étonnant de leur part, mais mon problème est résolu.

closed 2023-04-06 by deexiz