fairecasoimeme/Zlinky_TIC #192: Couleur lendemain mode Standard tarif TEMPO

dcyou, 2023-04-26:
Bonjour,  

j'essaye de retrouver la couleur du lendemain (elle n'est pas dans tomorrow color)

je suis bien en 
```
NGTF
Supplier pricing schedule name
TEMPO
LTARF
Current supplier price label
HP BLANC
```
Ma valeur de registre est la suivante
```
STGE
Register of Statutes
423A4C01
```
Soit
```
0100 0010 0011 1010 0100 1100 0000 0001

0 Contact sec 
100 Organe de coupure
0 État du cache-bornes distributeur
0 Non utilisé
1 Surtension sur une des phases
0 Dépassement de la puissance de référence
0 Fonctionnement producteur/consommateur
0 Sens de l’énergie active 
1110 Tarif en cours sur le contrat fourniture
10 Tarif en cours sur le contrat distributeur
0   Mode dégradée de l’horloge (perte de l’horodate de l’horloge interne)
1 État de la sortie télé-information 
0 Non utilisé
01 État de la sortie communication Euridis
10 Statut du CPL 
0 Synchronisation CPL 
00 Couleur du jour  
00 Couleur du lendemain
00 Préavis pointes mobiles
01 Pointe mobile (PM)
```
or j'ai 00 pour couleur du jour et lendemain
[image]

Soit Enedis ne renvoit pas correctement soit quelque chose m'échappe ? 
+ si vous avez une solution/piste pour retrouver cette information je suis preneur

peut être lié à ?
https://github.com/fairecasoimeme/Zlinky_TIC/issues/191

doc:
[Enedis-NOI-CPT_54E.pdf](https://github.com/fairecasoimeme/Zlinky_TIC/files/11331648/Enedis-NOI-CPT_54E.pdf)

fairecasoimeme, 2023-04-27:
Bonjour,

01000010001110100100110000000001 <--- il faut partir dans ce sens pour la numérotation des bits

1 Contact sec (bit 0)
000 Organe de coupure
0 État du cache-bornes distributeur
0 Non utilisé
0 Surtension sur une des phases
0 Dépassement de la puissance de référence
0 Fonctionnement producteur/consommateur
0 Sens de l’énergie active 
0011 Tarif en cours sur le contrat fourniture
01 Tarif en cours sur le contrat distributeur
0   Mode dégradée de l’horloge (perte de l’horodate de l’horloge interne)
1 État de la sortie télé-information 
0 Non utilisé
11 État de la sortie communication Euridis
01 Statut du CPL 
0 Synchronisation CPL 
10 Couleur du jour  --> **couleur du jour blanc**
00 Couleur du lendemain --> **pas d'annonce**
00 Préavis pointes mobiles
01 Pointe mobile (PM)

dcyou, 2023-04-27:
j'ai eu un doute en plus... merci ;) 
je surpris de n'avoir pas d'annonce pour le lendemain
[image]

il est précisé "historique tempo" donc ce qui m'échappe c'est pourquoi j'ai bien la valeur actuelle mais celle du lendemain.

Si vous avez une piste ça m'aiderait ;) sinon je pensais voir avec ENEDIS

fairecasoimeme, 2023-04-27:
Il me semble que ya des heures précises pour les annonces du lendemain : 
https://github.com/fairecasoimeme/Zlinky_TIC/issues/139#issuecomment-1344400845
Dans l'exemple c'est sur le mode historique mais je pense que c'est pareil pour le mode standard.

closed 2023-04-28 by dcyou

pipiche38, 2023-07-25:
le STGE renvoie bien le Registre des Status en mode standard?
Par contre la doc Enedis indique bien que les bits 26 et 27 "Couleur du Lendemain" sont à 0 hors contrat Historique Tempo.

[image]

fairecasoimeme, 2023-07-26:
Il ne faut pas confondre le mode de communication et l'abonnement (contrat).
Le registre des status STGE n'existe que dans le mode standard. (c'est le mode de communication)
Le contrat historique tempo correspond à un contrat spécifique au fournisseur EDF qui est le seul contrat qui gère les périodes de couleurs bleu blanc rouge.
Donc c'est tout à fait normal que hors contrat tempo (contrat historique edf), la valeur est à 0