fairecasoimeme/Zlinky_TIC #193: ZLinky_TIC non alimenté

uncle-buzz, 2023-04-30:
Bonjour,
J'ai un ZLinky_TIC  v1.1 firmware 4000-0012 du 20221121 (information zigbee2mqtt) que j'ai testé sur un linky en appartement : RAS, tout fonctionne bien
Je vais déménager dans une maison, j'emmène mon serveur zigbee2mqtt avec mes capteurs et le ZLinky_TIC  que j'installe sur le linky de la maison, sauf qu'avant de pouvoir tester l'appairage avec un répéteur zigbee (le compteur est en limite de terrain) je constate que la led du ZLinky_TIC  ne s'allume pas... Le linky est bien en mode TIC historique (comme à l'appartement)
Vérification du fusible du module TIC du linky : RAS, il n'est pas coupé
Que dois-je faire pour vérifier que les sorties TIC du linky fonctionnent ? je compte retester le ZLinky_TIC  sur le linky de mon appartement pour vérifier qu'il n'a subi aucun dommage entre le moment où il fonctionnait dans l'appartement et la tentative à la maison, mais côté linky comment détecté un problème ?
Merci

fairecasoimeme, 2023-05-03:
Vous pouvez utiliser un multimètre configuré sur voltmètre alternatif VAC. Côté Linky, vous placez une première sonde sur la borne A et l'autre sonde sur I1. la tension à vide doit être de 13VAC

uncle-buzz, 2023-05-03:
Bonjour,
Merci pour votre réponse, je testerai ce weekend quand je serai de nouveau à la maison, ici à l'appartement le ZLinky_TIC s'est bien allumé, il n'est donc pas en cause, y a-t-il une configuration de mon compte linky qui pourrait ne pas avoir activé la sortie TIC et qu'il faudrait vérifier ou bien s'agit-il forcément d'une défaillance ? J'avais déjà regardé la tension entre A et I1 mais je n'ai pas relevé la valeur, ne sachant pas ce que je devais trouver à ce moment là.
En fonction du résultat, une idée de quoi contrôler ensuite ?
Merci

fairecasoimeme, 2023-05-03:
Malheureusement probablement une défaillance du Linky.

nicocha75, 2023-06-08:
Bonjour
Petit problème identique, en A et I1, je n'ai que 1,8V, donc le zlinky ne fonctionne pas. Ce qui est étrange c'est que j'ai deja eu le cas, et j'ai due débranché le module durant un certain temps pour que la tension revienne. Comme si la puissance était limité par le linky à un moment. ça arrive souvent?
MErci

uncle-buzz, 2023-06-08:
Petit retour pour le suivi, donc contrôle de la tension d'alimentation sur le linky où il était précédemment fonctionnel à 11.8VAC de mémoire
ZLinky_TIC hors de cause donc, retour à la maison, contrôle de la tension : 11.8VAC... Je rebranche le ZLinky_TIC et là ça fonctionne...
Cause inconnue donc, en espérant que ce soit un faux contact et non pas une panne récurrente pouvant survenir à nouveau.

fairecasoimeme, 2023-06-09:
> Bonjour Petit problème identique, en A et I1, je n'ai que 1,8V, donc le zlinky ne fonctionne pas. Ce qui est étrange c'est que j'ai deja eu le cas, et j'ai due débranché le module durant un certain [...]

ça n'arrive pas souvent mais Enedis est au courant de ce genre de chose. Il semblerait que qu'un composant se mette en protection mais on ne connait pas encore la cause.

J'ai pu le reproduire il y a longtemps sur mon compteur lors de manipulations (mauvaises) sur le TIC mais impossible depuis à reproduire.

uncle-buzz, 2025-07-15:
Bonjour,
j'avais fait la mise à jour OTA vers la v16 depuis zigbee2mqtt, et peu après, plus de données (peu après, pas immédiatement, donc la v16 a fonctionné)
J'ai reflashé le zlinky manuellement avec la v16 sans option routeur, alimenté par le programmeur la led bleue flash correctement, en revanche sur mon linky, nada, aucune réaction quand je le branche.
J'ai toujours environ 11VAC entre les bornes A et I1 sur le linky pendant que le zlinky y est raccordé, mais aucun signe de vie du zlinky, une idée de quoi tester/vérifier ? Une demande à faire à ENEDIS pour vérifier le compteur ?
Ca fait quelques mois déjà, le zlinky a été débranché plusieurs jours, mais aucun changement quand je le reconnecte.
En cas d'erreur de communication il devrait y avoir un clignotement lent, mais pourquoi aucun clignotement alors qu'il y a une tension d 'alimentation ?

uncle-buzz, 2025-07-23:
Après avoir dessoudé puis ressoudé la puce que je pense être un [ME6214C50M5G](https://lcsc.com/product-detail/Voltage-Regulators-Linear-Low-Drop-Out-LDO-Regulators_MICRONE-ME6214C50M5G_C487908.html), ça semble refonctionner, j'ai 4.9V après le régulateur, et 3.3V après le 2nd, je verra ce week-end une fois remis sur le linky, mais si les schéma au moins pdf pouvais arriver sur le dépôt ça serait plus simple pour dépanner.
Pourquoi 2 ldo en série ?

fairecasoimeme, 2025-07-28:
Historiquement le premier régulateur 5V était destiné à un supercondensateur 5V. n'ayant pas besoin de celui ci, il a été remplacer par un autre condensateur.
Non pas possible d'avoir les schémas