fairecasoimeme/Zlinky_TIC #194: Too many events in Zigbee2MQTT log file for LiXee

tomav, 2023-05-10:
Hi there, thanks for the LiXee which I found reliable so far and easy to integrate.
I post this message in english in case @Koenkk wants to jump in the conversation.
I have several zigbee devices configured in my Z2M but LiXee is spreading a lot of similar info at the same time as you can see in the following extract:

```
zlinky config: standard, single_phase, false, BASE
Zigbee2MQTT:info  2023-05-10 08:21:50: MQTT publish: topic 'zigbee2mqtt/LiXee', payload '{"active_enerfy_out_d01":31608.11,"active_power":822,"active_power_max":3493,"active_power_ph_b":594,"active_register_tier_delivered":"","apparent_power":2493,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E230510082123","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":39167.61,"current_tarif":"BASE","current_tier1_summ_delivered":39167.6,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3316,"linkquality":69,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"0123456789","power_threshold":12,"relais":0,"rms_current":10,"rms_current_max":0,"rms_voltage":242,"site_id":"0123456789","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
Zigbee2MQTT:info  2023-05-10 08:21:50: MQTT publish: topic 'zigbee2mqtt/LiXee', payload '{"active_enerfy_out_d01":31608.11,"active_power":822,"active_power_max":3493,"active_power_ph_b":594,"active_register_tier_delivered":"","apparent_power":2493,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E230510082123","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":39167.61,"current_tarif":"BASE","current_tier1_summ_delivered":39167.6,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3316,"linkquality":72,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"0123456789","power_threshold":12,"relais":0,"rms_current":10,"rms_current_max":0,"rms_voltage":242,"site_id":"0123456789","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
Zigbee2MQTT:info  2023-05-10 08:21:50: MQTT publish: topic 'zigbee2mqtt/LiXee', payload '{"active_enerfy_out_d01":31608.11,"active_power":822,"active_power_max":3493,"active_power_ph_b":594,"active_register_tier_delivered":"","apparent_power":2493,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E230510082123","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":39167.61,"current_tarif":"BASE","current_tier1_summ_delivered":39167.6,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3316,"linkquality":72,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"0123456789","power_threshold":12,"relais":0,"rms_current":10,"rms_current_max":0,"rms_voltage":242,"site_id":"0123456789","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
Zigbee2MQTT:info  2023-05-10 08:21:50: MQTT publish: topic 'zigbee2mqtt/LiXee', payload '{"active_enerfy_out_d01":31608.11,"active_power":822,"active_power_max":3493,"active_power_ph_b":594,"active_register_tier_delivered":"","apparent_power":2493,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E230510082145","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":39167.61,"current_tarif":"BASE","current_tier1_summ_delivered":39167.6,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3316,"linkquality":69,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"0123456789","power_threshold":12,"relais":0,"rms_current":10,"rms_current_max":0,"rms_voltage":242,"site_id":"0123456789","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
Zigbee2MQTT:info  2023-05-10 08:21:50: MQTT publish: topic 'zigbee2mqtt/LiXee', payload '{"active_enerfy_out_d01":31608.11,"active_power":822,"active_power_max":3493,"active_power_ph_b":594,"active_register_tier_delivered":"","apparent_power":2493,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E230510082145","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":39167.61,"current_tarif":"BASE","current_tier1_summ_delivered":39167.6,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3316,"linkquality":72,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"0123456789","power_threshold":12,"relais":0,"rms_current":10,"rms_current_max":0,"rms_voltage":242,"site_id":"0123456789","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
Zigbee2MQTT:info  2023-05-10 08:21:50: MQTT publish: topic 'zigbee2mqtt/LiXee', payload '{"active_enerfy_out_d01":31608.11,"active_power":822,"active_power_max":3493,"active_power_ph_b":594,"active_register_tier_delivered":"","apparent_power":2493,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E230510082145","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":39167.61,"current_tarif":"BASE","current_tier1_summ_delivered":39167.6,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3316,"linkquality":72,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"0123456789","power_threshold":12,"relais":0,"rms_current":10,"rms_current_max":0,"rms_voltage":242,"site_id":"0123456789","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
Zigbee2MQTT:info  2023-05-10 08:21:50: MQTT publish: topic 'zigbee2mqtt/LiXee', payload '{"active_enerfy_out_d01":31608.11,"active_power":822,"active_power_max":3493,"active_power_ph_b":594,"active_register_tier_delivered":"","apparent_power":2493,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E230510082145","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":39167.61,"current_tarif":"BASE","current_tier1_summ_delivered":39167.6,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3316,"linkquality":72,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"0123456789","power_threshold":12,"relais":0,"rms_current":10,"rms_current_max":0,"rms_voltage":242,"site_id":"0123456789","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
Zigbee2MQTT:info  2023-05-10 08:21:50: MQTT publish: topic 'zigbee2mqtt/LiXee', payload '{"active_enerfy_out_d01":31608.11,"active_power":822,"active_power_max":3493,"active_power_ph_b":594,"active_register_tier_delivered":"","apparent_power":2493,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E230510082145","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":39167.61,"current_tarif":"BASE","current_tier1_summ_delivered":39167.6,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3316,"linkquality":72,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"0123456789","power_threshold":12,"relais":0,"rms_current":10,"rms_current_max":0,"rms_voltage":242,"site_id":"0123456789","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
Zigbee2MQTT:info  2023-05-10 08:21:50: MQTT publish: topic 'zigbee2mqtt/LiXee', payload '{"active_enerfy_out_d01":31608.11,"active_power":822,"active_power_max":3493,"active_power_ph_b":594,"active_register_tier_delivered":"","apparent_power":2493,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E230510082145","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":39167.61,"current_tarif":"BASE","current_tier1_summ_delivered":39167.6,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3316,"linkquality":72,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"0123456789","power_threshold":12,"relais":0,"rms_current":10,"rms_current_max":0,"rms_voltage":242,"site_id":"0123456789","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":false,"warn [... cut ...] rent_max":0,"rms_voltage":242,"site_id":"0123456789","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
```

This happens at every poll, currently configured at 15s.
I don't understand why there are some many lines in a short period of time.
I don't know neither if the problem comes from the LiXee firmware (`4000-0013`) or from Zigbee2MQTT (`1.30.4 commit: b2dd21e`).

Any clue?

Thanks.

tomav, 2023-05-10:
Seems known in Zigbee2MQTT:
https://github.com/Koenkk/zigbee2mqtt/issues/15679
https://github.com/Koenkk/zigbee2mqtt/issues/16987

tomav, 2023-05-10:
Can't tell you, installed Z2M and upgraded directly :(

fairecasoimeme, 2023-05-11:
could you try to just reboot Z2M or unpair ZLinky then reboot Z2M then re pair

trollix, 2023-05-24:
I have similar problem in zigbee2mqt logs (error logs only):
`
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
zlinky config: standard, single_phase, false, HC SEM WE LUNDI
`

tomav, 2023-05-26:
Note: Rebooting Z2M does not fix the issue.

trollix, 2023-05-27:
> could you try to just reboot Z2M or unpair ZLinky then reboot Z2M then re pair

Reboot ing Z2M - re-appairing Lixee does not fix the problem i have on Z2M logs

tomav, 2023-07-19:
Just had a look to my db, it's 16GB 🤯, mostly filled by zlinky.

This is what I get:

``` 
Zigbee2MQTT:info  2023-07-19 22:13:21: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":31871.33,"active_enerfy_out_d02":7899,"active_enerfy_out_d03":0,"active_enerfy_out_d04":0,"active_power":304,"active_power_max":3274,"active_power_ph_b":830,"active_register_tier_delivered":"","apparent_power":446,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E230719221256","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":40197.324,"current_tarif":"BASE","current_tier10_summ_delivered":0,"current_tier1_summ_delivered":40197.324,"current_tier2_summ_delivered":0,"current_tier3_summ_delivered":0,"current_tier4_summ_delivered":0,"current_tier5_summ_delivered":0,"current_tier6_summ_delivered":0,"current_tier7_summ_delivered":0,"current_tier8_summ_delivered":0,"current_tier9_summ_delivered":0,"days_number_current_calendar":0,"days_number_next_calendar":0,"days_profile_current_calendar":"00008001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE","days_profile_next_calendar":"","drawn_v_a_max_n1":3489,"linkquality":83,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":2,"rms_current_max":0,"rms_voltage":242,"site_id":"07593777070098","software_revision":2,"start_mobile_point1":0,"start_mobile_point2":0,"start_mobile_point3":0,"status_register":"003A4001","stop_mobile_point1":0,"stop_mobile_point2":0,"stop_mobile_point3":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
Zigbee2MQTT:info  2023-07-19 22:13:21: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":31871.33,"active_enerfy_out_d02":7899,"active_enerfy_out_d03":0,"active_enerfy_out_d04":0,"active_power":304,"active_power_max":3274,"active_power_ph_b":830,"active_register_tier_delivered":"","apparent_power":446,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E230719221256","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":40197.324,"current_tarif":"BASE","current_tier10_summ_delivered":0,"current_tier1_summ_delivered":40197.324,"current_tier2_summ_delivered":0,"current_tier3_summ_delivered":0,"current_tier4_summ_delivered":0,"current_tier5_summ_delivered":0,"current_tier6_summ_delivered":0,"current_tier7_summ_delivered":0,"current_tier8_summ_delivered":0,"current_tier9_summ_delivered":0,"days_number_current_calendar":0,"days_number_next_calendar":0,"days_profile_current_calendar":"00008001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE","days_profile_next_calendar":"","drawn_v_a_max_n1":3489,"linkquality":87,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":2,"rms_current_max":0,"rms_voltage":242,"site_id":"07593777070098","software_revision":2,"start_mobile_point1":0,"start_mobile_point2":0,"start_mobile_point3":0,"status_register":"003A4001","stop_mobile_point1":0,"stop_mobile_point2":0,"stop_mobile_point3":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
Zigbee2MQTT:info  2023-07-19 22:13:21: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":31871.33,"active_enerfy_out_d02":7899,"active_enerfy_out_d03":0,"active_enerfy_out_d04":0,"active_power":304,"active_power_max":3274,"active_power_ph_b":830,"active_register_tier_delivered":"","apparent_power":446,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E230719221256","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":40197.324,"current_tarif":"BASE","current_tier10_summ_delivered":0,"current_tier1_summ_delivered":40197.324,"current_tier2_summ_delivered":0,"current_tier3_summ_delivered":0,"current_tier4_summ_delivered":0,"current_tier5_summ_delivered":0,"current_tier6_summ_delivered":0,"current_tier7_summ_delivered":0,"current_tier8_summ_delivered":0,"current_tier9_summ_delivered":0,"days_number_current_calendar":0,"days_number_next_calendar":0,"days_profile_current_calendar":"00008001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE","days_profile_next_calendar":"","drawn_v_a_max_n1":3489,"linkquality":87,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":2,"rms_current_max":0,"rms_voltage":242,"site_id":"07593777070098","software_revision":2,"start_mobile_point1":0,"start_mobile_point2":0,"start_mobile_point3":0,"status_register":"003A4001","stop_mobile_point1":0,"stop_mobile_point2":0,"stop_mobile_point3":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
Zigbee2MQTT:info  2023-07-19 22:13:21: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":31871.33,"active_enerfy_out_d02":7899,"active_enerfy_out_d03":0,"active_enerfy_out_d04":0,"active_power":304,"active_power_max":3274,"active_power_ph_b":830,"active_register_tier_delivered":"","apparent_power":446,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E230719221307","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":40197.324,"current_tarif":"BASE","current_tier10_summ_delivered":0,"current_tier1_summ_delivered":40197.324,"current_tier2 [... cut ...] SAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":2,"rms_current_max":0,"rms_voltage":242,"site_id":"07593777070098","software_revision":2,"start_mobile_point1":0,"start_mobile_point2":0,"start_mobile_point3":0,"status_register":"003A4001","stop_mobile_point1":0,"stop_mobile_point2":0,"stop_mobile_point3":0,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":false,"warn_d_p_s":0}'
```

Any way to debug this?

fairecasoimeme, 2023-07-20:
Sorry but I don't know why there is a flood. It cannot be provided by ZLinky.
You have to verify polling from Z2M. From ZLinky, reports can't be less than 61 seconds

SheelGreg, 2023-09-04:
I had the same problem, I am at the origin of the 2 posts quoted above (and one new post, maybe not necessary i'll delete him)

I was able to work around the problem with these values in Z2M:
Retain: true (it's not for this problem but it can avoid bugs if you calculate your cost per kw/h and it avoids bad returns with, for example, days at +300€ after a reboot of HA)
Debounce: 20 (beyond this, packets may be lost)
Measurement poll interval: 120 (although this parameter doesn't seem to be taken into account, whatever the value, that's weird)
Measurement poll chunk : 1

With these values, I get a poll every 60 seconds (??? why exactly 60 ? Don't know. Sometimes ~50 sec), rarely more but it can happen...

I've created another post on this problem to push towards the 0xff66 cluster which is supposed to indicate the poll frequency
https://github.com/fairecasoimeme/Zlinky_TIC/issues/204

Hope it can help :)

SheelGreg, 2023-09-04:
Ok sorry it's not exactly the same problem since it's only the indexes that are brought up here, not the values of those indexes as was the case with me. On the other hand, I've just tested a solution given here: https://github.com/fairecasoimeme/Zlinky_TIC/issues/174#issuecomment-1632208423
It seems that this particular problem is solved with this solution

trollix, 2023-09-15:
Same problem for me in zigbee2mqtt logs

[image]

`
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI

`

tomav, 2023-10-12:
New installation, new hardware, same issue.

```
zlinky config: standard, single_phase, false, BASE
Zigbee2MQTT:info  2023-10-12 20:24:34: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32173.842,"active_power":620,"active_power_max":3469,"active_power_ph_b":400,"apparent_power":389,"available_power":12,"average_rms_voltage_meas_period":242,"current_date":"E231012202427","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41307.405,"current_tarif":"BASE","current_tier1_summ_delivered":41307.405,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3451,"linkquality":65,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":2,"rms_voltage":243,"site_id":"XXXXXXXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-12 20:24:34: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32173.842,"active_power":620,"active_power_max":3469,"active_power_ph_b":400,"apparent_power":389,"available_power":12,"average_rms_voltage_meas_period":242,"current_date":"E231012202427","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41307.405,"current_tarif":"BASE","current_tier1_summ_delivered":41307.405,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3451,"linkquality":65,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":2,"rms_voltage":243,"site_id":"XXXXXXXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-12 20:24:35: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32173.842,"active_power":620,"active_power_max":3469,"active_power_ph_b":400,"apparent_power":389,"available_power":12,"average_rms_voltage_meas_period":242,"current_date":"E231012202427","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41307.405,"current_tarif":"BASE","current_tier1_summ_delivered":41307.405,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3451,"linkquality":61,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":2,"rms_voltage":243,"site_id":"XXXXXXXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-12 20:24:35: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32173.842,"active_power":620,"active_power_max":3469,"active_power_ph_b":400,"apparent_power":389,"available_power":12,"average_rms_voltage_meas_period":242,"current_date":"E231012202427","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41307.405,"current_tarif":"BASE","current_tier1_summ_delivered":41307.405,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3451,"linkquality":61,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":2,"rms_voltage":243,"site_id":"XXXXXXXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-12 20:24:35: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32173.842,"active_power":620,"active_power_max":3469,"active_power_ph_b":400,"apparent_power":389,"available_power":12,"average_rms_voltage_meas_period":242,"current_date":"E231012202427","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41307.405,"current_tarif":"BASE","current_tier1_summ_delivered":41307.405,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3451,"linkquality":61,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":2,"rms_voltage":243,"site_id":"XXXXXXXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-12 20:24:35: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32173.842,"active_power":620,"active_power_max":3469,"active_power_ph_b":400,"apparent_power":389,"available_power":12,"average_rms_voltage_meas_period":242,"current_date":"E231012202427","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41307.405,"current_tarif":"BASE","current_tier1_summ_delivered":41307.405,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3451,"linkquality":65,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":2,"rms_voltage":243,"site_id":"XXXXXXXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-12 20:24:35: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32173.842,"active_power":620,"active_power_max":3469,"active_power_ph_b":400,"apparent_power":389,"available_power":12,"average_rms_voltage_meas_period":242,"current_date":"E231012202427","current_index_tarif":1,"current_price":"BASE","curr [... cut ...] ASE","current_summ_delivered":41307.405,"current_tarif":"BASE","current_tier1_summ_delivered":41307.405,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3451,"linkquality":61,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":2,"rms_voltage":243,"site_id":"XXXXXXXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
```

Koenkk, 2023-10-13:
Could someone provide the debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

tomav, 2023-10-15:
Hi @Koenkk, thanks for jumping in!
Please the [debug log file](https://github.com/fairecasoimeme/Zlinky_TIC/files/12909456/log.txt).
Device is named lixee / 0x00158d000638db70 / ZLinky

Koenkk, 2023-10-15:
Can you try to increase the polling time? The default is 60 seconds, in the z2m configuration.yaml:

```yaml
devices:
   # First device
  '0x00158d000638db70':
    measurement_poll_interval: 600 # <-- add this line, increases polling rate from 60 to 600 seconds
```

tomav, 2023-10-15:
Hi @Koenkk, my config was set to 61 (already tried to go above default setting) which did not change anything.
But 600 seems to do the trick!

tomav, 2023-10-17:
The new configuration with a poll set at 600 is really better.

That said I can see that the issue still occurs:

```
zlinky config: standard, single_phase, false, BASE
Zigbee2MQTT:info  2023-10-17 11:21:12: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32184.496,"active_power":382,"active_power_max":2867,"active_power_ph_b":574,"apparent_power":270,"available_power":12,"average_rms_voltage_meas_period":241,"current_date":"E231017112100","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41366.873,"current_tarif":"BASE","current_tier1_summ_delivered":41366.873,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3387,"linkquality":61,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":1,"rms_voltage":241,"site_id":"XXXXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-17 11:21:12: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32184.496,"active_power":382,"active_power_max":2867,"active_power_ph_b":574,"apparent_power":270,"available_power":12,"average_rms_voltage_meas_period":241,"current_date":"E231017112100","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41366.873,"current_tarif":"BASE","current_tier1_summ_delivered":41366.873,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3387,"linkquality":61,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":1,"rms_voltage":241,"site_id":"XXXXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-17 11:21:12: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32184.496,"active_power":382,"active_power_max":2867,"active_power_ph_b":574,"apparent_power":270,"available_power":12,"average_rms_voltage_meas_period":241,"current_date":"E231017112100","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41366.873,"current_tarif":"BASE","current_tier1_summ_delivered":41366.873,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3387,"linkquality":65,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":1,"rms_voltage":241,"site_id":"XXXXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-17 11:21:12: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32184.496,"active_power":382,"active_power_max":2867,"active_power_ph_b":574,"apparent_power":270,"available_power":12,"average_rms_voltage_meas_period":241,"current_date":"E231017112100","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41366.873,"current_tarif":"BASE","current_tier1_summ_delivered":41366.873,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3387,"linkquality":61,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":1,"rms_voltage":241,"site_id":"XXXXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-17 11:21:12: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32184.496,"active_power":382,"active_power_max":2867,"active_power_ph_b":574,"apparent_power":270,"available_power":12,"average_rms_voltage_meas_period":241,"current_date":"E231017112100","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41366.873,"current_tarif":"BASE","current_tier1_summ_delivered":41366.873,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3387,"linkquality":65,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":1,"rms_voltage":241,"site_id":"XXXXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-17 11:21:12: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32184.496,"active_power":382,"active_power_max":2867,"active_power_ph_b":574,"apparent_power":270,"available_power":12,"average_rms_voltage_meas_period":241,"current_date":"E231017112100","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41366.873,"current_tarif":"BASE","current_tier1_summ_delivered":41366.873,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3387,"linkquality":61,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":1,"rms_voltage":241,"site_id":"XXXXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-17 11:21:12: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32184.496,"active_power":382,"active_power_max":2867,"active_power_ph_b":574,"apparent_power":270,"available_power":12,"average_rms_voltage_meas_period":241,"current_date":"E231017112100"," [... cut ...] :"BASE","current_summ_delivered":41366.873,"current_tarif":"BASE","current_tier1_summ_delivered":41366.873,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3387,"linkquality":65,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":1,"rms_voltage":241,"site_id":"XXXXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
```

SheelGreg, 2023-10-17:
For me, the only thing is working is to set the **Poll Interval to 120 / Measurement Poll Chunk to 120 AND Debounce to 30 (the most important for me).
For some reason, it seems that the Poll Interval doesn't seam to work, i don't remember clearly but try to put 1 or 100 or 1000, it will be the same since the Lixee doesn't wait for a call before sending its data.

With theses data, i have a return each 60 sec, sometimes 30 sec but rarely.
But there is indeed a bug with this Lixee module. Either the Lixee sends info WAAAAAAAY too often and uncontrollably, or zigbee2mqtt has a problem with this module, since 80% of the parameters in z2mqtt don't seem to work.

fairecasoimeme, 2023-10-17:
There are no bug with this module. This module is used with a lot of other plugin and there aren't this issue.
ZLinky answers when you simply ask him an info. There is another way to get values with config reports and ZLinky allows only a minimal report of 60 seconds

SheelGreg, 2023-10-17:
So, there is a problem with z2mqtt because we have 60 reports every seconds when the parameters are set by default :/

fairecasoimeme, 2023-10-17:
As I can see, It's seem to be not the case for all users and I don't know why... But no link with ZLinky firmware. The best is to post this issue in the right place.

trollix, 2023-10-17:
> It's seem to be not the case for all users

Sorry, but it's the same for me, spam of logs:

`
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI
zlinky config: standard, single_phase, false, SEM WE LUNDI

`

Just I want to stop this..
And i don't know how to do do that.
Thank you

SheelGreg, 2023-10-17:
Yeah ! Thanks a lot :)

tomav, 2023-10-18:
Hi @Koenkk, just updated my docker image to latest-dev. Will report the result soon.

tomav, 2023-10-18:
I can see occurrences using `koenkk/zigbee2mqtt:latest-dev` :

```
Zigbee2MQTT:info  2023-10-18 16:54:07: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32186.844,"active_power":380,"active_power_max":3021,"active_power_ph_b":150,"apparent_power":269,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E231018165350","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41376.893,"current_tarif":"BASE","current_tier1_summ_delivered":41376.893,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":2867,"linkquality":54,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":1,"rms_voltage":243,"site_id":"XXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-18 16:54:07: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32186.844,"active_power":380,"active_power_max":3021,"active_power_ph_b":150,"apparent_power":269,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E231018165350","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41376.893,"current_tarif":"BASE","current_tier1_summ_delivered":41376.893,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":2867,"linkquality":54,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":1,"rms_voltage":243,"site_id":"XXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-18 16:54:07: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32186.844,"active_power":380,"active_power_max":3021,"active_power_ph_b":150,"apparent_power":269,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E231018165350","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41376.893,"current_tarif":"BASE","current_tier1_summ_delivered":41376.893,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":2867,"linkquality":51,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":1,"rms_voltage":243,"site_id":"XXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-18 16:54:07: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32186.844,"active_power":380,"active_power_max":3021,"active_power_ph_b":150,"apparent_power":269,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E231018165350","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41376.893,"current_tarif":"BASE","current_tier1_summ_delivered":41376.893,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":2867,"linkquality":51,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":1,"rms_voltage":243,"site_id":"XXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-18 16:54:07: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32186.844,"active_power":380,"active_power_max":3021,"active_power_ph_b":150,"apparent_power":269,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E231018165350","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41376.893,"current_tarif":"BASE","current_tier1_summ_delivered":41376.893,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":2867,"linkquality":54,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":1,"rms_voltage":243,"site_id":"XXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-18 16:54:07: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32186.844,"active_power":380,"active_power_max":3021,"active_power_ph_b":150,"apparent_power":269,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E231018165350","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41376.893,"current_tarif":"BASE","current_tier1_summ_delivered":41376.893,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":2867,"linkquality":54,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":1,"rms_voltage":243,"site_id":"XXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2023-10-18 16:54:07: MQTT publish: topic 'zigbee2mqtt/lixee', payload '{"active_enerfy_out_d01":32186.844,"active_power":380,"active_power_max":3021,"active_power_ph_b":150,"apparent_power":269,"available_power":12,"average_rms_voltage_meas_period":243,"current_date":"E231018165350","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":41376.893,"current_tarif":"BASE","current_t [... cut ...] e":"BASE","current_summ_delivered":41376.893,"current_tarif":"BASE","current_tier1_summ_delivered":41376.893,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":2867,"linkquality":58,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"061861595490","power_threshold":12,"relais":0,"rms_current":1,"rms_voltage":243,"site_id":"XXXXXXXXXX","software_revision":2,"status_register":"003A4001","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null}'
```

Koenkk, 2023-10-18:
@tomav can you provide the debug log of 10 minutes?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

tomav, 2023-10-19:
[Here it is](https://github.com/fairecasoimeme/Zlinky_TIC/files/13039573/lixee.log). Look at 09:34:23

Koenkk, 2023-10-20:
Next to the `measurement_poll_interval`, can you also set `measurement_poll_chunk: 5`?

tomav, 2023-10-23:
I don't see flood anymore with value 5 or 10. Sometimes 2 events in the same second, but never more.
But now I have an error that I have not seen before:

```
Failed to read zigbee attributes: Error: Read 0x00158d000638db70/1 liXeePrivate(["currentDate","currentIndexTarif","currentPrice","currentTarif","message1","message2","relais","statusRegister","drawnVAMaxN1"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 37989 - 1 - 32 - 65382 - 1 after 10000ms)
```

I can see multiple occurrences of this error.

Koenkk, 2023-10-23:
could you try a lower chunk value? e.g. 3 or 4

tomav, 2023-10-24:
No more error with a value of 4 during the past 8 hours.
Still have some clusters of data at the same second, but not that many.

```
2023-10-24 05:52:13: 'zigbee2mqtt/lixee',
2023-10-24 05:52:13: 'zigbee2mqtt/lixee',
2023-10-24 05:52:13: 'zigbee2mqtt/lixee',
2023-10-24 05:52:13: 'zigbee2mqtt/lixee',
```

Koenkk, 2023-10-24:
Great, set the default to 4 now.

tomav, 2023-10-24:
Thanks @Koenkk!

closed 2023-10-24 by tomav