fairecasoimeme/Zlinky_TIC #199: ZLinky semble non alimenté à certains moments

VianneyHENON, 2023-07-14:
Bonjour,
Après des mois de fonctionnement, dimanche dernier le Zlinky est apparu hors ligne dans HA, je l'ai débranché quelques heures, puis après rebranchement la LED a clignoté de nouveau et donc en ligne dans HA.

Aujourd'hui, le phénomène se reproduit, que faire ? Problème avec le Zlinky ou le Linky?

Merci d'avance.

VianneyHENON, 2023-07-18:
Bonjour,
La version est 4000-0012, je n'arrive pas a passer en 13 en OTA avec Home Assistant.
Mais je pense que j'ai eu le même souci que uncle-buzz, avec le souci de tension sur les bornes du tic.
https://github.com/fairecasoimeme/Zlinky_TIC/issues/193

[image]

fairecasoimeme, 2023-07-19:
Je remarque qu'en ce moment, il y a plus de cas de coupure de l'alimentation de la téléinformation des Linky que d'habitude. soit des mises à jours ou peut-être lié à la chaleur ... je vais remonter une liste de cas. pouvez-vous m'envoyer le modèle et la marque de votre linky ou une phot par mail ?

closed 2023-07-19 by fairecasoimeme

VianneyHENON, 2023-07-19:
Bonjour,
Linky de marque ITRON modèle T10716
Autre référence ITRON ITE411L3B

VianneyHENON, 2023-09-02:
Bonjour,
Le phénomène est de nouveau réapparu cette nuit.
Pour info.

louistrapadoux, 2023-11-23:
bonjour, il m'est arrivé aussi la même chose aujourd'hui, le zlinky ne semblait plus allimenté, après l'avoir deinstaller du compteur et attendu quelques heures comme suggeré, il s'est remi a fonctionner. la version du firmware est la dernière : 4000-0013