fairecasoimeme/Zlinky_TIC #200: Home Assistant - Discovered entities with a name that starts with the device name

kefrens, 2023-08-07:
Suite derniere mise a jour j'ai le warning suivant:

Some MQTT entities have an entity name that starts with the device name. This is not expected. To avoid a duplicate name the device name prefix is stripped of the entity name as a work-a-round. Please inform the maintainer of the software application that supplies the affected entities to fix this issue.

List of affected entities:
sensor.lixee_linky_fpm2
sensor.lixee_linky_east
sensor.lixee_linky_easd03
sensor.lixee_linky_easf06
sensor.lixee_linky_dpm2
...

Lien vers la nouvelle documentation : https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/

JigSawFr, 2023-08-07:
Same on my side ;)

fairecasoimeme, 2023-08-07:
Ces alertes ne concernent pas le produit ni son firmware mais plutôt Z2M. De ce que j’ai pu lire, il faut attendre une mise à jour de Z2M.
https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099

PEGARifrouf, 2023-08-16:
Pour info c'est un warning temporaire il y a des actions en cours qui vont le résoudre, aucune action des utilisateurs n'est requise.
J'en ai d'autres pour beaucoup de devices .