fairecasoimeme/Zlinky_TIC #201: Problème de fonctionnement : connexion distante entre Linky et Zlinky_TIC

scalmaa, 2023-08-09:
Bonjour à toute la communauté.

Mes connaissances en électronique sont parties en fumée depuis bien longtemps et je sollicite l’avis et l’aide précieuse de la communauté.

Mon objectif est d’installer un module Lixee fonctionnel et raccordé à mon installation HA. (NUC, Z2M, MQTT, le tout dans les dernières versions disponibles).

Mon contexte est que le compteur Linky est en limite de propriété et qu’il y a 50m entre le compteur Linky et mon garage ou se trouve mes tableaux électriques. Je croyais astucieux de passer un câble électrique (câble de 1,5mm2 dans les goulottes existantes: ce que j’ai fait non sans galère) pour raccorder le Lixee placé dans le garage et à portée du réseau zigbee de la maison aux bornes de mon compteur Linky. Le module Lixee est donc connecté au compteur Linky via un câble électrique de 50m qui transite dans les goulottes enterrées.

Le souci est que le module Lixee reçoit mal les informations TIC du compteur Linky (clignotement de la LED bleu toutes les 2s et effectivement malgré un parfait raccordement au réseau Zigbee, les informations remontées ne sont pas fiables et pas vraiment cohérentes. Extrait de la doc Lixee: L’appareil clignote plusieurs fois lentement. Zlinky_TIC rencontre des erreurs lors de l’acquisition des données du Linky.

Je cherche donc une solution à la situation et des explications parmi :

Le câble électrique n’est pas une bonne idée : à remplacer par un câble téléphonique ?

installer un répéteur zigbee pour étendre le réseau : assez compliqué dans le contexte

, installer un amplificateur de signal ?

J’ai bien trouver une documentation sur le TIC et Linky mais qui ne m’a pas vraiment aidé.

Bref je cherche… et j’ai besoins de conseils pour explorer les pistes possibles.

Merci par avance pour votre aide.

SilentT-FR, 2023-08-10:
Le câble électrique n’est pas une bonne idée : à remplacer par un câble téléphonique ? => le cable telephonique est pas forcement mieux isolé que le câble électrique, mettre du câble RJ45 cat 6 avec les bonne lettre de blindage SFTP je crois serai le mieux, par contre c'est en terme de perte en tension que le câble RJ45 va avoir du mal, en effet plus la section du câble électrique est gros moins la tension baisse , si tu a mis du 1.5² garde le pour le  +&- et le RJ45 pour la RX

cyber-g, 2023-08-21:
Je confirme que le câble électrique n'est pas adapté pour ce cas de transmission. 

Concernant le câble téléphonique, je suis un peu plus optimiste. Les documents fournis par Enedis mentionnent que le système TIC (du moins la partie "émission") est basé sur la norme EURIDIS et que la longueur de câble maximum supportée est de 500m. 

Je cite les passages intéressants de la fiche [SéQuelec N°17 - La TIC des compteurs « Linky »](https://www.enedis.fr/media/3005/download): 
> Pour assurer son bon fonctionnement et le respect des caractéristiques électriques, la longueur maximale du
bus d’information ne doit pas dépasser les 500 m (topologie quelconque).

> Le câble de raccordement est un câble téléphonique intérieur de type :
> - paire torsadée simple avec écran aluminium et conducteur de drain,
> - conducteur monobrin en cuivre étamé de diamètre 0,5 mm,
> - isolant PVC.

("bus d’information" == câble ; "écran aluminium" == blindage de la paire torsadée ; "conducteur de drain"==le blindage possède un fil qui permet de le connecter à la terre (explications plus amples [ici](https://www.elandcables.com/fr/the-cable-lab/faqs/faq-what-are-drain-wires-in-electrical-cables))

En conclusion, une paire torsadée blindée, dont le blindage est mis à la terre, devrait fonctionner pour 50m. 
Et un cable RJ-45 SSTP garantira ces propriétés. 

Pour infos, comme indiqué dans la fiche elle même, cette fiche est en fait un résumé court (voire incomplet) de: 
 - [Sortie de télé-information client des appareils de comptage électronique utilisé par Enedis](http://www.enedis.fr/sites/default/files/Enedis-NOI-CPT_02E.pdf)
 - [Sortie de télé-information client des appareils de comptage Linky utilisé en généralisation par
Enedis](http://www.enedis.fr/sites/default/files/Enedis-NOI-CPT_54E.pdf)
dont je conseille la lecture (au moins en diagonale) pour clarifier certains éléments obscures de la fiches (notamment les définitions des tensions)

Il y a aussi une autre fiche SéQuelec qui peut valoir le détour : [SéQuelec N°16 - Le circuit de communication du branchement](https://www.enedis.fr/media/2224/download) qui parle plus spécialement d'EURIDIS. 

Ceci étant dit : ces documents donnent les spécifications attendues pour un appareil voué à être branché sur la sortie TIC. Mais rien ne dit que le Lixee est totalement conforme à la norme ; en particulier, je n'ai pas vu de marquage "Linky Ready" qui permettrait de garantir cette conformité...

Peut-être que @fairecasoimeme pourrait donner quelques réponses sur la possibilité de brancher le Lixee via une paire torsadée blindée ?

fairecasoimeme, 2023-08-22:
Bonjour,
le problème est surtout l'alimentation du ZLinky qui ne tiendra pas sur cette distance même avec un câble cat6 STP. D'ailleurs, aucun appareil ne peut être alimenté par le Linky sur cette distance.
Sur ce genre de distance, je recommande plutôt les modules TIC filaires alimentés.
La labellisation est assez rare sur les matériels en vente aujourd'hui. De plus la labellisation est un peu "hasbeen" car elle ne garantit absolument pas l'intégrité des données sur les ERLs communicant. 
Le ZLinky est en cours de labellisation "Linky ready" mais le hardware ne sera plus le même. Le hardware actuel du ZLinky ne permet pas de capter des signaux en dessous de 2v et ne permet pas le montage en série.
Le fait qu'il n'y ait pas la labellisation ne veut pas dire que l'objet ne soit pas conforme à la norme.

scalmaa, 2023-08-22:
Bonjour

Tout d'abord, merci beaucoup pour votre intérêt et vos contributions pour m'aider à résoudre ma difficulté.

La dernière réponse de fairecasoimeme m'embête un peu car pour partager l'information, j'ai passer un câble Ethernet Cat6 STP dans mes fourreaux et je peux constater le bon fonctionnement du Lixee. Le module Lixee est bien alimenté en permanence (diode bleu fixe) malgré la distance et semble fonctionner correctement.

Malgré tout, je constate quelques trous (heures manquantes) dans le tableau de bord Energie de HA. Je n'ai pas investigué.

Quelle conclusion peut-on en tirer si cela ne devrait pas fonctionner correctement ?

SilentT-FR, 2023-08-22:
Firmware et/ou [Errata](https://github.com/fairecasoimeme/Zlinky_TIC#errata-hardware) ?

scalmaa, 2023-08-23:
Errata :-)
Je contacte Lixee

scalmaa, 2023-09-11:
Bonjour. 
J'ai bien désormais un module avec le condensateur. Pour autant, j'ai toujours des heures absentes dans le module energie. Cela signifie-t-il qu'un changement avec le firmware limited est requis ?
Merci de vos réponses

SilentT-FR, 2023-09-11:
Tu est passé en v13?

scalmaa, 2023-09-11:
Bonsoir. Oui je suis au dernier niveau de version. Le tdb energie présente des "trous" (données easd01) alors que lorsque je consulte l'historique de la donnée CCASN il ne semble pas y avoir de données manquantes. J'aimerais pouvoir vérifier la continuité de collecte mais je cherche où je peux modifier la durée de collecte du fichier de logs