fairecasoimeme/Zlinky_TIC #202: Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>

cyber-g, 2023-08-20:
Hi,
I try to read values using the `bellows` (zigpy) command line. 

For information, I use a [Sonoff Zigbee 3.0 Dongle Plus-E](https://www.sonoff.be/a-69234890/accessoires/sonoff-zigbee-3-0-dongle-plus-e/#description) as coordinator (as shown in the table of this post: [ZBDongle-E](https://github.com/Koenkk/zigbee2mqtt/discussions/13373#discussion-4272840) , it is an EZSP ; hence the usage of `bellows`)

I base my command line on this post : [How to get values from a device?](https://github.com/zigpy/zigpy/issues/206#issuecomment-526828258)

Network forming and joining has been done with: 
```
bellows -d /dev/ttyACM0 -b 115200 form
bellows -d /dev/ttyACM0 -b 115200 permit
```
Please note: the arguments ` -d /dev/ttyACM0 -b 115200` are required for my Sonoff Dongle ; and the lixee "link" button has been pushed to reset the link during the permit join command, the blue LED of the lixee turns stable as expected. 

I use the following command in my case (node ID is obfuscated): 
```
bellows -d /dev/ttyACM0 -b 115200 zcl '00:15:xx:xx:xx:xx:xx:xx' 1 1794 read-attribute 0
```

which asks for : 
Endpoint : `1`
Cluster: `1794` (Metering)
Attribute: `0` (Index Base)

But the **command fails with the error**: 
```
Failed to deliver message: <EmberStatus.DELIVERY_FAILED: 102>
```

fairecasoimeme, 2023-08-21:
Hi, 
I don't know where is the issue but I confirm that ZLinky can respond to cluster 1794 and attribute 0

cyber-g, 2023-08-21:
Hi, thank you for your answer. 

I ran the above command **several times and sometimes it works** providing the following output:
```
0=7464438
```

I feel that the command line bellows application async operation does not support the communication mode of ZLinky. 

Just in case it would help for my further tests, is there any specific periodicity of the Zlinky transmission ?

fairecasoimeme, 2023-08-22:
hi, what does the error code mean ?
There is no specific periodicity for transmission.
This is normal that the number is every time larger than previous because this is an index.