fairecasoimeme/Zlinky_TIC #206: Flasher le firmware limited

adri123, 2023-09-14:
Bonjour, 
Je rencontre des difficultés pour flasher le firmware limited.
J'utilise un USB-TTL avec lequel j'ai déjà réussi à flasher des ESP 32.
Je branche le USB en maintenant le bouton flash appuyé et je vois COM3 qui s'ajoute dans mes périphérique. Je relache le bonton après quelques secondes, puis je lancer la commane suivante :
`C:\NXP\DK6ProductionFlashProgrammer>DK6Programmer -s COM3 -P 115200 -Y -p .\ZLinky_router_v13_limited.bin`

Puis j'obtiens:
`┌───────────────────────────────────── NXP DK6 Device Programmer (Build 2282) ──────────────────────────────────────┐│COM3                                                                                                               ││                                                                                                                   ││Error opening connection: Read error                                                                               ││                                                                                                                   ││───────────────────────────────────────────────────────────────────────────────────────────────────────────────────││                                                                                                                   ││                                                                                                                   ││                                                                                                                   ││                                                                                                                   ││                                                                                                                   ││                                                                                                                   ││                 `

J'ai vérifié tous les branchements, je ne comprends pas où je me trompe...
Merci pour votre aide :)

fairecasoimeme, 2023-09-15:
Bonjour,
avez vous essayé d'inverser RX et TX ?

adri123, 2023-09-16:
Tout bêtement , ... Merci !

closed 2023-09-16 by fairecasoimeme

JigSawFr, 2023-11-19:
J'avais également le même souci,
Finalement résolu par une notion de "timing";
- Branchement du TTL avec appui sur le bouton flash
- Relâchement du bouton
- Lancement de la commande de flash (LED GND rouge)
Le passage entre l'étape 2 à 3 doit être assez rapide, sinon j'obtiens également l'erreur de lecture.

Sinon pour les drivers du TTL: https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads
Le lien sur le site mentionné par le README est HS.