fairecasoimeme/Zlinky_TIC #208: active_power toujours à 0 - Zigbee2mqtt

AntoineLemaire, 2023-09-25:
Bonjour,
J'ai installé ma clef Lixee Zlkiny fraîchement reçue, elle est en version 13. Je l'ai configurée sur Home Assistant via Zigbee2MQTT en mode historique HCHP (après avoir vérifié sur l'écran du Linky).

[image]

Je reçois bien les index HC/HP ainsi que quelques autres infos, cependant le `active_power` est lui toujours à 0

```
{
    "active_enerfy_out_d01": 0,
    "active_enerfy_out_d02": 0,
    "active_enerfy_out_d03": 0,
    "active_enerfy_out_d04": 0,
    "active_power": 0,
    "active_power_max": 0,
    "active_power_ph_b": 0,
    "active_register_tier_delivered": "HP..",
    "apparent_power": 550,
    "available_power": 30,
    "average_rms_voltage_meas_period": 0,
    "current_date": "",
    "current_index_tarif": 0,
    "current_price": "",
    "current_summ_delivered": 0,
    "current_tarif": "HC..",
    "current_tier10_summ_delivered": 0,
    "current_tier1_summ_delivered": 7828.158,
    "current_tier2_summ_delivered": 7452.224,
    "current_tier3_summ_delivered": 0,
    "current_tier4_summ_delivered": 0,
    "current_tier5_summ_delivered": 0,
    "current_tier6_summ_delivered": 0,
    "current_tier7_summ_delivered": 0,
    "current_tier8_summ_delivered": 0,
    "current_tier9_summ_delivered": 0,
    "days_number_current_calendar": 0,
    "days_number_next_calendar": 0,
    "days_profile_current_calendar": "",
    "days_profile_next_calendar": "",
    "drawn_v_a_max_n1": 0,
    "linkquality": 255,
    "message1": "",
    "message2": "",
    "meter_serial_number": "******************",
    "relais": 0,
    "rms_current": 2,
    "rms_current_max": 90,
    "rms_voltage": 0,
    "schedule_h_p_h_c": 0,
    "site_id": "",
    "start_mobile_point1": 0,
    "start_mobile_point2": 0,
    "start_mobile_point3": 0,
    "status_register": "",
    "stop_mobile_point1": 0,
    "stop_mobile_point2": 0,
    "stop_mobile_point3": 0,
    "update": {
        "installed_version": 13,
        "latest_version": 13,
        "state": "idle"
    },
    "warn_d_p_s": 0
}
```

J'ai tenté :
- De cliquer sur "reconfigure" dans la liste des appareils Zigbee2MQTT
- De redémarrer Zigbee2MQTT
- De tout mettre sur "auto"

Mais rien n'y fait, j'ai toujours le `active_power` sur zéro

Une idée de ce qui pourrait être mal configuré ?

Merci à vous

fairecasoimeme, 2023-09-25:
Bonjour,

si "active_power" représente la puissance active, elle n'existe pas dans le Linky. En mode historique, il n'y a que la puissance apparente qui est disponible sur le Linky.
pour la puissance active on peut (je crois la calculer), j'ai trouvé ceci : https://github.com/fairecasoimeme/Zlinky_TIC/issues/125#issuecomment-1486882150

closed 2023-09-25 by fairecasoimeme

AntoineLemaire, 2023-09-25:
Ah je pensais qu'il s'agissait de la consomation actuelle car le  `apparent_power` ne me semblait pas réaliste, et est toujours en dizaine

fairecasoimeme, 2023-09-25:
Apparent power, c'est la puissance en temps réel en VA. Du coup, le cos Phi n'est pas intégré et si certains appareils ont un mauvais rendement, le VA ne correspond pas au W. Malheureusement, c'est ce que donne le Linky.
D'autre part, si vous souhaitez avoir plus d'informations et un apparent power qui ne soit pas en dizaine, il faut faire la demande auprès de votre fournisseur d'énergie de passer le Linky en mode standard.

1.	Contacter votre fournisseur d'électricité ;
2.	Demander le passage de votre compteur Linky en mode standard en précisant qu’il s’agit de la prestation F185 auprès d'Enedis.
3.	Et c'est terminé !
Cette opération est gratuite et en général, le changement est effectué dans les 24 heures ouvrées. Les effectifs étant souvent sous l’eau, il n’est pas rare que cela prenne quelques jours supplémentaires.