fairecasoimeme/Zlinky_TIC #209: Écrire une documentation mise à jour pour ZHA

SeraphicRav, 2023-10-04:
Bonjour,

J'ai cliqué sur le "tutoriel" pour ZHA et ça m'a envoyé sur un commentaire d'une issue de plus d'un an que j'ai, je dois l'avouer, du mal à déchiffrer.

Je pense qu'il faudrait faire une documentation à jour, un peu plus pédagogique et illustrée pour un meilleur onboarding des utilisateurs.

fairecasoimeme, 2023-10-04:
Bonjour,
Si vous êtes à jour sur ZHA, vous n'avez rien à faire si vous avez un abonnement "classique".
Les entités les plus courantes vont se créer automatiquement.
Le lien du turoriel permet d'aborder de manière avancée l'exploitation du Zlinky en ajoutant de nouvelles entités personnalisées.
Je vais modifier le README pour spécifier que le lien concerne un paramétrage avancé.

closed 2023-10-04 by fairecasoimeme

jarthod, 2023-12-14:
Bonjour,

Petite question sur le même sujet, je viens d'installer mon Linxee hier avec ZHA, je suis en abonnement "base", et j'ai les info suivantes qui sont remontés:
[image]

J'ai remarqué que:
- Le courant remonte mais arrondi à 1A près (limitation du linky?)
- La Puissance Apparente semble aussi dépendre du courrant arrondi car elle saute entre quelques valeurs significatives uniquement.
- La puissance en W reste à 0
- La tension reste à 0
- Le total en kWh semble OK

Est-ce que c'est le comportement attendu ? Si oui pourquoi les attributs sont crées quand même mais à zero ?

Ce sont probablement des information interessante pour la documentation qui est (je suis d'accord avec @SeraphicRav) un peu légère pour le moment ^^. Du coup on ne sais pas vraiment si ce qu'on voit c'est normal ou c'est un bug ou encore une mauvaise version. On ne sais pas non plus ce qu'il est possible de faire ou non avec une configuration avancé et si on en a besoin ou pas. (et encore je suis développeur mais je n'imagine meme pas un néophite dans ce thread)

Merci pour ce chouette produit tout de même :clap:

fairecasoimeme, 2023-12-15:
Bonjour,

* Le courant arrondi à 1A près --> limitation du Linky  
* la Puissance apparente est arrondi à 10 VA --> limitation du Linky en mode historique. Demandez le passage en mode standard à votre fournisseur d'électricité pour ne plus avoir cette limitation  
* la puissance en W n'existe pas dans le Linky
* la tension reste à 0 parce que vous n'êtes pas en mode standard

Ce comportement est attendu car il existe  2 modes possibles (historique et standard) avec le Linky et au moins 10 abonnements différents. De ce fait, les développeurs du plugin ont créés  ce qui leur semblait utile mais sans pouvoir prendre en compte le mode dans lequel vous êtes ou votre abonnement. Libre à vous de sélectionner les entités qui vous sont disponibles et utile.

Il vous faut plutôt consulter la documentation de la téléinformation Linky si vous voulez comprendre les valeurs.

jarthod, 2023-12-15:
Ok merci beaucoup pour ces information !