fairecasoimeme/Zlinky_TIC #210: Don't receive Tempo PTEC and many nulls (while OK few hours ago)

paramecie, 2023-10-13:
Khadas VIM1S, Home Assistant very latest version.
Zigbee2MQTT 1.33.1

**I received today our Lixee brand new from Lixee, installed, etc.**

No option changed (like Reports, Parameters or specific ones...)
Everything was working fine!!!

But then I've got maybe the DUMB idea to update it using OTA.
Now it's in version 13.

**2 hours later, many infos are missing, and the MOST important PTEC.**

Tried many thing, restarted HA, rebooted system, removed the device (forced), unplug LiXee, reinstalled, no way.
I checked, Linky shows everything normal including "HC BLEU" (for PTEC "HCJB").

See the current JSON with many null:
```
{
    "apparent_power": 1270,
    "current_summ_delivered": 0,
    "current_tarif": "BBR(",
    "current_tier1_summ_delivered": 21866,
    "current_tier2_summ_delivered": 29641,
    "current_tier3_summ_delivered": 0,
    "current_tier4_summ_delivered": 0,
    "current_tier5_summ_delivered": 0,
    "current_tier6_summ_delivered": 0,
    "linkquality": 123,
    "rms_current": 5,
    "start_notice_e_j_p": 0,
    "tomorrow_color": "----",
    "warn_d_p_s": 0,
    "MOTDETAT": null,
    "active_register_tier_delivered": null,
    "available_power": null,
    "meter_serial_number": null,
    "rms_current_max": null,
    "schedule_h_p_h_c": null,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null
}
```

Last log received with version 1:

`info  2023-10-12 19:26:46: MQTT publish: topic 'zigbee2mqtt/0x00158d0005d29c35', payload '{"MOTDETAT":null,"active_register_tier_delivered":"HPJB","apparent_power":2000,"available_power":45,"current_summ_delivered":51489,"current_tarif":"BBR(","current_tier1_summ_delivered":21855,"current_tier2_summ_delivered":29634,"current_tier3_summ_delivered":0,"current_tier4_summ_delivered":0,"current_tier5_summ_delivered":0,"current_tier6_summ_delivered":0,"linkquality":48,"meter_serial_number":"031864700077","rms_current":9,"rms_current_max":90,"schedule_h_p_h_c":0,"start_notice_e_j_p":0,"tomorrow_color":"----","update":{"installed_version":1,"latest_version":13,"progress":100,"remaining":20,"state":"updating"},"update_available":null,"warn_d_p_s":0}'`

One of the last logs recived with version 13:

`info  2023-10-12 21:31:57: MQTT publish: topic 'zigbee2mqtt/LiXee', payload '{"MOTDETAT":null,"active_register_tier_delivered":"HPJB","apparent_power":2000,"available_power":45,"current_summ_delivered":51489,"current_tarif":"BBR(","current_tier1_summ_delivered":21855,"current_tier2_summ_delivered":29634,"current_tier3_summ_delivered":0,"current_tier4_summ_delivered":0,"current_tier5_summ_delivered":0,"current_tier6_summ_delivered":0,"linkquality":132,"meter_serial_number":"031864700077","rms_current":9,"rms_current_max":90,"schedule_h_p_h_c":0,"start_notice_e_j_p":0,"tomorrow_color":"----","update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null,"warn_d_p_s":0}'`

Few minutes later, version 13, no more "active_register_tier_delivered", many nulls:

`info  2023-10-12 21:36:35: MQTT publish: topic 'zigbee2mqtt/LiXee', payload '{"MOTDETAT":null,"active_register_tier_delivered":null,"apparent_power":1450,"available_power":null,"current_summ_delivered":null,"current_tarif":"BBR(","current_tier1_summ_delivered":null,"current_tier2_summ_delivered":null,"current_tier3_summ_delivered":null,"current_tier4_summ_delivered":null,"current_tier5_summ_delivered":null,"current_tier6_summ_delivered":null,"linkquality":123,"meter_serial_number":null,"rms_current":null,"rms_current_max":null,"schedule_h_p_h_c":null,"start_notice_e_j_p":null,"tomorrow_color":null,"update":{"installed_version":13,"latest_version":13,"state":"idle"},"update_available":null,"warn_d_p_s":null}'`

What can I do?

It is soooooo important for us to know the EDF Tempo color - I bought Lixee for this (also).
Thank you!

PS: Français, désolé j'avais commencé à taper en anglais. Un énorme merci pour ce que vous faites !

fairecasoimeme, 2023-10-13:
Bonjour,

Après une mise à jour sur Z2M, il faut souvent redémarrer Z2M et réappairer le ZLinky.

paramecie, 2023-10-13:
Wow - ça a marché, enfin pour l'instant, je confirmerai plus tard ;-)

Manip complète

**Arrêt:**
* débrancher le lixee du compteur
* dé-appairer le lixee dans HA en forçant
* arrêt propre HA et du host
* débrancher la clef USB zigbee

**Redémarrage:**
* brancher la clef zigbee
* redémarrer host et HA
* HA, stats, supprimer les résidus de stats de l'ancien lixee
* lancer appairage
* rebrancher lixee, appuyer sur bouton appairage.

Bon, je confirmerai que ça tient - merci encore !

llaumgui, 2023-10-20:
On est OK qu'avec le firmware 4000-0013 en date du 20230214 c'est bon ? Car j'ai encore ce message d'erreur sur z2mqtt / Jeedom.

paramecie, 2023-11-04:
@llaumgui J'ai le 4000-0013 mais sur Zigbee2MQTT / Home Assistant en tous cas depuis 3 semaines, ça fonctionne impeccablement. Je ne connais pas Jeedom, pas testé, désolé.

Silverrider5, 2023-11-15:
j'ai le même problème, ça marche pendant plusieurs heures puis tout à coup plus rien les infos ne remontent plus. obligé de ré-appairer pour que ça refonctionne.

Silverrider5, 2023-11-21:
j'ai le même problème, ça marche pendant plusieurs heures puis tout à coup plus rien les infos ne remontent plus. obligé de ré-appairer pour que ça refonctionne. Help !!!