fairecasoimeme/Zlinky_TIC #215: EASD03 / EASD04

technoman63, 2023-11-02:
Bonjour,

Je cherche désespérément a faire remonté les deux valeurs:  EASD03 / EASD04

Je suis sur Zigbee2MQTT 1.33.2
Je suis en HC/HP et j'ai besoin de EASD03 / EASD04 pour les HP/HC saison haute

Si vous avez une petite piste pour m'aider ?

py-g, 2023-11-24:
Bonjour,

C'est un sujet qui concerne autant zigbee2mqtt que la ZlinkyTIC, pas facile de trouver l'endroit idéal pour en causer.
Je viens de regarder mais je ne suis pas en HP/HC comme toi. Donc si ça se trouve il y a plus simple...
Voici comment j'ai fait:

## Faire remonter l'information de la ZLinky TIC
### Ajouter des rapports pour les valeurs concernées
Comme nous l'a généreusement partagé @fairecasoimeme dans sa [doc exhaustive](https://github.com/fairecasoimeme/Zlinky_TIC), ces informations sont disponibles sur les clusters/attributs suivants:

| Commande TIC | CLUSTER | Attribut | Droit | data type | size max | unit | designation |
| -----------------------  | ----------- | ----------- | ------------- | ------------- | -------------- | --------- | --------------- |  
| EASD03 | 0xFF66 | 0x0205 | RP | Uint32 | 9 car | Wh | Energie active soutirée Distributeur, index 03 |
| EASD04 | 0xFF66 | 0x0206 | RP | Uint32 | 9 car | Wh | Energie active soutirée Distributeur, index 04 |

Il faut donc demander à Zigbee2MQTT de les faire remonter en MQTT pour en disposer dans Home Assistant.
Pour ce faire, un petit coup d’œil [dans le code](https://github.com/Koenkk/zigbee-herdsman-converters/blob/0fd3f35ff5627ea80acd236c3061fe20b8f61144/src/devices/lixee.ts#L146) des [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters/tree/0fd3f35ff5627ea80acd236c3061fe20b8f61144) nous permet de tomber sur ces lignes : 
```
    lixee_private_fz: {
        cluster: 'liXeePrivate', // 0xFF66

                (...)

                /* 0x0203 */ 'activeEnerfyOutD01',
                /* 0x0204 */ 'activeEnerfyOutD02',
                /* 0x0205 */ 'activeEnerfyOutD03',
                /* 0x0206 */ 'activeEnerfyOutD04',
```
Tu retrouves ici le cluster **0xFF66** (liXeePrivate) et les attributs **0x0205** et **0x0206** associés aux étiquettes TIC **EASD03** et **EASD04**.
Par la même occasion, on en profite pour noter que pour z2m, ces valeurs s'appellent "**activeEnerfyOutD03**" et "**activeEnerfyOutD04**" (en passant, il semble que: soit l'auteur a un feveux sur la langue, soit la touche F du clavier se trouve à côté de la touche G, et en traduction simultanée on peut comprendre activeEner**g**yOutD03 !)

Armé de ces précieuses informations, il est maintenant possible de configurer des rapports (car ces attributs sont "Reportables" comme l'indique "RP" dans la colonne "Droit" du tableau). C'est à dire que la ZLinky TIC enverra d'elle même ces valeurs à zigbee2mqtt.

Pour ça, il faut aller dans la configuration de zigbee2mqtt, sélectionner la ZLinky TIC, et cliquer sur l'onglet "Rapports" : 

[image]

Tout en bas, il faut ensuite ajouter deux lignes pour configurer des rapports pour les deux valeurs qui t'intéressent : 

[image]

Il faut commencer par sélectionner le **point de terminaison** : **1**
Ensuite tu sélectionnes le **groupe de fonctionnalités** (cluster) : **liXeePrivate** (qui correspont à 0xFF66)
Et enfin l'**attribut**: **activeEnerfyOutD03** (qui correspond à 0x0205, soit EASD03 en language TIC)

Les valeurs "Interval min réponse", "Interval max réponse" et "Changement min réponse" portent bien leurs noms:
- **Interval min réponse** est le temps minimum entre deux envois de rapports
- **Interval max réponse** est le temps maximum entre deux envois de rapports 
- **Changement min réponse** est le changement minimum de la valeur qui provoque l'envoi du rapport (pour éviter d'envoyer trop de rapports sur les valeurs qui changent souvent)

Tu cliques sur "Appliquer" et z2m devrait te dire : 
[image]

Pour la deuxième valeur, c'est pareil mais tu choisis l'attribut **activeEnerfyOutD04** (EASD04).

## Vérifier que l'information remonte correctement
### MQTT Explorer
Pour vérifier que les infos remontent, je ne saurais que trop te conseiller d'utiliser l’excellent [MQTT Explorer](https://mqtt-explorer.com/).
Après une (toute) petite attente, tu devrais voir apparaître les nouvelles valeurs dans le JSON envoyé en MQTT:

[image]

## Ajouter un capteur dans Home Assistant
### Edition du fichier de configuration

Pour ajouter le capteur d'énergie à Home Assistant, j'ai édité le fichier de configuration **configuration.yaml**.

Si tu ne l'as pas, ajoute le module complémentaire "File Editor" : 

[image]
[... 23 lines cut ...]
Tu n'as plus qu'à faire la même chose pour **EASD04**. Un petit copier/coller à partir de **- name: "ZLink....** et tu ajoutes le deuxième capteur pour EASD04 en faisant les modifications nécessaires.

## Vérification de la configuration !
Après tout changement dans le fichier de configuration, il est fortement conseillé de vérifier que le fichier de config est valide. Tu peux le faire dans les outils de développement, en cliquant sur le bouton "Vérifier la configuration" : 

[image]

Home Assistant devrait te dire ça : 

[image]

## Chargement de la configuration

Ensuite, un peu plus bas dans la même page, il faut recharger les entités MQTT configurées manuellement : 

[image]

## Un nouveau capteur d'énergie est disponible

Tu vas ensuite dans "**Paramètres / Appareils et Services / (onglet) Entités**" et tu devrais trouver ton nouveau capteur (enfin... les deux normalement mais j'en ai fait un seul pour tester)

[image]

Voilà. Un peu long à expliquer mais plutôt rapide à mettre en place.
En espérant que ça aide :wink:

technoman63, 2023-11-28:
Merci pour ton retour, je vais essayé tout cela quand j'aurais un peu de temps