fairecasoimeme/Zlinky_TIC #216: Zigbee2mqtt 1.33

algernon2607, 2023-11-04:
Bonjour

Depuis la mise à jour de zigbee2mqtt 1.33 , l'intervalle de réception des infos est trop élevé (10 mn pour certains éléments).
Le changement de tarif est remonte parfois avec 10 mn de retard . Les infos d’énergie sont trop parcellaires et j'ai une dérive par rapport à la réalité (alors qu'avant c’était nickel) . Quand aux mesures de puissance , c'est devenu inexploitable .

J'ai bien essayé de modifier les polls ainsi que les intervalles mais rien n'y fait . Le z_linky est devenu inutile.
Avez une solution autre que de restaurer zigbee2mqtt 1.32  ??

al

 Core 2023.11.0
Supervisor 2023.10.1
Operating System 11.1
Interface utilisateur 20231030.1

fairecasoimeme, 2023-11-06:
avez-vous lu cette issue : https://github.com/fairecasoimeme/Zlinky_TIC/issues/194 ?

algernon2607, 2023-11-06:
Bonjour
Oui j’ai lu cette issue et dans mon cas, j'y avais pallié avec la whitelist et avec le filtrage .
Mais bon tant pis ....!
Je vais trouver autre chose ...
Merci en tous cas de votre réponse ..

closed 2023-11-06 by algernon2607