fairecasoimeme/Zlinky_TIC #219: TEMPO et HomeAssistant, bascule HC HP ne se fait pas plus a 22h ni 6h

kefrens, 2023-11-22:
Bonjour,

Depuis la nuit du 15 au 16 Novembre la bascule des HC / HP ne se fait plus a 22h ni a 6h mais vers 1h30 du matin jusque 9h30..
Est-ce normal?
J'ai pourtant bien un forfait tempo et je viens de verifier sur mon compte c'est bien 22h / 6h.

Est-ce que d'autres sont dans le meme ca?
Je n'ai pas touche a la config, mais HA derniere version est installe tout comme zlinky v13. (tout est configure en auto, mais je vois que c'est le mode standard qui a ete choisi)

py-g, 2023-11-23:
Bonjour,
Je suis aussi en Tempo et je ne rencontre pas ce problème.
Effectivement, en principe c'est 22h/6h pour tous les Tempo, mais les horaires des heures creuses sont définis par Enedis. Ils dépendent de la gestion du réseau électrique local et peuvent être modifiés pour des raisons de gestion de la demande en électricité.

kefrens, 2023-11-23:
apres verification mon linky n'affiche pas la bonne heure ... avec exactement 3h30 de decalage... le service client d'enedis a l'air peu perdu.. visiblement il faut que quelqu'un vienne sur place pour regarder

fairecasoimeme, 2023-11-23:
Si le Linky n'est pas à la bonne heure, c'est qu'il n'arrive pas à communiquer avec les serveurs Enedis.
Le produit aura permis de déceler le problème ;)

closed 2023-11-23 by fairecasoimeme