fairecasoimeme/Zlinky_TIC #220: Alimentation externe

OlivierCazade, 2023-11-23:
Bonjour,

Mon Linky est assez éloigné de ma maison (environ 50m) et difficile de passer un nouveau cable de telereport. Je peux par contre reutiliser les cables existant de contact sec pour le chauffe eau pour remonter a la place i1 et i2 au niveau de mon coffret elec.

J'ai vu que le TIC-DIN vendu aussi sur Lixee n'a besoin que de ces deux entrées parce qu'il prend son alimentation sur le port USB.

Je prefererai utiliser le Zlinky car ma box domotique n'est pas pret du coffret et j'ai deja un reseau Zigbee.

Est-il possible d'alimenter differement le Zlinky pour supporter ce genre de montage?

starox, 2025-03-28:
> Bonjour, Je suis exactement dans la même situation que celle décrite par Olivier Cazade.
> 
>     * Mon Linky est en limite de propriété trop loin de ma maison pour une liaison zigbee

Peut-être un répéteur zigbee ? Mon zlinky est à 10m de la maison. Au début, j'avais une mauvaise liaison avec le coordinateur, mais à force d'ajouter des appareils sigbee qui font office de répéteur, je n'ai plus eu de problème.
Il faut que je vérifie par quel appareil il passe maintenant.

Mon garage  éloigné de la maison de 30m à des appareils en zigbee qui sont bien connecté. J'ai mis une clef sonoff en mode routeur pour faire la liaison avec la maison, mais tous ne l'utilise pas.

>     * Le câblage de la TIC et du contact sec issus de mon Linky arrivent dans le tableau électrique dans la maison ce qui semble idéal
> 
>     * mais le TIC DIN nécessite d’être connecté en USB à la box pour l’alimentation et l’échange de données ce qui implique d’avoir la box proche du tableau ou d’acheter une box rackable fort chère
> 
> 
> L’idéal serait effectivement un TIC DIN ZIGBEE avec une alimentation AC et intégrant un adapteur d’alimentation ac/dc 3,3v. Non ?
> 
> Sinon la solution des wagos et des soudures …
> 
> Suis preneur de toute autre idée Cordialement Thierry

Je ne comprends pas bien cette histoire de TIC (Je ne comprends pas l'acronyme). Le zlinky est connecté en  filaire à l'intérieur du bâtiment sur le tableau électrique, c'est ça ? Il faut une alimentation externe pour le zlinky ?

ThierrydeBiot, 2025-03-28:
> Peut-être un répéteur zigbee ? Mon zlinky est à 10m de la maison. Au début, j'avais une mauvaise liaison avec le coordinateur, mais à force d'ajouter des appareils sigbee qui font office de [...]

Dans mon cas, le Linky est situé à 25m de la première prise où je pourrais installer un répéteur mais c’est aussi à 4 ou 5m de dénivelé en contrebas d’un talus.  Je ne vois donc  pas bien comment placer un répéteur .

> Je ne comprends pas bien cette histoire de TIC (Je ne comprends pas l'acronyme). Le zlinky est connecté en filaire à l'intérieur du bâtiment sur le tableau électrique, c'est ça ? Il faut une alimentation externe pour le zlinky ?

TIC : Télé Information Client. 
Ce que je comprends : 
Le produit TIC DIN est un appareil qui se contente de décoder le flux de données de la télé information client du compteur Linky. Il reçoit ces informations du Linky via une paire de câble, mais il a  besoin d’une alimentation pour ce faire et aussi d’une liaison série pour les communiquer à la box. C’est ce que permet le cable usb entre le TIC DIN et la box domotique. Mais cela implique une proximité avec la box domotique 

Quand on l’utilise de façon normale, l’alimentation  du ZLinky est quant à elle fourni par le compteur Linky lui même.  
Mais alors que le signal portant la TIC supporte
de grande longueur de câble, l’alimentation de très basse tension fournie par le Linky ne supporte pas des grandes longueurs de câbles (manifestement certains témoignages montrent que cela a déjà été essayé). 
On ne peut donc pas déporter le zlinky en utilisant l’alimentation fournie par le Linky.  Reste donc à l’alimenter par autre chose. 
Mais il faut bidouiller …

starox, 2025-03-28:
C'est bon, j'ai compris ! Voilà ce que j'en pense:

_1) Déporter le zlinky dans la maison_

résolution: 
- amener des câbles du compteur linky à un endroit dans le bâtiment
- alimenter le zlinky avec une alim externe, nécessite de souder ...
- trouver l'alim ...

J'y vois un inconvénient, vu la longueur des câbles, je me demande si le signal sera bon et l'éléphant dans la maison, tirer les câbles.

_2) utiliser un linky tic sachant que la box n'est pas à côté du compteur_

résolution: 
- amener les câbles
- déporter le port usb via réseau rj45/wifi

J'ai plusieurs idée pour déporter le port usb mais ça nécessite d'investir (mini-pc  ou autre appareil à hacker qui a un port usb et une carte wifi, détournement de serveur d'impression wifi, tirer un long câble usb, etc ...)

Inconvénient, nécessite de tirer des câbles et de passer du temps à mettre en place.

_3)  répéter le signal zigbee_

résolution:
- utiliser une grande rallonge électrique pour mettre entre le bâtiment et le compteur zlinky
- connecter un répéteur sur la rallonge.

Inconvénient, nécessite de protéger le tout des intempéries.

Ma préférence va pour le 3. Eventuellement peut de transformer en borne électrique au milieu du terrain pour les guirlandes de noël ou autre ...
 
Mon ordre de préf (mais c'est subjectif): 3 - 1 - 2

ThierrydeBiot, 2025-03-29:
Merci beaucoup d’avoir pris le temps de réfléchir à mon problème. En ce qui concerne les options :

1/ la longueur du câblage du bus TIC ne pose pas de problème. S’agissant d’une liaison à très bas débit Enedis indique qu’on peut aller jusqu’à 500m sur une paire téléphonique. Trouver l’alim pour qcq euros n’est pas un problème. Reste effectivement deux soudures à faire et bricoler le casing de l’ensemble pour faire propre. 

2/j’avoue que je ne connais pas de solution permettant à la fois de fournir l’alimentation et de « déporter  le port usb » par voie radio jusqu’à la box domotique. S’il faut tirer un long câble usb depuis le tableau électrique, ce n’est pas raisonnable compte tenu des distances et de l’architecture de la maison. 

3/ Plutôt que de tirer une ligne électrique pour alimenter un répéteur à l’extérieur,  avec un répéteur zigbee utilisable en extérieur et alimenté  par une petite batterie chargée par un petit  panneau solaire je pourrais éventuellement imaginer l’installer en haut du talus pour relayer les flux jusqu’à la maison. Mais bon, tout ça pour ça !! 
Pour ma part je privilégie 1,3 et 2. Mais j’ajoute désormais une alternative  existante construite autour d’un autre produit qui combine le décodage du bus TIC et l’émission vers la box en wifi. Resterait juste à l’alimenter en 12V ce qui n’est pas un pb.

starox, 2025-03-29:
Je comprends :-)
L'idée d'avoir un répéter zigbee extérieure alimentée en solaire avec batterie est excellente.
C'est bizarre que ça ne soit pas déjà trouvable en commerce.