fairecasoimeme/Zlinky_TIC #221: Surr Tempo: HCHP n'est pas le total the HP ni est HCHC le total HC, c'est que jours blanc

vingerha, 2023-12-01:
Voir dessus
HCHP/HCHC manquent les données des jours white/red

[image]

vingerha, 2023-12-04:
La question c'est si HCHP ou HCHC deviennent du Linky ou Lizee ou de l'intégration

fairecasoimeme, 2023-12-05:
Pour mieux comprendre les étiquettes, il faut lire le [readme](https://github.com/fairecasoimeme/Zlinky_TIC#subscription-optarif-values)

Un couple cluster / attribut peut correspondre à plusieurs abonnement.

Dans votre cas les étiquettes remontée par Z2M : BBRHCJB / HCHC et EJPHN correspondent à la même valeur : 0x0702 | 0x0100
Je pense que vous devez sélectionner votre abonnement dans Z2M pour que ce soit plus clair.

vingerha, 2023-12-05:
Z2M est sur Standard - Tempo ce qui n'est pas dans la docu
BBRHPJW change HCHP pas...des suggestions bienvenu :)
[image]

fairecasoimeme, 2023-12-05:
Normal HCHP n'existe pas sur un abonnement tempo

vingerha, 2023-12-05:
OK... donc je l'ignore ...c'est bon a savoir, merci

closed 2023-12-05 by vingerha