fairecasoimeme/Zlinky_TIC #222: Consommations Tempo Jour Rouge non remontées

adrienball, 2023-12-04:
J'utilise le zlinky_tic avec une passerelle Conbee II, j'ai récemment basculé sur l'option Tempo et autant les consommations des jours bleus et blancs (heures pleines et heures creuses) ne posent pas de soucis, autant les consommations des jours rouges ne sont pas correctement remontées.
Initialement je pensais que le problème provenait du plugin deCONZ (lié a la passerelle Conbee II), mais après avoir pas mal creusé et étant donné que la configuration (fichier DDF, etc) est identique pour les autres couleurs de jours, je commence à me demander si le problème n'est pas du côté du zlinky_tic.
Plus précisément, sur l'interface du plugin deCONZ, les attributs 0x0108 et 0x010A (HC jour rouge, HP jour rouge) sont correctement lus lorsqu'on demande une lecture manuelle des attributes, mais l'API REST exposée par le plugin ne renvoie pas de valeurs, et se comporte donc comme si elle ne recevait jamais de mise a jour pour ces attributs.

[image]

@fairecasoimeme est-il possible qu'il y ait une spécificité liée au jour rouge, côté zlinky_tic, qui pourrait expliquer ce problème ?
Est-il utile de reboot le zlinky_tic suite au passage en option tempo ? Etant donné que les autres jours fonctionnent correctement, j'en ai déduit que ce n'était pas nécessaire.
Je suis arrivé à bout de tous mes efforts de debugging, en vain, et c'est très frustrant d'avoir un dashboard grafana sur lequel il manque juste les jours rouges 😬 
Merci pour votre aide!

fairecasoimeme, 2023-12-05:
Bonjour,
Si les valeur 0x0108 et 0x010A sont correctement remontée sur deConz, et que les valeurs ne le sont pas sur le plugin, c'est : 
* Soit un problème de DDF ou de l'API ou du plugin
* soit le config report qui n'a pas été bien configuré.

adrienball, 2023-12-05:
Voici les bindings que j'ai défini dans mon fichier DDF pour l'option tempo:
```
{
  "bindings": [
    {
      "bind": "unicast",
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "change": "0x00000001",
          "dt": "0x23",
          "max": 300,
          "min": 60
        },
        {
          "at": "0x0001",
          "change": "0x00000001",
          "dt": "0x23",
          "max": 300,
          "min": 60
        },
        {
          "at": "0x0100",
          "change": "0x00000001",
          "dt": "0x23",
          "max": 300,
          "min": 60
        },
        {
          "at": "0x0102",
          "change": "0x00000001",
          "dt": "0x23",
          "max": 300,
          "min": 60
        },
        {
          "at": "0x0104",
          "change": "0x00000001",
          "dt": "0x23",
          "max": 300,
          "min": 60
        },
        {
          "at": "0x0106",
          "change": "0x00000001",
          "dt": "0x23",
          "max": 300,
          "min": 60
        },
        {
          "at": "0x0108",
          "change": "0x00000001",
          "dt": "0x23",
          "max": 300,
          "min": 60
        },
        {
          "at": "0x010A",
          "change": "0x00000001",
          "dt": "0x23",
          "max": 300,
          "min": 60
        }
      ],
      "src.ep": 1
    },
    {
      "bind": "unicast",
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "change": "0x00000001",
          "dt": "0x21",
          "max": 300,
          "min": 60
        },
        {
          "at": "0x0508",
          "change": "0x00000001",
          "dt": "0x21",
          "max": 300,
          "min": 60
        },
        {
          "at": "0x050F",
          "change": "0x00000001",
          "dt": "0x21",
          "max": 300,
          "min": 60
        }
      ],
      "src.ep": 1
    },
    {
      "bind": "unicast",
      "cl": "0xFF66",
      "report": [
        {
          "at": "0x0005",
          "change": "0x00000001",
          "dt": "0x21",
          "max": 300,
          "min": 60
        }
      ],
      "src.ep": 1
    }
  ]
}
```

Cela se traduit de cette façon dans le plugin deCONZ (pour les deux derniers attributs correspondant a Tempo Rouge):

[image]

Ai-je raté quelque chose ?

adrienball, 2023-12-09:
> Pouvez-vous faire un config report manuel sur ces deux valeurs et avoir les statut de retour ?

Je ne suis pas sur de savoir ce que ça veut dire, et comment faire ça, pouvez-vous m'éclairer un peu ?
Merci!

adrienball, 2023-12-09:
J'ai modifié les bindings car le type spécifié pour les différents index était u32 alors que la doc du zlinkty_tic spécifie du u48, je ne suis pas sur que ça règle mon problème car les index de jours bleus et blancs fonctionnaient correctement malgré ça.
Par ailleurs, lorsque je lis la report configuration des attributs 0x0108 et 0x010A, tout semble normal:
[image]

adrienball, 2023-12-10:
Bon j'ai fini par désappairer / ré-appairer le zlinky_tic et tous les attributs sont désormais remontés correctement.
Merci pour votre aide en tout cas

closed 2023-12-10 by adrienball