fairecasoimeme/Zlinky_TIC #227: ZigbeeMQTT ZLINKY / updates available again and again

Dites33, 2023-12-10:
Hello Everyone

I have a small question about updates.
HomeAssistant / Zigbee2MQTT ask me to update my Zlinky again and again. I tried more than 5 times ...

[image]

when i run it from here, the installer works, end fine, everything is ok and Zlinky is offline at the end of the update process (seems normal)
I need to remove it and to install it again.
Then Home Assistant ask me to update it again ....
Same with the Zigbee2MQTT interface

[image]

Here is the info in Zigbee2MQTT

`{
    "apparent_power": 0,
    "current_summ_delivered": 59977,
    "current_tarif": "BASE",
    "linkquality": 102,
    "rms_current": 4,
    "status_register": "003AC301",
    "update": {
        "installed_version": 1,
        "latest_version": 13,
        "state": "available"
    },
    "warn_d_p_s": 0,
    "MOTDETAT": null,
    "active_register_tier_delivered": null,
    "available_power": null,
    "device": {
        "applicationVersion": 12,
        "dateCode": "20230214",
        "friendlyName": "0x00158d0005d29a3b",
        "hardwareVersion": 1,
        "ieeeAddr": "0x00158d0005d29a3b",
        "manufacturerID": 4151,
        "manufacturerName": "LiXee",
        "model": "ZLinky_TIC",
        "networkAddress": 60611,
        "powerSource": "Mains (single phase)",
        "softwareBuildID": "4000-0013",
        "stackVersion": 2,
        "type": "Router",
        "zclVersion": 3
    },
    "meter_serial_number": null,
    "rms_current_max": null,
    "update_available": null
}`

How can i do a full update ?

thanks a lot

sanvalerius, 2024-01-05:
FYI I have the exact same situation.
Looking for a solution as well, tks

Dites33, 2024-01-05:
For your information, I got an answer through the manufacturer's contact form. His answer is that for updates, it would seem that Z2M is not accessing the right value for the version.
All Zlinkys since Mars are on the latest version. Z2M's update request should be ignored
Maybe we should open a Z2M Ticket ?

fairecasoimeme, 2024-01-08:
please ignore the update request.
I'll try to fix for the next firmware version.