fairecasoimeme/Zlinky_TIC #229: z2m: warn dps

dinde, 2023-12-30:
Hello,

Petite question, je suis passé en mode tic standard.
Adps n'existe plus disponible qu'en mode historique.
J'ai vu dans les états de z2m warn_d_p_s mais il n'y a rien d'exposé et donc aucune entité dans home assistant.

Comment la faire remonter ?

Sinon j'ai remarqué dans les états une typo:
active_enerfy_out_d01
J'imagine que c'est energy.

Merci !

Home assistant + z2m + zlinky sur compteur monophasé et mode tic standard (tout est à jour)

fairecasoimeme, 2024-01-02:
Bonjour,
En mode standard, l'alarme de dépassement de puissance se trouve dans l'étiquette STGE. cluster : 0xFF66 attribut : 0x0217
Cette valeur en hexadécimal est décrite dans le document technique d'Enedis : Enedis-NOI-CPT_54E.pdf (page 24)
le bit 7 de la valeur correspond au dépassement de seuil. Je ne connais pas bien Z2M et si c'est possible mais il faudrait faire un script en masquant ce bit pour connaitre l'état du dépassement de seuil.
Cette information est mise à jour dans Z2M sur chaque modification de valeur

dinde, 2024-01-25:
Hello,
Ok je vais voir pour faire une automatisation directement sur le superviseur en surveillant le changement du bit 7 de ce cluster afin de recevoir une notification voir désactiver des équipements le cas échéant.
Merci pour la réponse en tout cas !

closed 2024-01-25 by dinde