fairecasoimeme/Zlinky_TIC #230: Passage chez Octopus Energy, depuis plus rien

gsemet, 2024-01-01:
Bonjour,
Le 16 decembre je suis passé de Total Energie vers Octopus Energy, et depuis le zlinky ne remonte plus aucune donnée. Tous les compteurs semblent bloqués dans zigbee2mqtt. Le zlinky est avec sa led bleue continue et il arrive à communiquer avec le réseau.

Qu'est qu'il faut demander à octopus enegie à réactiver sur le linky? J'étais en mode standard que j'avais demandé chez Total Energie et j'avais bien tous mes compteurs de consommation (base) et de production (revente).

Merci

florianbelhomme, 2024-01-01:
A tout hasard, une opération que vous pouvez tester c'est :
- supprimer & bannir le Zlinky de Z2M
- débrancher le Zlinky du Zlinky
- attendre 2 minutes
- le rebrancher
- le débannir de Z2M
- lancer une nouvelle association

Et en laissant les paramètres auto dans un premier temps, voir si il remonte des infos ?

tadikwa, 2024-01-04:
> A tout hasard, une opération que vous pouvez tester c'est :
>
> * supprimer & bannir le Zlinky de Z2M
> * débrancher le Zlinky du Zlinky
> * attendre 2 minutes
> * le rebrancher
> * le débannir de Z2M
> * lancer [...]

Je confirme cette méthode fonctionne.

J'ai eu le même problème et grâce aux conseil de @florianbelhomme j'ai résolu cela.

Merci