fairecasoimeme/Zlinky_TIC #232: MAJ Non OTA sur nouvel hardware sans connecteur noir

tadikwa, 2024-01-04:
Hello,
je me permet d'ouvrir un ticket pour demander une précision par rapport à la mise à jour non ota sur le nouveau dongle.

En effet, depuis que je l'ai, il est bloqué en v1 (d'après mqtt sur home asssitant) et je souhaitai le faire via le module usb ttl.
Sauf que, j'ai pas vraiment envie de me lancer dans de la soudure, les connecteurs étant juste des trous, auriez vous une solution simple pour faire le contact provisoirement sans soudure pour faire la maj v13 ?

JigSawFr, 2024-01-05:
Personnellement j'ai utilisé ça: Hailege 2pcs CP2102 USB 2.0 to TTL UART Serial Converter Module 5Pin STC PRGMR with Dupont Wire https://amzn.eu/d/2ySRu3R

Le tout en maintenant les broches dans les trous ;) sans soudures !

florianbelhomme, 2024-01-05:
Attention Z2M/MQTT peuvent être menteur sur la version du firmware si vous aviez déjà associé le Zlinky par le passé.
Si vous avez effectué la MAJ du firmware et que vous avez eu un message de succès sur le logiciel de MAJ, mais qu'après ré-association vous voyez toujours le numéro de firmware précédent dans HA, voici une manip que vous pouvez tester pour être sûr :

- une fois le Zlinky associé
- sélectionnez le dans la liste des appareils
- cliquez sur l'onglet "console dev"
- dans "groupe de fonctionnalités" sélectionnez "genBasic"
- puis dans "attribut" sélectionnez "swBuildId"
- enfin cliquez sur le bouton vert "lire"

Z2M va demander en direct le numéro de version au Zlinky, elle va s'afficher en JSON en dessous du bouton. Comparez avec ce que vous attendiez puis avec ce que Z2M affiche dans les pages "user friendly".

CF : https://github.com/fairecasoimeme/ZiPulses/issues/21#issuecomment-1793840520

tadikwa, 2024-01-05:
J'ai le même module. Cependant, c'est difficile d'obtenir un contact ok.
Je voyais sur l'autre ticket du ZiPulse, des gens galéré comme moi : https://github.com/fairecasoimeme/ZiPulses/issues/21

Je trouve compliqué d'arrivé à brancher la clef en tenant le module de travers pour faire contact tout en appuyant sur le bouton flash puis le relâchant en maintenant la clef et le module dans leur position.  J'ai juste le droit à un sapin de noël avec les voyants.

Savez qu'elle devrais être l'état des voyants du USBTTL au branchement ?

fairecasoimeme, 2024-01-05:
Il est très probable que votre ZLinky soit déjà à jour. le plugin Z2M ne va pas chercher la bonne valeur de version au bon endroit.
Pour avoir la bonne version de votre appareil, il faut interroger manuellement le cluster : 0x0000 attribut : 0x4000. Si la valeur retournée est 4000-0013 c'est que vous êtes à jour. (ce qui est surement le cas car ça va bientôt faire un an que cette version a été release)

tadikwa, 2024-01-05:
[image]
[image]

j'ai testé avec mes p'tites rallonges mais toujours pareil, soit read error ou rien.

Si je branche la clef avec le branchement et le bouton flash appuyer alors j'ai aucun voyant.

Si je branche la clef sans appuyer sur flash, alors j'ai le tx en fixe et le RX qui fini par clignoter au bout de 5 secondes et tjrs le mm résultat read error au flash

tadikwa, 2024-01-05:
Bon en effet si je lis via Z2M, le firmware semble déjà à jour.

[image]

JigSawFr, 2024-01-05:
J'avais du inverser le sens du branchement sur la mienne et j'ai utilisé du scotch pour maintenir les contacts 🤣

Dites33, 2024-01-05:
Hello

On pourrait ouvrir un ticket Z2M pour faire corriger le support et la lecture de la bonne valeur ? :)

tadikwa, 2024-01-05:
> J'avais du inverser le sens du branchement sur la mienne et j'ai utilisé du scotch pour maintenir les contacts 🤣

J'ai testé d'inverser le tx et le rx (on ne sait jamais) mais non pas de résultat.

Au final, j'ai rebrancher le dongle, z2m doit surement se tromper sur la valeur à remonter pour le firmware installé.

Tant que ça marche c'est ok :)

tadikwa, 2024-01-05:
> Hello
> 
> On pourrait ouvrir un ticket Z2M pour faire corriger le support et la lecture de la bonne valeur ? :)

Bonne idée.
En plus sur la nouvelle version 1.35.0 y'a des trucs en + pour le lixee

fairecasoimeme, 2024-01-08:
D'après les captures d'écran (led bleu allumé sur le module TTL), je pense que vous avez grillé l'entrée RX du MCU (en faisant une mauvaise manip)
Vous pourrez mettre à jour OTA mais malheureusement, la mise à jour par USB ne pourra plus se faire.
Fred

tadikwa, 2024-01-10:
Ok, par contre j'ai testé avec deux modules USB TTL neuf et les deux ont le même comportement.

De toute façon j'abandonne la MAJ et j'ouvre un ticket chez Z2M pour l'histoire de remontée de version

closed 2024-01-10 by tadikwa