fairecasoimeme/Zlinky_TIC #233: Zlinky_TIC do not pair when moving from ZHA to Zigbee2MQT

pouilld, 2024-01-05:
HAOS 11.3 CORE 2024.0 
HW RPI 4 
controller Zigbee 3.0 USB Dongle plus (E)

Using the ZHA Zigbee integration, the Zlinky TIC module has been recognized, and paired without problem.

I decided to move from the ZHA integration to zigbee2mqtt without changing anything the zigbee network(controller & devices).
As recommended, I first remove the ZHA integration before installing Zigbee2mqtt (all zigbee devices and entities were removed).
After install/configuration of Zigbee2mqtt, I was able to pair ALL my zigbee devices BUT the Zlink_TIC module (after resetting it), even after installing a additional router near of the module (this was not necessary when using the ZHA integration). The module led flash quickly (waiting pairing) but do not pair.
I am stuck at that point and help will be appreciated.

nerthazrim, 2024-01-10:
I recently moved from a PiZigate+ on a RPI3 to the same Sonoff dongle on a virtualized server and also moved from ZHA to Z2M. I did not have any issue.

When you "Permit join" on Z2M, do you select the coordinator only? Or do you Permit Join (All)? Sometimes devices do not like to be routed and need direct link with the coordinator, have you tried that?

Also, have you tried disconnecting the ZLinky completely from the Linky, re-attaching it and retried the pairing?
Last idea: try to push the reset button for more than 15s, release, then Permit join on Z2M and reset Zlinky for 5s. That's what I did when I moved, without problem.

pouilld, 2024-01-19:
I have replaced my actual coordinator (SonOFF  Zigbee 3.0 USB Dongle plus (E)  EFR32MG21) by a ZIGATE V2 USB dongle (firmware  ZIgate2-3a0.
Now the Zlinky **pair with Z2M** (as well as all other devices).
So, the problem seems related to an incompatibility between Z2M  and this version of the Sonoff coordinator.
However, with the ZIgate coordinator, it is not possible to upgrade the Zlinky firmware using OTA and it s is a problem already known by the Zlinky manufacturer.

nerthazrim, 2024-01-19:
I seriously doubt incompatibility is the reason, as I've done pretty the same thing as you very recently and changed my coordinator from a PiZigate+ with ZHA to a SonOFF Zigbee 3.0 USB Dongle plus (E) with Z2M.
Worked like a charm without any problem for pairing.

pouilld, 2024-01-19:
In that case, I don't see  an explanation why by just replacing the Sonoff coordinator by the ZIgate, the Zlinky that was unable to pair with Z2M was now able to pair with Z2M. 
Please note that in both case, the network was reinitialized by the coordinator on the same Zigbee channel (11).

pouilld, 2024-01-23:
After doing a "full reset" (re-initialization of the zigbee network, removing devices and entities in HA, uninstall of Z2M, deleting the Z2M configuration folder in ZHA, and installing Z2M again, I was able to successfully  pair the Zlinky with Z2M.
I also was able update thru OTA the Zlinky firmware to V13.
Unfortunately after the update, some information about my "HP-HC" were not anymore sent by the Zlinky.
Thanks for your help,
Daniel

pouilld, 2024-01-23:
subject closed, thanks

closed 2024-01-23 by fairecasoimeme