fairecasoimeme/Zlinky_TIC #236: Coupure de l'alimentation du zlinky par le Linky : que faire ?

jessycler, 2024-01-21:
Bonjour,
Je possède un zlinky depuis environ 9 mois, et j'étais ravi de son fonctionnement sous HA via Z2M.
Il y a 15j, je n'avais plus de remontée des infos zlinky dans HA. En allant voir le Linky, j'ai vu que le voyant bleu du zlinky était éteint...
Je l'ai donc débranché du Linky, puis rebranché, sans qu'il ne redémarre. Je l'ai laissé débranché, pensant qu'il était en panne. Le lendemain, par acquis de conscience,  je l'ai rebranché. Et là "miracle", la led bleue s'allume, et directement les infos remontent dans HA. Ouf !
Sauf qu'hier, bim, plus de remontée dans HA, je vais voir le zlinky, il est éteint...

J'ai lu par ailleurs sur un forum, que le Linky pouvait mettre en "protection" l'alimentation via les bornes TIC. Cela expliquerait que le zlinky s'éteigne. Est ce vrai ? Et si oui, pourquoi se met il en protection ? Est ce un défaut du Linky ou du zlinky ? Faut il que je fasse une demande de SAV pour mon zlinky ? Ou pour le Linky chez Enedis ?

Merci de partager avec moi vos connaissances sur le sujet ;-)

fairecasoimeme, 2024-01-22:
Bonjour,
Oui par expérience cela peut arriver sur un Linky qui se met à jour ou lors d'une mauvaise manipulation ou lors de microcoupure. C'est propre au Linky et Enedis investigue sur ce genre de phénomène.
Fred

jessycler, 2024-01-22:
Merci Fred pour la réponse,
Donc il faudrait que je demande à Enedis s'ils peuvent faire quelque chose ?
J'ai oublié de préciser que j'avais en parallèle 2 fils branchés sur le bornier TIC du Linky (bornes 1 et 2), en plus du zlinky, qui transmettent le signal TIC à une borne de recharge de VE (permet de faire du délestage).
Je pensais que ça ne poserait pas de problème (quand le zlinky ne s'éteignait pas, je n'avais pas ces deux fils), mais est ce que ça pourrait avoir un lien avec une "mise en protection" du Linky ?

nicolas-r, 2024-02-06:
Salut,
je pense que j'ai le même problème, mon zlinky ne s'allume plus et j'ai essayé avec un autre modele sans antenne et il ne s'allume pas non plus. Je penche donc pour un probleme coté Linkly mais je sens venir la galère avec Enedis...

Je vais tenter de les contacter et on verra bien

nicocha75, 2024-02-08:
bonjour, perso depuis 2 jours, plus de remonté, LED éteinte.
J'ai donc rapatrier à la maison pour alimenté via la pris de programmation., et ça fonctionne. Le problème vient donc du linky et certainement d'une mise a jours qui a coupé l'alimentation TIC.

nicolas-r, 2024-02-08:
J’ai envoyé un mail à Enedis il y a deux jours mais pas encore de réponse. Mais quand on voit que l’espace client est en rade depuis plusieurs jours, plus rien ne m’étonne.

Je vais les appeler demain pour tenter d’avancer

fairecasoimeme, 2024-02-08:
Essayez de rebrancher le ZLinky. Après une courte ou longue période de coupure, l'alimentation revient parfois.

nicolas-r, 2024-02-09:
Je l'ai rebranché ce matin et toujours rien. J'ai appelé Enedis qui m'a dit que cela arriver parfois en effet et m'a demander d'appeler mon fournisseur d'électricité afin qu'il fasse une demande F920.

J'ai donc appelé EDF qui a initié la demande et Enedis devrait m'appeler lundi pour convenir d'un rendez-vous

A suivre

nicolas-r, 2024-02-15:
Bonjour

toujours pas de nouvelles de la part d'Enedis et/ou d'EDF. Je vais le rebrancher dans la journée pour voir s'ils n'auraient pas fait un truc à distance au cas où

nicocha75, 2024-02-15:
Bonjour,
Si enedis 1 reseter le linky, et samedi matin, après un débranchement durant 2h, ça a refonctionné.
Ça n'a pas marché pour toi?
La classe, EDF m'a bien rappelé pour savoir si tous fonctionnait.

nicolas-r, 2024-02-16:
Toujours rien pour moi et pas de news d'EDF/Enedis

Je vais faire tirer un câble entre mon compteur en limite de propriété et l'intérieur de la maison pour ne plus dépendre de l'alimentation de la prise TIC

Allsimon, 2024-05-08:
est-ce que vous avez trouvé un fix ? mon linky est down depuis le 7 avril et ne s'est jamais rallumé depuis

nicocha75, 2024-05-19:
Salut, WE prolongé, et pas de chance certainement enedis qui à refait un update du linky, plus d'alimentation. Galère cette gestion à la noix du linky. Question, savez vous si on peut directement alimenté le module entre I1 et A avec une alimentation 5V, afin de ne plus dépendre de l'alimentation du linky?

nicolas-r, 2024-05-19:
Hello

je viens de tester mes deux ZLinky sur le compteur d'un ami et aucun ne s'allument. Il a un compteur Sagem et moi un Honeywell. On a testé avec un appareil de mesure et il y a bien du courant qui sort du port TIC.

Pour ma part, le câble a été tiré donc je n'ai plus besoin du Zlinky pour le moment

VianneyHENON, 2024-06-13:
> Essayez de rebrancher le ZLinky. Après une courte ou longue période de coupure, l'alimentation revient parfois.

Bonjour, est-ce que utilisé une [alimentation externe](https://github.com/fairecasoimeme/Zlinky_TIC/issues/262) avec un module 5V pourrait éviter ce désagrément ?

fairecasoimeme, 2024-06-17:
Bonjour, Oui si vous n'utilisez pas l'alimentation de la prise TIC du Linky, vous n'aurez pas ce type de problème.

Deejayfool, 2024-10-02:
Hello tout le monde,
J'ai exactement ce même soucis et j'avoue que débrancher/rebrancher le ZLinky est plutôt ennuyant. Le problème revient quasiment tous les 2 mois.
Est-ce que quelqu'un a des news de la part d'EDF/Enedis sur une correction de ce problème ?
Ou bien est-ce que quelqu'un a une solution de contournement simple ? (parce que la solution de l'alimentation externe semble hors de mes compétences 😅)

fairecasoimeme, 2024-10-03:
Contactez-moi sur le mail de lixee pour tenter avec un échange... sait-on jamais.

fairecasoimeme, 2024-10-07:
Bonjour,

Non pas encore de version plus récente (bientôt) mais si vous le souhaitez on peut tenter aussi de faire un échange aussi mais sans garanti de changement. (surtout si cela provient du Linky)

CyrilM81, 2025-01-15:
Bonjour à tous, 

Ce qui vous arrive avec les modules tic qui stoppent de fonctionner m'est arrivé plusieurs fois . Je n'en connais pas la cause mais le resultat est l'arrêt total du fonctionnement du module installé sur le TIC.
De mon coté , J'ai résolu le problème de manière simple et efficace . Sur le coté de l'emplacement du TIC se trouve un fusible dans un morceau de plastique noir, dévissez le , sortez le , attendez 30 secondes, et replacez le fusible !
Dans mon cas , le fusible n'a jamais grillé , mais le fait de le sortir affectue un RESET de l'electronique qui gère la fonctionnalité TIC dans le Linky .

Pour moi cette technique a marché à chaque fois ....

J'espère que pour vous aussi ...

Deejayfool, 2025-06-19:
Bonjour,
J'ai envoyé un mail en octobre dernier comme demandé afin de tenter un échange mais je n'ai jamais eu de retour.
Du coup je voulais savoir si certaines personnes étaient passé au Lixee v2 et si le problème survenait encore avec cette nouvelle version ?

fairecasoimeme, 2025-06-19:
avez-vous envoyé un mail à contact[at]lixee.fr ? Retenter ce jour en mettant le lien ou votre pseudo dans le mail pour que je puisse faire le lien

Deejayfool, 2025-06-19:
Oui c'était bien à cette adresse que j'avais écrit. Je viens de vous renvoyer un mail à l'instant. Merci à vous 🙏.

chmtc94, 2026-05-03:
LiXee finalement OK à nouveau au rebranchement après deux heures débranché ...

eekamouze, 2026-08-16:
Bonjour,
Je possède la version 1 avec antenne déportée depuis plusieurs années maintenant. A chaque orage, micro coupures, mises à jour du Linky (supposée) je perds la connexion et suis obligé de le débrancher 1 bonne heure avant de le rebrancher. Mon compteur est à l'extérieur et ça devient franchement pénible, sans compter quand ça arrive en mon absence. J'ai entendu parler de remplacement et j'aimerais un retour d'expérience pour savoir si le remplacement par un v2 corrige le problème ou si le problème est propre au linky et que donc il n'y a rien à faire. Merci