fairecasoimeme/Zlinky_TIC #238: ZLinky ne s'allume pas

philuxe, 2024-02-06:
Bonjour,

Je viens de recevoir le module mais ce dernier ne s'allume pas:

[image]

que faut il faire ?

philuxe, 2024-02-06:
j'ai tenté la mesure avec un multimètre mais je pense que les sondes sont trop epaisses pour faire contact donc forcément 0 courant.

du coup j ai essayé de plugin le Zlinky dans un autre Linky et ca ne marche pas non plus, tjs pas de LED

[image]

ketourneau, 2024-02-09:
J'ai le même problème avec le ZLinky reçu hier. Il ne s'allume pas non plus.

ketourneau, 2024-02-09:
Je pense que c'est en lien avec #236

ketourneau, 2024-02-09:
J'ai contacté EDF qui va faire la demande à ENEDIS pour passer la prise TIC du Linky en mode "Standard" à voir si ça corrige le problème.

https://particuliers.engie.fr/aide-contact.html?question=comment-changer-le-mode-de-fonctionnement-de-mon-compteur

philuxe, 2024-02-16:
recu nouveau module, il s'allume correctement

closed 2024-02-16 by philuxe

ketourneau, 2024-02-16:
Après avoir vérifié avec un voltmètre j'ai bien une tension. Le boîtier zlinky est donc défectueux, quel est la procédure pour un échange ?

hichamC31300, 2024-08-29:
Bonjour
Je rencontre le même problème. Tension de 16V aux bornes du Linky. Mon zlinky reste quand même éteint. Pensez-vous qu'il est défectueux ?