fairecasoimeme/Zlinky_TIC #242: Pas de Remontée d'info du tout

Christof71800, 2024-02-22:
Bonjour j'ai instalé le Zlinky
reconnu dans HA via Z2M 
je suis en TEMPO EDF
j'ai configuré comme ca 
[image]

cependant je n'ai aucune info qui remonte 
[image]

j'ai retiré et réapairé le Zlinky, essayé un autre sans succes 

est ce un defaut de mon compteur, enedis peut intervenir ? 

merci

fairecasoimeme, 2024-02-22:
Bonjour,
Avez-vous fait la demande auprès de votre fournisseur d'énergie la demande de passage en mode standard ?

crazymikefra, 2024-03-01:
Bonjour,

Je suis également en Tempo mais j'ai tout laissé sur les valeus par défaut (auto).
As-tu essayé en auto ?

Par contre, ce qui m'étonne, c'est que tu es en "three phase", tu as réellement du tri-phasé chez toi ?