fairecasoimeme/Zlinky_TIC #243: Z2T Cannot not able to receive attribute reporting after subscription

xav-git, 2024-03-14:
Hello, 

I have integrated the Lixee dongle on a Sonoff ZBbridge Pro running Tasmota 13.4.0.2, using Z2T to push the Sensor information to MQTT broker.  Zigbee integration is successfull and I can `read` data:
```
20:16:14.302 CMD: ZbSend { "device":"0xCDAB", "endpoint":"0x01", "cluster":"0x0702", "read":["0x0000", "0x0100","0x0102"]}
20:16:14.317 MQT: stat/tasmota_lixee/RESULT = {"ZbSend":"Done"}
20:16:14.858 MQT: tele/tasmota_lixee/SENSOR = {"ZbReceived":{"0xCDAB":{"Device":"0xCDAB","Name":"Lixee","CurrentSummationDelivered":"0x000004A36C4D","0702/0100":"0x00000293D209","0702/0102":"0x0000020F9A44","Endpoint":1,"LinkQuality":29}}}

```

I am also able to bind to cluster `0x0702` :
```
21:45:32.948 CMD: Zbbind {"Device":"0xCDAB","Cluster":"0x0702","Endpoint":1}
21:45:32.958 MQT: stat/tasmota_lixee/RESULT = {"ZbBind":"Done"}
21:45:33.098 MQT: tele/tasmota_lixee/RESULT = {"ZbBind":{"Device":"0xCDAB","Name":"Lixee","Status":0,"StatusMessage":"SUCCESS"}}
```

And Verify bind is successful :
```
21:46:11.373 CMD: zbbindstate 0xCDAB
21:46:11.385 MQT: stat/tasmota_lixee/RESULT = {"ZbBindState":"Done"}
21:46:11.476 MQT: tele/tasmota_lixee/RESULT = {"ZbBindState":{"Device":"0xCDAB","Name":"Lixee","Status":0,"StatusMessage":"SUCCESS","Total":1,"Start":1,"Bindings":[{"Cluster":"0x0702","Endpoint":1,"ToDevice":"0x00124B002C400527","ToEndpoint":1}]}}
```

However the issue I have is that the lixee is not sending the reportable information availble in cluster `0x0702`

yes I am not sure if the problem comes from Tasmota or Lixee, but I would like to confirm that sensor reporting is supported and that the bind to cluster is the correct way to receive such reports.

Thanks.

fairecasoimeme, 2024-03-14:
Hi,
You need to do a "Configure Report" after a "Bind" to get reportable information.
Just a bind is not enough
Fred

xav-git, 2024-03-21:
This problem has been resolved in https://github.com/arendst/Tasmota/discussions/20952 with a fix in tasmota.
Tasmota now successfully receives sensor report from Lixee.

However it seems that `ReportableChange` is not taken into account by Lixee. Indeed the lixee will always report sensor values after `MaxInterval` is passed, although it can be verified that counters always changes.
Is this an expected behavior ?
Thanks.

fairecasoimeme, 2024-03-22:
Sorry but reportable change works with others systems

xav-git, 2024-03-22:
Thanks, for your answer. 
From your comment I understand that you suspect that the coordinator (in this case Tasmoata Z2T) is the problem ? 

So if if this is the case it should be the  report config of `ReportableChange` that is not  achieved correctly ?