fairecasoimeme/Zlinky_TIC #244: Passer d'un firmware `legacy` à un `limited` pour un ZLinky v1.2 : comment faire ?

MilesTEG1, 2024-03-15:
Bonjour,

J'ai ce ZLinky depuis plusieurs mois maintenant, et j'ai de plus en plus d'équipement Zigbee dans mon réseau. Et j'ai constaté que de plus en plus se connecte sur le ZLinky au lieu de se connecter sur d'autres routeurs à peine plus loin (comme des prises connectées).
Le souci est que mon Linky est en TEMPO chez EDF en mode standard, donc remonte beaucoup d'informations, et les capteurs de températures connectés sur le ZLinky envoient eux aussi pas mal de données. J'ai constaté des indisponibilités fréquentes de ces capteurs ainsi que des remontées d'informations du ZLinky que je pense dues au firmware du ZLinky qui est en mode `legacy`.
J'aimerais passer le ZLinky en firmware `limited` pour ne plus avoir ces connexions sur lui, mais je ne sais pas comment procéder.
J'ai acheté mon ZLinky le 03/12/2023, donc assez récemment, et je ne sais pas s'il est possible de flasher en OTA la version limited par-dessus une version legacy, car dans la documentation GitHub ce n'est pas très clair.

Pouvez-vous m'aider ? Et s'il faut, me prêter un module pour faire cette MAJ ?

PS : voilà le ZLinky que j'ai reçu (modèle avec antenne déporté) : 
[image]

Merci par avance

MilesTEG1, 2024-03-17:
@fairecasoimeme 
Bonsoir,
J'essaie désespérément de changer le firmware en limited, avec un module USB-TTL ( https://www.amazon.fr/dp/B089QJZ51Z?psc=1&ref=ppx_yo2ov_dt_b_product_details ) 
[image]

J'ai pu le connecter sur le ZLinky, et le maintenir avec du scotch noir, la connexion se fait puisque le ZLinky clignote en bleu.
Mais quoi que je tente j'ai toujours cette erreur :
[image]
J'ai essayé d'intervertir le RX et le TX, d'appuyer sur le bouton du ZLinky avant de brancher l'USB sur l'ordinateur, et de lancer la commande le plus vite possible une fois branché et le bouton laché...
Mais rien à faire, je n'y arrive pas.
J'ai mis le USB-TTL en port COM70, avec un baudrate de 115200 dans le gestionnaire de périphérique (j'ai donc installé le pilote), la commande que je lance est:
```
.\DK6Programmer.exe -s COM70 -P 115200 -e FLASH -Y -p .\ZLinky_router_v13_limited.bin
```
J'ai aussi tenté la commande qui n'efface pas les appairages : 
```
.\DK6Programmer.exe -s COM70 -P 115200 -Y -p .\ZLinky_router_v13_limited.bin
```

Mais à chaque fois, j'ai le "Read error"...

Pouvez-vous m'aider ?
Merci d'avance.

MilesTEG1, 2024-03-17:
Bonsoir,
Bon, j'ai réussi à flasher le firmware Limited.
Je n'avais pas vu qu'il y avait deux boutons sur le ZLinky...
Il se peut grandement que j'appuyais avant sur le bouton d'appairage... au lieu de l'autre bouton.
Mais j'ai aussi placé mes connecteurs avec une autre inclinaison, toujours avec du scotch.

Méthode suivie :

1. Télécharger et installer les pilotes ici (le lien dans la doc n'est plus valide) : https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads
Moi j'ai pris la version [CP210x Universal Windows Driver v11.3.0 du 6/24/2023](https://www.silabs.com/documents/public/software/CP210x_Universal_Windows_Driver.zip)
2. Brancher l'adaptateur USB-TTL seul, pour configurer le port COM et la baudrate : 
   - Choisir le pilote dans le panneau de configuration s'il n'est pas installé seul ;
   - Dans les propriétés du pilote, choisir un port COM loin des autres au besoin si vous en utilisez déjà (j'ai choisi COM70)
   - Paramétrer le baudrate sur 115200 ;
   - Valider les préférences.
4. Télécharger le programme de flashage donné dans la doc : https://zigate.fr/documentation/mise-a-jour-de-la-zigate-2/
5. Télécharger le firmware Limited : https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v13.0/ZLinky_router_v13_limited.bin
6. Placer tous les fichiers dans le même dossier ;
7. Préparer la commande à valider d'une touche dans une invite PowerShell administrateur (pas sûr que ce soit nécessaire) :
   ```powershell
   .\DK6Programmer.exe -s COM70 -P 115200 -e FLASH -Y -p .\ZLinky_router_v13_limited.bin
   ```
8. Brancher l'adaptateur USB-TTL sur le ZLinky (avec des fils dupont, ou autre nappe, ou directement + utiliser du scotch d'électricien isolant pour maintenant la connexion) ;
9. Appuyer sur le bouton du ZLinky situé de l'autre côté du PCB que le bouton d'appairage ;
10. Maintenir le bouton appuyé, et brancher l'USB sur l'ordinateur ;
11. Rapidement valider la commande avec la touche entrée ;
12. Relâcher rapidement le bouton du ZLinky ;
13. Constater le bon déroulement du flashage (si tout s'est bien passé 😄).
[image]

PS : Pour le point 2 et le baudrate, je ne suis pas sûr que ça ait une incidence... car j'ai lancé avant la commande du point 7 la commande suivante qui s'est passée avec succès également :
```powershell
.\DK6Programmer.exe -s COM70 -P 500000 -Y -p .\ZLinky_router_v13_limited.bin
```

### Maintenant questions :

1. Est-ce normal que dans Z2M le ZLinky soit toujours considéré comme un router ?
   [image]

   Pour le moment, je n'ai plus rien de connecter sur lui : 
   [image]

2. Z2M me dit qu'il y a une MAJ de firmware :
   [image]
   J'ai peur que si je la lance ça me fasse bugger le ZLinky...
   La MAJ sera-t-elle bien la limited.OTA ?
   Je demande ça car lorsqu'il y aura une MAJ dans le futur, je voudrais la faire correctement.
   
Merci d'avance

fairecasoimeme, 2024-03-18:
Réponse :
1- Oui c'est normal qu'il soit en routeur car en "limited", il reste un routeur. Juste qu'il limitera le routage d'un seul appareil.
2- Z2M ne va pas chercher au bon endroit pour connaître la mise à jour du ZLinky. De plus, en limited, vous êtes désormais en 4001-0013. Z2M ne va chercher que la mise à jour du mode "legacy". Pour les prochaines mises à jour et pour le moment, il faudra le faire avec le module USB

MilesTEG1, 2024-03-18:
@fairecasoimeme 
Merci pour la réponse.

1- OK, merci pour la précision.

2- C'est dommage pour les MAJ impossible en OTA, mais Ok, ça ne prend pas tant de temps que ça à faire en USB maintenant que j'ai compris comment faire ^^

3- j'ai constaté que Z2M n'indique pas la bonne version du firmware :
[image]
It should be seen as `4001-0013` not `4000-0013`.

Et dans la console dev du ZLinky (dans Z2M), j'ai ceci comme informations :
swBuildId = `4001-0013` alors que  hwVersion = `1`.
[image]

Est-ce que c'est normal ? Comment corriger cela pour que Z2M ne cherche plus à faire les maj OTA impossible à faire et ne plus avoir les notifications de MAJ disponibles ?

ShagoY, 2024-03-19:
@MilesTEG1 Bonjour, le flashage en mode limited t'a amélioré tes pertes/indisponibilités ?

MilesTEG1, 2024-03-19:
> @MilesTEG1 Bonjour, le flashage en mode limited t'a amélioré tes pertes/indisponibilités ?

@ShagoY bonsoir, je n'ai pas encore assez de recul mais je n'ai pas encore eu de décrochage de valeurs remontées par le Zlinky.

Mais en // du firmware du Zlinky j'ai aussi mis un nouveau firmware sur ma clé SkyConnect et passé le driver en `ember` au lieu de `ezsp` via la version dev de zigbee2mqtt dans Home Assistant.
😅

luke7101, 2024-03-21:
> Bonsoir, Bon, j'ai réussi à flasher le firmware Limited. Je n'avais pas vu qu'il y avait deux boutons sur le ZLinky... Il se peut grandement que j'appuyais avant sur le bouton d'appairage... au lieu de l'autre bouton. Mais j'ai aussi placé mes connecteurs avec une autre inclinaison, toujours avec du scotch.
> 
> Méthode suivie :
> 
> 1. Télécharger et installer les pilotes ici (le lien dans la doc n'est plus valide) : https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads
>    Moi j'ai pris la version [CP210x Universal Windows Driver v11.3.0 du 6/24/2023](https://www.silabs.com/documents/public/software/CP210x_Universal_Windows_Driver.zip)
> 2. Brancher l'adaptateur USB-TTL seul, pour configurer le port COM et la baudrate :
>    
>    * Choisir le pilote dans le panneau de configuration s'il n'est pas installé seul ;
>    * Dans les propriétés du pilote, choisir un port COM loin des autres au besoin si vous en utilisez déjà (j'ai choisi COM70)
>    * Paramétrer le baudrate sur 115200 ;
>    * Valider les préférences.
> 3. Télécharger le programme de flashage donné dans la doc : https://zigate.fr/documentation/mise-a-jour-de-la-zigate-2/
> 4. Télécharger le firmware Limited : https://github.com/fairecasoimeme/Zlinky_TIC/releases/download/v13.0/ZLinky_router_v13_limited.bin
> 5. Placer tous les fichiers dans le même dossier ;
> 6. Préparer la commande à valider d'une touche dans une invite PowerShell administrateur (pas sûr que ce soit nécessaire) :
>    ```powershell
>    .\DK6Programmer.exe -s COM70 -P 115200 -e FLASH -Y -p .\ZLinky_router_v13_limited.bin
>    ```
> 7. Brancher l'adaptateur USB-TTL sur le ZLinky (avec des fils dupont, ou autre nappe, ou directement + utiliser du scotch d'électricien isolant pour maintenant la connexion) ;
> 8. Appuyer sur le bouton du ZLinky situé de l'autre côté du PCB que le bouton d'appairage ;
> 9. Maintenir le bouton appuyé, et brancher l'USB sur l'ordinateur ;
> 10. Rapidement valider la commande avec la touche entrée ;
> 11. Relâcher rapidement le bouton du ZLinky ;
> 12. Constater le bon déroulement du flashage (si tout s'est bien passé 😄).
>     ![image](https://private-user-images.githubusercontent.com/29298312/313504938-9803d645-c1ee-4f4f-9767-b97d53809577.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEwNDIxODIsIm5iZiI6MTcxMTA0MTg4MiwicGF0aCI6Ii8yOTI5ODMxMi8zMTM1MDQ5MzgtOTgwM2Q2NDUtYzFlZS00ZjRmLTk3NjctYjk3ZDUzODA5NTc3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIxVDE3MjQ0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdhOGYxYzRmYzVjNThlMmI0OWI3Y2M2M2VmZjc0YjU1MGFhMDI5OGU4YzUzY2FkM2FjOTRkMGI5MDU5Mzc4NTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.0FMCgS7kZDV4nzG6fj0vzvEr0SeZR4xE4tpHjivE13A)
> 
> PS : Pour le point 2 et le baudrate, je ne suis pas sûr que ça ait une incidence... car j'ai lancé avant la commande du point 7 la commande suivante qui s'est passée avec succès également :
> 
> ```powershell
> .\DK6Programmer.exe -s COM70 -P 500000 -Y -p .\ZLinky_router_v13_limited.bin
> ```
> 

Merci pour ce tuto en tout cas ! Je cherchais ça il n'y a pas très longtemps et puis j'ai laissé tomber !

MilesTEG1, 2024-09-03:
Bonjour,
Je viens de réutiliser ma méthode pour flasher le ZLinky en v14, mais cette fois-ci avec le firmware standard, j'en ai marre de voir Z2M me dire qu'il n'est pas à jour.
Le processus fonctionne encore bien.
Mais j'ai dû lancer une machine windows pour le faire, car je n'ai pas trouvé le binaire macos pour faire le flashage.
Si quelqu'un a un binaire pour macOS, je suis preneur ;)

Sleinous, 2024-09-03:
J'ai suivi ton exemple pour le flashage (quelques heures avant mon issue et ta réponse d'ailleurs) mais je n'arrive pas a obtenir autre réponse que ton premier message : Error opening connection : read error.
Je continue donc a tenter les maj via OTA, j'en suis a ma 14ème tentative, j'ai l'habitude maintenant.... ayant fait des maj depuis la v5...

elmincho80, 2024-09-03:
Salut les gars 👍 je vais faire la MAJ aussi en no OTA car plantage systématique du Zlinky en OTA ... grrr

Autre question @MilesTEG1  Ton passage de EZSP a Ember tu as eux besoin de re apairer tes services ? 

Merci 

Stephane

elmincho80, 2024-09-03:
J'ai réussi sans problème par contre il affiche tjrs l'ancienne version :( 

[image]

MilesTEG1, 2024-09-03:
> J'ai réussi sans problème par contre il affiche tjrs l'ancienne version :( 
> 
> 
> 
> [image]
> 
> 
> 
> 

Oui j'ai eu le cas moi aussi. J'ai fait une Maj OTA ensuite et c'est ok maintenant since n'est la date du firmware 🤪

elmincho80, 2024-09-03:
Ouai je viens de relancer l'OTA et qui vient de planter à 66% grrrr

Sleinous, 2024-09-03:
Il faut continuer à insister :)

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

elmincho80, 2024-09-03:
Ouai c'est ce que je fais ... mais bon aucun intérêt de l'Upgrade en non OTA lol vu que t'es obligé de pousser l'OTA ensuite. Et relou d'aller le débrancher rebrancher à chaque fois. Chance qu'il n'y a pas de MAJ tous les mois lol

MilesTEG1, 2024-09-03:
> Salut les gars 👍 je vais faire la MAJ aussi en no OTA car plantage systématique du Zlinky en OTA ... grrr
> 
> 
> 
> Autre question @MilesTEG1  Ton passage de EZSP a Ember tu as eux besoin de re apairer tes services ? 
> 
> 
> 
> Merci 
> 
> 
> 
> Stephane 

Alors pour mon passage à ember , ça a été plus compliqué car j'avais des soucis d'erreurs dans Z2M, et j'ai aussi changé de coordinateur (avant une Clé SkyConnect) pour passer sur une SLZB-06M (Adaptateur PoE).
Donc j'ai complètement cassé mon réseau Zigbee en le recréant et donc en reappairant les devices.