fairecasoimeme/Zlinky_TIC #245: Problème avec abonnement HC et Week-End

anthonycorreia, 2024-03-17:
Bonjour,

J'ai un abonnement HC et Week-End sur un Linky en TIC mode standard et a chaque changement de tarif (HC, HP, WE) le module arête d'envoyer les infos.

Ma config :

Zlinky_TIC 13
Zigbee2MQTT 1.36.0-1
Mosquitto broker 6.4.0
Home Assistant Core 2024.3.1, Supervisor 2024.03.0, Operating System 12.1

Dispo pour tout autre info

fairecasoimeme, 2024-03-18:
Bonjour,
Le ZLinky n'a aucun traitement particulier lors d'un changement de tarif. Il n'y a aucune raison directe engendrant un arrêt d'envoi selon ce paramètre.
Je ne pense pas que le ZLinky arrête d'envoyer des données mais plutôt qu'une perturbation bloque la réception.
Il faut vérifier qu'un changement de tarif n'engendre pas le démarrage d'un moteur (qui peut entrainer des perturbations) ou qu'une prise connectée n'inonde pas le réseau etc ...

DolmensDude, 2024-03-25:
not certain, this is the same, but I have roughly the same config:

Core. 2024.3.3
Supervisor. 2024.03.0
Operating System. 12.1
Frontend. 20240307.0

Device info
Lixee ZLinky (ZLinky_TIC)
by LiXee
Connected via [Zigbee2MQTT Bridge]
Firmware: 4000-0013

Issue being that regardless of whether config'd via ZHA or zigbee2MQTT, the Zlinky is report all consumption on in the 1st rate, and also always shows the current tariff as BASE, even though on the Linky itself it shows my tariffs as HC, PH and Weekends...  

I have no idea how to get the Zlinky to properly report parameters correctly.  The usage per hour is correct, it's just not well aligned with the Linky itself, with no idea who to look to for fixing this?

In ZHA, the quirk file is seemly not doing anything?, does this need to be an exact match firmware and quirk file?
in MQTT mode, which is the one recommended in the instructions here, again, doesn't match.
EDF & Moi app on the iPhone *does* split out by tarif, but is a day "late" in reporting.
Enedis, using the ConsoAPI via Linky integration is also correct, but again sees all consumption as one tariff.

Who do I look to to get this resolved?  Have gone back and forth twice now between ZHA and MQTT mode... only advantage to MQTT was that I could get the update via OTA as that's currently disabled in ZHA...

Sensors
 
ADSC
--------------
ADIR1
0 A
 
ADIR2
0 A
 
ADIR3
0 A
 
Energy
105,385 kWh
 
IINST1
1 A
 
IINST2
2 A
 
IINST3
1 A
 
IMAX1
60 A
 
IMAX2
60 A
 
IMAX3
60 A
 
ISOUSC
30 A
 
MOTDETAT
None
 
NGTF
BASE
 
PAPP
1,190 VA
 
PMAX
6,410 W
 
PPOT
None
 
PTEC
TH.

KanaChan06, 2024-04-27:
Bonjour,

J'ai le même problème que DolmensDude.
Je suis en EDF Zen Weekend (sans HC) et tout remonte sur la première ligne de BASE, même le weekend.
Savez-vous comment faire pour mapper le changement de tarification ?
Merci.

DolmensDude, 2024-04-27:
this problem is resolved for me as of 5 days ago.

You need to log into your EDF account online and change under your preferences (profile) that you want "standard" data, so no historical data.

***AND*** (this too is required)

EDF needs to request of Enedis, since they manage the Linky, that Enedis do a so-called "F185 prestation".

The next day, you will start to receive data, but this too will still be garbage, but once it's different garbage, you have only one more step to do.

**Last step**, disconnect HomeAssistant from the Linky (lose the bluetooth connection), and then reconnect (via bluetooth) to the Linky... It will then "know"...

You will receive separate data for tier 1, tier 2, and tier 3, tier 1 is lower rates during the week, tier 2 is normal rate, and tier 3 is the weekend, even if this is the same as your tier 1 rate.

It took weeks to get this fixed, mostly due to EDF saying that Enedis required proof that this data was wrong.  Waited 3 weeks and I sent a reminder email to the customer service folks, and the next evening it was all done.

It is that I do not write French, otherwise I would want to help write up the steps needed when using standard zigbee versus MQTT bluetooth.  I'm a retired technologist - this was not easy to do for "normal folks". These days, I'm trying to be more like a normal person instead of a technologist. :-)

At the end of the day, it *is* working here now and I can begin to do with this what I needed/wanted to do!

DolmensDude, 2024-04-27:
adding to help close this one - I switched to ZHA, and then went the steps above. It works.

There might be benefit to going the zigbee2MQTT route, but I really wanted to use "standard" ZHA. Ultimately, I got this to work and have been happy to see this data come into HA.  There is still work to be done, but at least consumption per hour data is good.

KanaChan06, 2024-05-02:
> this problem is resolved for me as of 5 days ago.
> 
> You need to log into your EDF account online and change under your preferences (profile) that you want "standard" data, so no historical data.
> 
> _**AND**_ (this too is required)
> 
> EDF needs to request of Enedis, since they manage the Linky, that Enedis do a so-called "F185 prestation".
> 
> The next day, you will start to receive data, but this too will still be garbage, but once it's different garbage, you have only one more step to do.
> 
> **Last step**, disconnect HomeAssistant from the Linky (unpair the bluetooth connection), and then re-pair (via bluetooth) to the Linky... It will then "know"...
> 
> You will receive separate data for tier 1, tier 2, and tier 3, tier 1 is lower rates during the week, tier 2 is normal rate, and tier 3 is the weekend, even if this is the same as your tier 1 rate.
> 
> It took weeks to get this fixed, mostly due to EDF saying that Enedis required proof that this data was wrong. Waited 3 weeks and I sent a reminder email to the customer service folks, and the next evening it was all done.
> 
> It is that I do not write French, otherwise I would want to help write up the steps needed when using standard zigbee versus MQTT bluetooth. I'm a retired technologist - this was not easy to do for "normal folks". These days, I'm trying to be more like a normal person instead of a technologist. :-)
> 
> At the end of the day, it _is_ working here now and I can begin to do with this what I needed/wanted to do!

Hello,
Thanks for the help.
I have set the TIC up from the Website but, even if I have received an email from EDF to tell me I am in standard mode, then the Linky says I am in "historique" mode for the TIC onto its interface.
I have sent a demand from a form, I will see if it works better.

SeafireGit, 2024-09-14:
Issue can be closed, right ?

KanaChan06, 2024-09-14:
Hello,

 

Yes. It can be.

 

Thanks.

closed 2024-09-14 by fairecasoimeme